esphome/esphome-core #321: mqtt_json platform depreciated in Homeassistant 0.84

cbrherms, 2018-12-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

All

**ESP (ESP32/ESP8266/Board/Sonoff):**

All

**Affected component:**

All components that use the mqtt_json platform as output

**Description of problem:**
Changes to mqtt lights in homeassistant made `mqtt_json` platform depreciated, this is now incorporated in to one platform, `light.mqtt`, with the type of mqtt light configured via a schema. 
This should only affect devices set up via mqtt discovery using the old type of config, and users who manually setup their lights in configuration.yaml
Pull that implemented this and describes the change is here: https://github.com/home-assistant/home-assistant/pull/18227

**Problem-relevant YAML-configuration entries:**

Old configuration entry
```yaml
light:
  - platform: mqtt_json
    command_topic: "home/rgb1/set"
```

New entry
```yaml
light:
  - platform: mqtt
    schema: json
    command_topic: "home/rgb1/set"
```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2018-12-14:
What's the issue here? This is a [documented](https://www.home-assistant.io/blog/2018/12/12/release-84/#breaking-changes) breaking change, which do happen from time to time.

thubot, 2018-12-15:
So it seems like there is still some work ahead for #322, right?
From what I can understand, fixing this directly would only require a minor change to the discovery message format.
Are you working on that? If not, I might look into it, because right now this breaks all my lights.

OttoWinter, 2018-12-15:
@thubot No, it doesn't break any lights - they might have been _renamed_ to `light.the_light_2` but that cannot be fixed (HA issue).

The `mqtt_json` removal is a deprecation and is backwards-compatible for the MQTT protocol (I know, if you look at the PR that implemented it on the HA side you'll see me as a reviewer).

I won't change this to the new "schema" format until something like esphomelib v1.11.0 or even 1.12. The reason is that many users don't immediately update their HA installations and if I upgrade to the newer format it would break it for everybody with older versions of HA.

If I don't change it just yet, it is only a minor annoyance as a warning message gets logged, but lights still work.

PR #322 (2018-12-18): Native Esphomelib API
## Description:

**Related issue (if applicable):** fixes #321

**Pull request in [esphomedocs](https://github.com/OttoWinter/esphomedocs) with documentation (if applicable):** OttoWinter/esphomedocs#<esphomedocs PR number goes here>
**Pull request in [esphomeyaml](https://github.com/OttoWinter/esphomeyaml) with YAML changes (if applicable):** OttoWinter/esphomeyaml#<esphomeyaml PR number goes here>

## Checklist:
  - [x] The code change is tested and works locally.
  - [x] Check this box if you have read, understand, comply, and agree with the [Code of Conduct](https://g [... cut]
closed 2018-12-18 by OttoWinter