esphome/esphome-core #325: Color values are relative instead of absolute

WoLpH, 2018-12-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Not relevant for this issue, but I'm using Python 2.7.15 on FreeBSD 11.1

**ESP (ESP32/ESP8266/Board/Sonoff):**
[ESP8266 WeMos D1 mini Lite](https://docs.platformio.org/en/latest/boards/espressif8266/d1_mini_lite.html)

**Affected component:**

FastLED Clockless Light (but most likely, all RGB light components): https://esphomelib.com/esphomeyaml/components/light/fastled_clockless.html

**Description of problem:**

When setting the color values through MQTT it appears that all values are relative, that makes it harder to configure specific colors.

For example, the following commands all give the same results:
```
{"color": {"r": 0.1, "g": 0, "b": 0}}
{"color": {"r": 1, "g": 0, "b": 0}}
{"color": {"r": 255, "g": 0, "b": 0}}
{"color": {"r": 1023, "g": 0, "b": 0}}
```

I am aware of the brightness parameter but that would mean I would have to recalculate all colors to relative values before sending them. Is this intended or is this a bug?

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: D3
    num_leds: 50
```

OttoWinter, 2018-12-23:
Yes that's how Home Assistant expects it - and the MQTT platform sticks to the HA format.

Also it makes sense from the internal logic too. Color and brightness should not affect each other - color is only for specifying the tint of the light, and brightness should only affect the overall brightness. This is particularily useful with the HSV color wheel.

closed 2018-12-23 by OttoWinter

WoLpH, 2018-12-23:
It's odd though, I would expect this from HSV but not with RGB values. It's easy enough to recalculate the values using HSV though, that should work around it :)

Thanks for the explanation!

OttoWinter, 2018-12-23:
Yes the plan is to eventually move to HSV, as that will also enable better color transitions.

However that will require a bit more work (making sure HSV<->RGB conversion is lossless, making the calculation quick, etc)

WoLpH, 2018-12-23:
I'm not sure how much effect the truncating to 256 hsv values has in practice, but I've had great performance with the fastled hsv conversion. Upgrading 300 leds 60 times per second using hsv values with no issues at all.

The code inside fastled is rather optimized: https://github.com/FastLED/FastLED/blob/master/hsv2rgb.cpp

OttoWinter, 2018-12-27:
I know that code. But that's optimized for bytes - esphomelib uses `float`s for lights everywhere up until the last stage (writing to LEDs), as some LED controllers support more precision than just 0-255.

The other issue, as noted before, is that conversion needs to be lossless. So for example converting 255,0,0 to HSV and back should yield the same result.

WoLpH, 2018-12-27:
But you're using fastled to actually write the colors, right? In that case the results will still be truncated in the end as far as I'm aware. The performance is most likely a non-issue since the esp chips are rather fast and the colors are set per strip (so a single calculation).

Although I agree that the hsv to rgb conversion could be better, the 8 bit hsv values don't map fantastically to 3x8 bit rgb. Especially looking at the future, some led strips such as the ones equipped with TLS3001/TLC5940/TLC5947 actually support 36 bits colors so a more detailed hsv2rgb conversion would be useful. As far as I am aware, 16 bit per color support is planned for the future of fastled but I wouldn't get my hopes up for a quick solution. If you want greater color depth, it might be better to look for an alternative library for the time being. Although a temporary hack using dithering could do the trick: https://github.com/FastLED/FastLED/wiki/FastLED-Temporal-Dithering

Another important thing to keep in mind is which kind of conversion you're going to use. A spectrum or a rainbow map (as fastled calls it: https://github.com/FastLED/FastLED/wiki/FastLED-HSV-Colors#color-map-rainbow-vs-spectrum ). That greatly affects the way colors are displayed. I personally prefer the rainbow hue chart but it depends on the led strip as well.

WoLpH, 2018-12-27:
Ah, I see. I assumed you used fastled in all cases. That makes more sense :)