esphome/esphome-core #326: Capability to send UDP packets inside a fast loop to other devices, while keeping mqtt .

gnkarn, 2018-12-21:
**Is your feature request related to a problem/use-case? Please describe.**

not able to use UDP packets at the same time im using Mqtt .
fast mqtt packets are used in my case to send audio over UDP to other devices (slaves) at a fast rate while mqtt is used to receive commands from Has .
**Describe the solution you'd like**

enable UDP, and mqtt  write , and after that be able to write my own sketch to read analog input and send UDP packets to other ESPs .

**Additional context**

just that simple .

OttoWinter, 2018-12-23:
You're going to need to provide more context here.

You're using you're `not able to use UDP packets at the same time im using Mqtt`. It seems like you're using custom code from what I can gather. So in that case it could also be an issue with your custom code.

I know for a fact that UDP packets do work together with the MQTT client. The mDNS server uses UDP, as does the old ArduinoOTA code.

gnkarn, 2018-12-24:
the issue , is  mqtt and udp seams to conflict when fast udp packets are received , on slow , one at a time udp packet , there is no problem at all , not in your implementation, just in a general esp8266 , +mqtt+udp

so , i was asking , how to try it  with your library, ( i wasnt able to load a single sketch yet , as esphomelib , is not properly detecting the usb port on my mac connected to a nodemcu) , i red the docs, but no luck so far , back to mqtt and udp, how to add udp  in your environment , so i can receive audio streams ( for leds animations), as to be able to receive mqtt for commands. thanks

brandond, 2018-12-25:
I'm guessing whatever code you're running to handle the UDP stuff just isn't getting called frequently enough to avoid dropping packets out of the UDP buffer, when MQTT is enabled? Keep in mind that these things don't have a ton of power and are not really running a RTOS; the more you ask them to do the slower they're going to do it.

gnkarn, 2018-12-25:
thanks for your comment brandon, but im using UDP , just because of its characteristic of not being forced , to read all packets sent by the server , in my case as it is audio,  the missing packets are not important, just to pick the last one available to the client .

i still don't have clarity , on why this problem is generated , probably there are things i do not fully understand about UDP .

closed 2019-02-02 by OttoWinter