esphome/esphome-core #328: on_message MQTT Payload is invalid option

arnie580, 2018-12-21:
Using the on_message > payload option fails at build time:

`C:\Users\arnie>esphomeyaml conservatory.yaml run
[32mINFO Reading configuration...[0m
Failed config
 mqtt:
   - Invalid config for [mqtt]: [payload] is an invalid option for [mqtt]. Check: mqtt->on_message->0->payload. (See conservatory.yaml, line 10).
    - broker: 192.168.0.253
      on_message: ?
        - payload: 1
          then: ?
            - output.turn_on: cons_light_status
          topic: conservatory/light/status
      password:
      username:`

This is my yaml:

`mqtt:
  broker: '192.168.0.253'
  username: ''
  password: ''
  on_message:
    - topic: "conservatory/light/status"
      payload: '1'
      then:
        - output.turn_on: cons_light_status`

Version 1.9.0

closed 2018-12-21 by arnie580

HFsi, 2018-12-21:
Did you manage to use payloads someway?
I have the same problem

arnie580, 2018-12-21:
I didn't, I have opened a new issue as I was using esphomeyaml so created it on that repository instead.