esphome/esphome-core #331: dev branch: error: 'class esphomelib::WiFiComponent' has no member named 'set_sta'

DavidDeSloovere, 2018-12-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
hassio

**ESP (ESP32/ESP8266/Board/Sonoff):**
esp8266 wemos d1 mini

**Affected component:**
wifi

**Description of problem:**
Compile with **DEV** branch (`esphomelib_version: dev`) not working with wifi settings.

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.71
    gateway: 192.168.0.1
    subnet: 255.255.255.0
```

**Traceback (if applicable):**
```
Compiling .pioenvs/wemos_bandit_scrappy/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:14:19: error: 'class esphomelib::WiFiComponent' has no member named 'set_sta'
_wificomponent->set_sta(::WiFiAp{
^
src/main.cpp:14:27: error: '::WiFiAp' has not been declared
_wificomponent->set_sta(::WiFiAp{
^
*** [.pioenvs/wemos_bandit_scrappy/src/main.cpp.o] Error 1
```

**Additional information:**
Also tried commenting out `manual_ip` block, but that does't seem to matter.

emwap, 2018-12-24:
I think you need the `dev` branch of both esphomelib and esphomeyaml together.

Maybe reinstalling esphomeyaml triggered or updated something, but it solved the issue for me.

brandond, 2018-12-25:
Yeah, if you're using `esphomelib_version: dev` you should probably install esphomeyaml from git as well, otherwise odds are good that esphomelib is going to generate code that doesn't compile against the newer version of esphomelib.

```pip install --upgrade git+https://github.com/OttoWinter/esphomeyaml```

DavidDeSloovere, 2018-12-25:
Great. I'm using the `esphomeyaml-edge` hassio addon now instead of `esphomeyaml-beta` and it works.

I was mislead by the naming. Neither edge or beta of the addon align with dev from the lib. So I thought beta and dev would be a good match. It's not. Edge and dev are the correct ones.

Just figured out it's in the docs.
https://esphomelib.com/esphomeyaml/guides/faq.html?highlight=edge#how-do-i-use-the-latest-bleeding-edge-version

@OttoWinter maybe something to keep tabs on, if more people make the incorrect assumption, changing the description of the 3 different addon versions might help.

closed 2018-12-25 by DavidDeSloovere