esphome/esphome-core #340: Sonoff Basic + Dual + Slampher becomes non-responsive

meijerwynand, 2018-12-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Xubuntu 18.04
HA 0.84.3 
Mosquito MQTT

**ESP (ESP32/ESP8266/Board/Sonoff):**

* Sonoff Basic
* Sonoff Basic RF R2 POWER v1.0
* Sonoff Dual
* Sonoff Slampher

**Affected component:**
Appears to hang Sonoff devices on espyaml 1.9.3 

**Description of problem:**

- The Sonoff will flash correctly and connect to WiFi and MQTT and be visible in HA for a short time. 
- Roughly 3 minutes later the device will stop responding (unable to ping, does not refresh DHCP)
- ```status_led``` does indicate when attempting to connect MQTT and remains stable once connected. It remains in lit/on state if MQTT is disabled on the server
- Device does not register 'hostname' value when connecting to DHCP.
- Sonoff flashed devices (current 1.9.3) become unusable.
- OTA reflash is possible withing the 3 minute window of operation / before device hangs

**Problem-relevant YAML-configuration entries:**
```

esphomeyaml:
  name: !secret xxx
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
  # esphomelib_version: 1.9.1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  power_save_mode: light

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

# Enable logging
logger:
  level: DEBUG

ota:

## Quick GPIO reference
## https://esphomelib.com/esphomeyaml/devices/sonoff_basic.html
#
status_led:
  ## https://esphomelib.com/esphomeyaml/components/status_led.html
  pin: GPIO13

sensor:
  ## https://esphomelib.com/esphomeyaml/components/sensor/uptime.html
  - platform: uptime
    name: "esp_xxx_system_uptime"
    id: esp_xxx_system_uptime

  ## https://esphomelib.com/esphomeyaml/components/sensor/wifi_signal.html
  - platform: wifi_signal
    name: "esp_xxx_system_wifi_signal"
    id: esp_xxx_system_wifi_signal
    update_interval: 15s

## https://esphomelib.com/esphomeyaml/guides/automations.html
## switch - use build in API functions to manage local device pins without mqtt

light:
  # Status LED
  - platform: monochromatic
    name: "esp_xxx_gpio13"
    output: esp_xxx_gpio13

  ## OUT 1 - Green
  - platform: binary
    name: "esp_xxx_gpio12"
    id: esp_xxx_gpio12
    output: output_esp_xxx_gpio12

binary_sensor:
  ## https://esphomelib.com/esphomeyaml/components/binary_sensor/status.html
  - platform: status
    name: "esp_xxx_system_status"
    id: "esp_xxx_system_status"

## relays used as lights
## set as invertered to be on by fault
output:
  - platform: gpio
    id: "output_esp_xxx_gpio12"
    pin:
      number: GPIO12
      inverted: true

  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: esp_xxx_gpio13
    pin: GPIO13

```

**Additional information:**
- I initially started with esphomelib v1.8(ish), I only recently got time to revisit my pet project from v1.9.3
- I have multiple esp8266 devices (nodemcu/wemos/variety of sonoffs [touch (3gangUK/3gangUS), basic(old and new revisions)/dual/slampher])
- NodeMCU/Wemos seem to have no issue with current 1.9.x firmware, I frequently modify/reflash and fiddle with them, I redid my hydroponics pump and water level sensor (wemos d1 mini pro) this weekend and still running 100%
- The initial sonoff (~v1.8.2) flashing I did was added to HA and is still running fine to date.
- Recently I revisited my sonoff basics for a new project.
- I noticed the sonoff basic displaying symptoms as above
-- connects and appear to hang within 3 minutes
-- unable to connect to device / ping device 
-- hostname value was not passed during DHCP registration (I am running on Mikrotik (which does not support mDNS, hence I created a DHCP lease script which adds a static DNS entry once a esphomelib device connects to the specific IoT SSID/network))
- After experiencing the issue I suspected I might have an Sonoff Basic clone, I followed various sources I could find relating to the symptoms and tried multiple variations in my esphomeyml config but to no avail.
- I suspected a possible version issue so I decided not to reflash working sonoff basic devices on my HA deployment.
- Today I purchased 2 original sonoff devices (Sonoff Basic RF R2 POWER v1.0 and Sonoff Slampher) [Official importer in our country]
- After reflashing both, I experienced the same symptoms. leading me to suspect the issue is software related and not hardware/clone related.
- I decided to OTA flash a working/"production" Sonoff (Sonoff Dual) and it displayed exactly the same symptoms, one extra unresponsive device :(. Confirming the suspicion this issue appears to be software related.
- I then attempted to revert back to version 1.8.2, but only to be faced with new challenges 
-- ```esphomelib_version: 1.8.2``` gives error when trying to compile ```src/main.cpp:10:7: error: 'class esphomelib::Application' has no member named 'set_compilation_datetime'```
-- after manually editing the ```main.cpp``` file, and also trying combinations on ```DUSE_NEW_OTA``` and ```DUSE_OTA``` flags on ```platformio.ini``` with still the same end result and "hanging" of all my v1.9.3 sonoff devices.
- Running in ```logger.level: DEBUG``` I can see the unit connecting to MQTT, doing some updates and then becoming non responsive (even to pins)
- https://i.imgur.com/kT6LcyT.jpg

I currently have exhausted my personal resources and capacity in a best effort to troubleshoot and resolve the issue,  I am appealing to you for advice and suggestions.

Thank you very much, looking forward to your inputs.

OttoWinter, 2018-12-30:
> I then attempted to revert back to version 1.8.2, but only to be faced with new challenges
-- esphomelib_version: 1.8.2

The esphomeyaml and -lib versions need to match in order for compilation to work. So if you want to try with 1.8.2 you need to `pip install esphomeyaml==1.8.2` first

As for the other issues, there's nothing magic I can help you with here, sorry. I've seen semi-similar symptoms some times but then they vanished again for whatever reason. The only way to debug this kind of stuff is to get the USB logs. 

They _usually_ contain a lot more info regarding WiFi etc and of course also work when WiFi breaks. I know it's kind of hard to have a USB connection to these sonoff devices, but without logs I can't really help here.

meijerwynand, 2018-12-31:
Some feedback:

> versions need to match in order for compilation to work
- I was able to downgrade using ```pip install esphomeyaml==1.8.2``` and manually setting ```esphomeyam.esphomelib_version: 1.8.2```
- All sonoff devices are now stable with none of the reported issues (boot, alive for about 3 mins, hangs).
- I have noticed my ```time``` option for scheduling / cron events are not working.
- Some automation lambas do not work, I will crosscheck with the changelogs to see if features I used are compatible.

> get the USB logs
- Now that I have a base line and working system, I will gradually start upgrading (1.9.0 and increment). 
- Once I found the version causing the issues I will do ```VERY_VERBOSE``` log while connected to USB if that will help, and add the log here as a txt attachment. 
- Are there any other logs or information I should provide to assist in the trouble shooting?

meijerwynand, 2019-01-02:
More feedback:

### Attempting to isolate the issue
I used 1 Sonoff Basic which was known to hang/become non-responsive with 1.9.3 and working on 1.8.2.

### Environment
Xubuntu 18.04
HA 0.84.6 (pip)
esphomeyaml (pip)

### Symptoms
- Sonoff would boot, and become non-responsive within a ~3 minute period on v1.9.3
- Sonoff would boot and stay responsive on v1.8.2

### Procedure followed 

- [ ] Specify version
1. use ```pip install esphomeyaml==1.8.2``` to install the cli tool
2. specify the same version in the device_to_flash.yaml ```esphomeyaml.esphomelib_version: 1.8.2```

- [ ] Flash Sonoff 
1. the initial 1.8.2 flash was done via USB
2. all flashes after 1.8.2 was done via OTA

- [ ] Verify success
1. Doing trace via ```$ esphomelib some.yaml run``` and wait for ```system_uptime``` to stop in the trace ~3min. 
2. Wait for Sonoff to become non-responsive (~3mins)
3. If Sonoff is still responsive after ~10mins do 'hardware' test
4. Use programed button to toggle relay (Generally GPIO0) - i.e is device still functional on 'internal automations' 
5. Try next version of esphomelib

### Observations
1. I revered back to 1.8.2 and started the procedure as outlined above. I installed each version in the following order based on the [releases](https://github.com/OttoWinter/esphomelib/releases)
- 1.9.0
- 1.9.1
- 1.9.2
- 1.9.3

2. By installing and specifying each version in sequence I ended up with v1.9.3 currently running successfully on the [originally reported issue](#issue-394691450) 
3. Subsequently, after the tests appear successful I continued I OTA reflashed another 'problematic' Sonoffs from 1.9.3 (problematic) or 1.8.2 to 1.9.3 (version-by-version pip upgrade) and for the past 2 days they have been stable... MQTT is giving some challenges but I will address that at another stage
- Sonoff Dual - stable. 
- Sonoff Slampher - stable 
- Sonoff Basic (RF R2 V1) - stable
- Sonoff Touch's - Not flashed yet, status/outcome unknown

4. I have been able to even make modifications to the yaml
```
## Fish pond sonoff
# Starts with pump in off position
# Button toggles pump
# Cron'd
esphomeyaml:
  name: !secret xxx
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
  esphomelib_version: 1.9.3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

# Enable logging
logger:
  level: DEBUG

ota:

## Quick GPIO reference
## https://esphomelib.com/esphomeyaml/devices/sonoff_basic.html
status_led:
  ## https://esphomelib.com/esphomeyaml/components/status_led.html
  pin: GPIO13

sensor:
  ## https://esphomelib.com/esphomeyaml/components/sensor/uptime.html
  - platform: uptime
    name: "esp_xxx_system_uptime"
    id: esp_xxx_system_uptime

  ## https://esphomelib.com/esphomeyaml/components/sensor/wifi_signal.html
  - platform: wifi_signal
    name: "esp_xxx_system_wifi_signal"
    id: esp_xxx_system_wifi_signal
    update_interval: 15s

## https://esphomelib.com/esphomeyaml/guides/automations.html
light:
  # Status LED
  - platform: monochromatic
    name: "esp_xxx_gpio13"
    output: esp_xxx_gpio13

  ## OUT 1 - Green
  - platform: binary
    name: "esp_xxx_gpio12"
    id: esp_xxx_gpio12
    output: output_esp_xxx_gpio12

binary_sensor:
  ## https://esphomelib.com/esphomeyaml/components/binary_sensor/status.html
  - platform: status
    name: "esp_xxx_system_status"
    id: "esp_xxx_system_status"

  ## button - toggle relay
  - platform: gpio
    name: "esp_xxx_gpio0"
    pin:
      number: GPIO0
      inverted: true
    on_press:
      then:
        - light.toggle: esp_xxx_gpio12

## relays used as lights
output:
  - platform: gpio
    id: "output_esp_xxx_gpio12"
    pin:
      number: GPIO12

  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: esp_xxx_gpio13
    pin: GPIO13

## Use time
time:
  - platform: sntp
    id: sntp_time
    timezone: Africa/Johannesburg
    on_time:
    # Switch on on the hour
      - seconds: 0-5
        minutes: 0
        hours: '*'
        then:
          - light.turn_on: esp_xxx_gpio12
      # Switch off 10 minutes later
      - seconds: 0-5
        minutes: 10
        hours: '*'
        then:
          - light.turn_off: esp_xxx_gpio12
```

### Conclusion
* I cannot definitely state the reason for the fix, whether it be pip (possible upgrading sequence of versions) related or what anomaly caused (still is causing) this issue. 
* I can confirm, the methods used above in trying to resolve the matter has to date been delivering successful results for my environment. 
* There are still untested devices (Sonoff Touch's) I have to flash and confirm stability.
* Other concerns I did take note of does not appear to be part or have impact on this matter and will be addressed separately and appropriately.

_For now, this looks like a win... yay!_

OttoWinter, 2019-01-02:
Well it may be a win, but it still doesn't really solve the problem. If you give me the USB logs there's much more I can do. Oh and also there have been a couple of changes in the most recent dev version for wifi, maybe they could fix this? See https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version

meijerwynand, 2019-01-02:
> Well it may be a win, but it still doesn't really solve the problem

What is the possibility that is issue may be in the pip deployment or do all pip files get replace irrespective of the version upgrade e.g. (0.0.1 -> 0.0.2) or  (0.0.1 -> 0.1.3)

> If you give me the USB logs there's much more I can do.

Sure, but to what extent will be helpful for you? (DEBUG|VERY_VERBOSE)

I still have some Sonoff Touch's I need to reflash ( well first remove, resolder and then reflash). However, I will update them with current configuration and pips, if they work, I will revert back to 1.8.2 and upgrade to 1.9.3 direct and see if that replicates the issue. 

I will close this issue now and once I redo the Touch's I will update.

> most recent dev version for wifi, maybe they could fix this

Once I am happy and better understand the (reoccurring) issues on all devices I will do a bulk update and use ```dev```. Running various versions on various devices just creates various headaches while still learning.

Thanks

closed 2019-01-02 by meijerwynand