esphome/esphome-core #343: Festnight Sonoff Smart Switch Wifi (Sonoff Basic clone)

Styx85, 2019-01-01:
**What new integration would you wish to have?**
I don't know if new integration is the right thing, but I'd like to report the experience.
If I get some guidance I can also write a short manual for the module.

I recently bought a Sonoff Basic Smart Switch clone ([Amazon](https://www.amazon.de/gp/product/B07KHC1BTS/ref=oh_aui_detailpage_o09_s00?ie=UTF8&psc=1)).
It differs from the original Sonoff that it is Tuya/Smart Life based.
Also flashing the board is a bit different.

**If possible, provide a link to an existing library for the integration:**
All things are set in esphomelib.

**Please describe your use case for this integration and alternatives you've tried:**
I just bought this aforementioned clone.
There are sone Problems you have to think about.

**Additional context:**
There is a different PIN-out:
First of all the Relay is in GPIO4, the Status LED is the usual GPIO13.
I couldn't find out which GPIO the button is.

I used following template to get it up and running. at least the relay is working:

```
## https://esphomelib.com/esphomeyaml/guides/automations.html
## switch - use build in API functions to manage local device pins without mqtt

light:
  # Status LED (red)
  - platform: monochromatic
    name: "esp_Bad_gpio13"
    output: esp_Bad_gpio13

  ## OUT 1 - Blue
  - platform: binary
    name: "esp_Bad_gpio04"
    id: esp_Bad_gpio04
    output: output_esp_Bad_gpio04

binary_sensor:
  ## https://esphomelib.com/esphomeyaml/components/binary_sensor/status.html
  - platform: status
    name: "esp_Bad_system_status"
    id: "esp_Bad_system_status"

## relays used as lights
## set as invertered to be on by default
output:
  - platform: gpio
    id: "output_esp_Bad_gpio04"
    pin:
      number: GPIO04
#      inverted: true
```

Than version 1.9.0 onwards is not working, the clone becomes unresponsive. That maybe is related to Issue #340. Version 1.8.2 is working flawlessly.

Another thing I tested was the Sonoff basic instruction from the manual, but with that configuration OTA updates crashes reproducible. Maybe it is not that good an idea to have GPIO4 as INPUT_PULLUP.

> ## configuration crashes OTA
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: GPIO4
>       mode: INPUT_PULLUP
>       inverted: True
>     name: "Sonoff Basic Button"
>   - platform: status
>     name: "esp_Bad_system_status"
>     id: "esp_Bad_system_status"
> 
> switch:
>   - platform: gpio
>     name: "Sonoff Basic Relay"
>     pin: GPIO4
> 
> output:
>   - platform: esp8266_pwm
>     id: basic_blue_led
>     pin:
>       number: GPIO13
>       inverted: True
> 
> light:
>   - platform: monochromatic
>     name: "Sonoff Basic Blue LED"
>     output: basic_blue_led

meijerwynand, 2019-01-02:
Greetings, 

I was able to get my Sonoff devices stable by installing and defining each versions and then update them version by version [see update on issue](https://github.com/OttoWinter/esphomelib/issues/340#issuecomment-450809439).

If possible, use the method describe here and check if that improves / changes the stability for you devices.

Styx85, 2019-01-02:
And some pictures for the ones interested.

[image]
[image]
[image]
[image]
[image]

Styx85, 2019-01-02:
> Greetings,
>
> I was able to get my Sonoff devices stable by installing and defining each versions and then update them version by version [see update on [...]

Thanks for the info @meijerwynand , I switched to bleeding edge for my sonoff clone and that one is running flawlessly so far but breakung OTA for my Wemos D1 mini.

BTW, have a look at my modified yaml. I took yours for inspiration and modified the "button" part.

meijerwynand, 2019-01-03:
@Styx85 

>  I switched to bleeding edge for my sonoff clone and that one is running flawlessly so far but breakung OTA for my Wemos D1 mini.

I am running a Wemos D1 Mini Pro on v1.9.3 without any issue. However after copy/paste from other configs I noticed I flashed the Wemos possibly incorrect (but it's functional and stable for almost a week now). That includes a functional OTA.

The "invalid" but functional Wemos config - I have the board as ```nodemcuv2```
```
esphomeyaml:
  name: !secret 5ccf7f2c0f45
  platform: ESP8266
  board: nodemcuv2
  esphomelib_version: 1.9.3
```

@brandond 

> What the heck do you have going on with the relay configured as a light!?

Thank you so much for the config you posted. I will surely be able to better my configs and make a more appropriate use of the ```template``` options/features.

The initial reasoning for using the relay as a ```light``` was:

1. There is a lamp connect to the sonof's relay
2. The HA UI looks and works better with the ```light``` component 
3. I am still learning all the in's and out's

Your submission of configs as above really helps out, it is much appreciated.

OttoWinter, 2019-02-02:
Closing because this doesn't seem to be an issue, just documentation on how to set up a device.

closed 2019-02-02 by OttoWinter