esphome/esphome-core #353: OTA for wemos D1 mini is broken in latest dev version

Styx85, 2019-01-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
```
HA on Arch Linux via pip/systemd service
HA v0.84.6
```

**ESP (ESP32/ESP8266/Board/Sonoff):**
```
Wemos D1 mini
```

**Affected component:**
```
binary sensor / PIR
```

**Description of problem:**
OTA is working correctly with esphomelib v1.9.3 but broken with lates dev version

> INFO Successfully compiled program.
> INFO Resolving IP address of pir1.local
> ERROR Error resolving IP address of pir1.local. Is it connected to WiFi?
> ERROR (If this error persists, please set a static IP address: https://esphomelib.com/esphomeyaml/components/wifi.html#manual-ips)
> ERROR Error resolving IP address: [Errno -2] Name or service not known
> WARNING OTA v2 method failed. Trying with legacy OTA...
> INFO Starting on 0.0.0.0:48397
> INFO Upload size: 351968
> INFO Sending invitation to pir1.local 
> ERROR Failed
> ERROR Host pir1.local Not Found

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: pir1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: '!ssid
  password: '!pwd_wifi'

mqtt:
  broker: '!broker'
  username: '!user_mqtt'
  password: '!pwd_mqtt'

# Enable logging
logger:

ota:
  password: '!pwd'
binary_sensor:
  - platform: gpio
    pin: 
      number: D6
    name: "PIR Sensor"
    device_class: motion
  ## https://esphomelib.com/esphomeyaml/components/binary_sensor/status.html
  - platform: status
    name: "esp_xxx_pir_system_status"
    id: "esp_xxx_pir_system_status"

sensor:
  ## https://esphomelib.com/esphomeyaml/components/sensor/uptime.html
  - platform: uptime
    name: "esp_xxx_pir_system_uptime"
    id: esp_xxx_pir_system_uptime

  ## https://esphomelib.com/esphomeyaml/components/sensor/wifi_signal.html
  - platform: wifi_signal
    name: "esp_xxx_pir_system_wifi_signal"
    id: esp_xxx_pir_system_wifi_signal
    update_interval: 15s
```

**Traceback (if applicable):**

```

```

**Additional information:**

Styx85, 2019-01-02:
@OttoWinter Maybe this error message is helping:
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Updating esphomelib copy from git (pir1/.piolibdeps/esphomelib)
Bereits aktuell.
INFO Compiling app...
INFO Running:  platformio run -d pir1
Processing pir1 (platform: espressif8266@>=1.8.0; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.10.0-dev #866b198
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0
|   |   |-- <EspSoftwareSerial> 3.4.1
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/pir1/src/main.cpp.o
Compiling .pioenvs/pir1/libfa1/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs/pir1/libfa1/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling .pioenvs/pir1/libfa1/ESP8266WiFi/WiFiUdp.cpp.o
Compiling .pioenvs/pir1/libaf5/ESP8266mDNS/ESP8266mDNS.cpp.o
Compiling .pioenvs/pir1/libe0b/ArduinoOTA/ArduinoOTA.cpp.o
Compiling .pioenvs/pir1/lib61e/Hash/Hash.cpp.o
Compiling .pioenvs/pir1/lib61e/Hash/sha1/sha1.c.o
Compiling .pioenvs/pir1/lib10c/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
Compiling .pioenvs/pir1/lib10c/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o
Archiving .pioenvs/pir1/lib61e/libHash.a
Compiling .pioenvs/pir1/lib10c/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
Indexing .pioenvs/pir1/lib61e/libHash.a
src/main.cpp: In function 'void setup()':
src/main.cpp:16:19: error: 'class esphomelib::WiFiComponent' has no member named 'set_sta'
_wificomponent->set_sta(::WiFiAp{
^
src/main.cpp:16:27: error: '::WiFiAp' has not been declared
_wificomponent->set_sta(::WiFiAp{
^
Compiling .pioenvs/pir1/lib10c/ESPAsyncTCP_ID305/SyncClient.cpp.o
*** [.pioenvs/pir1/src/main.cpp.o] Error 1
===================================== [ERROR] Took 3.44 seconds =====================================
```
I encountered it when having version 1.9.3 installed but modified the yaml to dev-version.
```
pip2 install --user esphomeyaml==1.9.3
```
```
esphomeyaml:
  name: pir1
  platform: ESP8266
  board: d1_mini
  esphomelib_version: dev
```

brandond, 2019-01-03:
Are you sure the device is currently on the network? If mDNS usually works in your environment, the most likely cause for the error is what it says in the logs:

```
ERROR Error resolving IP address of pir1.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphomelib.com/esphomeyaml/components/wifi.html#manual-ips)
ERROR Error resolving IP address: [Errno -2] Name or service not known
```

The compilation failure is a known issue caused by using different versions of esphomelib and esphomeyaml, and should not be related to your mDNS name resolution failure.

Styx85, 2019-01-03:
@brandond I just rechecked with latest git

with 
```
pip2 install --user esphomeyaml==1.9.3
```
```
[caberliner@home config]$ python2 -m esphomeyaml pir1kueche.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d pir1
Processing pir1 (platform: espressif8266@>=1.8.0; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.9.3 #9d5e290
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0
|   |   |-- <EspSoftwareSerial> 3.4.1
|   |-- <Wire> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/pir1/src/main.cpp.o
Linking .pioenvs/pir1/firmware.elf
Retrieving maximum program size .pioenvs/pir1/firmware.elf
Building .pioenvs/pir1/firmware.bin
Checking size .pioenvs/pir1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  45.2% (used 37004 bytes from 81920 bytes)
PROGRAM: [===       ]  32.7% (used 341132 bytes from 1044464 bytes)
================================================= [SUCCESS] Took 4.77 seconds =================================================
INFO Successfully compiled program.
INFO Resolving IP Address of pir1.local
INFO  -> 192.168.178.29
INFO Uploading pir1/.pioenvs/pir1/firmware.bin (345280 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from pir1/debug
[18:55:23][I][ota:364]: OTA update finished!
[18:55:23][I][helpers:258]: Rebooting safely... Reason: 'ota'
[18:55:32][I][mqtt.client:115]: MQTT Connected!
[18:55:32][I][application:089]: setup() finished successfully!
[18:55:32][I][application:094]: You're running esphomelib v1.9.3 compiled on Jan  3 2019, 18:55:14
```

and with 
```
pip2 install --user git+git://github.com/OttoWinter/esphomeyaml.git
```
```
[caberliner@home config]$ python2 -m esphomeyaml pir1kueche.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Updating esphomelib copy from git (pir1/.piolibdeps/esphomelib)
Bereits aktuell.
INFO Compiling app...
INFO Running:  platformio run -d pir1
Processing pir1 (platform: espressif8266@1.8.0; framework: arduino; board: d1_mini)
---------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.10.0-dev #535f602
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0
|   |   |-- <EspSoftwareSerial> 3.4.1
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
Retrieving maximum program size .pioenvs/pir1/firmware.elf
Checking size .pioenvs/pir1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  48.5% (used 39756 bytes from 81920 bytes)
PROGRAM: [===       ]  33.4% (used 348952 bytes from 1044464 bytes)
========================================= [SUCCESS] Took 4.29 seconds =========================================
INFO Successfully compiled program.
INFO Resolving IP address of pir1.local
ERROR Error resolving IP address of pir1.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphomelib.com/esphomeyaml/components/wifi.html#manual-ips)
ERROR Error resolving IP address: [Errno -2] Name or service not known
WARNING OTA v2 method failed. Trying with legacy OTA...
INFO Starting on 0.0.0.0:38446
INFO Upload size: 353104
INFO Sending invitation to pir1.local 
ERROR Failed
ERROR Host pir1.local Not Found
```

mvturnho, 2019-01-03:
I had the same problem. I added my domain name to the wifi part in the yaml file. Then the upload worked but I than ran into another error that I am still investigating.

Styx85, 2019-01-03:
This is the DEBUG output for latest git for the wifi part:
```
DEBUG Executing task for domain=wifi
DEBUG Adding: WiFiComponent *wificomponent = App.init_wifi()
DEBUG Registered variable wificomponent of type WiFiComponent
DEBUG Adding: WiFiAP wifiap = WiFiAP()
DEBUG Registered variable wifiap of type WiFiAP
DEBUG Adding: wifiap.set_ssid("!ssid")
DEBUG Adding: wifiap.set_password("!pwd")
DEBUG Adding: wificomponent->add_sta(wifiap)
DEBUG  -> wifi finished
DEBUG Executing task for domain=ota
DEBUG Adding: OTAComponent *otacomponent = App.init_ota()
DEBUG Registered variable otacomponent of type OTAComponent
DEBUG Adding: otacomponent->set_auth_password("!pwd2")
DEBUG Adding: otacomponent->start_safe_mode()
DEBUG  -> ota finished
```

And this for 1.9.3
```
DEBUG Executing task for domain=wifi
DEBUG Adding: ::WiFiComponent *_wificomponent = App.init_wifi();
DEBUG Registered variable _wificomponent of type ::WiFiComponent
DEBUG Adding: _wificomponent->set_sta(::WiFiAp{
    .ssid = "!ssid",
    .password = "!pwd",
    .channel = -1,
});
DEBUG  -> wifi finished
DEBUG Executing task for domain=ota
DEBUG Adding: ::OTAComponent *_otacomponent = App.init_ota();
DEBUG Registered variable _otacomponent of type ::OTAComponent
DEBUG Adding: _otacomponent->set_auth_password("!pwd2");
DEBUG Adding: _otacomponent->start_safe_mode();
DEBUG  -> ota finished
```

brandond, 2019-01-03:
It looks like the problem is in mDNS, although you don't help things by pinging the IP instead of the .local hostname to prove that it's up.

If you assign the device an static IP address in the YAML, you can bypass the DNS lookup, just to see if that changes your results any.

Styx85, 2019-01-03:
It is indeed an IP resolving problem when no static IP is set.

here's the DEBUG output from a minute ago:

This is the DEBUG output for latest git for the wifi part:
```
DEBUG Executing task for domain=wifi
DEBUG Adding: WiFiComponent *wificomponent = App.init_wifi()
DEBUG Registered variable wificomponent of type WiFiComponent
DEBUG Adding: WiFiAP wifiap = WiFiAP()
DEBUG Registered variable wifiap of type WiFiAP
DEBUG Adding: wifiap.set_ssid("!ssid")
DEBUG Adding: wifiap.set_password("!pwd")
DEBUG Adding: wifiap.set_manual_ip(ManualIP{
    .static_ip = IPAddress(192, 168, 178, 29),
    .gateway = IPAddress(192, 168, 178, 1),
    .subnet = IPAddress(255, 255, 255, 0),
    .dns1 = IPAddress(0, 0, 0, 0),
    .dns2 = IPAddress(0, 0, 0, 0),
})
DEBUG Adding: wificomponent->add_sta(wifiap)
DEBUG  -> wifi finished
```

And this for 1.9.3
```
DEBUG Executing task for domain=wifi
DEBUG Adding: ::WiFiComponent *_wificomponent = App.init_wifi();
DEBUG Registered variable _wificomponent of type ::WiFiComponent
DEBUG Adding: _wificomponent->set_sta(::WiFiAp{
    .ssid = "!ssid",
    .password = "!pwd",
    .channel = -1,
});
DEBUG  -> wifi finished
```

both are uploading correctly
git:
```
================================================================================================== [SUCCESS] Took 15.16 seconds ==================================================================================================
INFO Successfully compiled program.
INFO Connecting to 192.168.178.29
INFO Uploading pir1/.pioenvs/pir1/firmware.bin (353312 bytes)
DEBUG MD5 of binary is 6383f1e6aa2a791fe0e59fd349492221
DEBUG Auth: Nonce is 10428cd16c3b634ba390dea7b2538bad
DEBUG Auth: CNonce is 0739229c3ea25268ead35deb939d30cd
DEBUG Auth: Result is 4dbe05df1c75f21426cc4592fb6eb3ef
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from pir1/debug
[19:32:43][I][mqtt.client:209]: MQTT Connected!
[19:32:44][I][application:089]: setup() finished successfully!
```
v1.9.3
```
================================================================================================== [SUCCESS] Took 4.84 seconds ==================================================================================================
INFO Successfully compiled program.
INFO Connecting to 192.168.178.29
INFO Uploading pir1/.pioenvs/pir1/firmware.bin (345376 bytes)
DEBUG MD5 of binary is 11b84509ec4d609546cffbfdb656e8da
DEBUG Auth: Nonce is 93c6694b7572f80b74c88975bbdee61a
DEBUG Auth: CNonce is 3cba9a12fa5d6f0cdf377196ff31d1a4
DEBUG Auth: Result is 3153faf108880ea4b392cb31967722ac
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from pir1/debug
[19:30:53][I][helpers:258]: Rebooting safely... Reason: 'ota'
[19:31:02][I][mqtt.client:115]: MQTT Connected!
[19:31:02][I][application:089]: setup() finished successfully!
[19:31:02][I][application:094]: You're running esphomelib v1.9.3 compiled on Jan  3 2019, 19:30:43
```

without manual_IP it is the same result as before. git is failing(not resolving the ip), v1.9.3 is working

Styx85, 2019-01-03:
The heck. Solved it.
I don't know why but with latest git you have to specify the domain in wifi part. Then it is working flawlessly. In v1.9.3 it is working without specifing the domain.

This is the DEBUG output for latest git for the wifi part:
```
DEBUG Executing task for domain=wifi
DEBUG Adding: WiFiComponent *wificomponent = App.init_wifi()
DEBUG Registered variable wificomponent of type WiFiComponent
DEBUG Adding: WiFiAP wifiap = WiFiAP()
DEBUG Registered variable wifiap of type WiFiAP
DEBUG Adding: wifiap.set_ssid("!ssid")
DEBUG Adding: wifiap.set_password("!pwd")
DEBUG Adding: wificomponent->add_sta(wifiap)
DEBUG  -> wifi finished
```
this is the yaml wifi part
```
wifi:
  ssid: '!ssid'
  password: '!pwd'
  domain: .fritz.box
#  manual_ip:
    # Set this to the IP of the ESP
#    static_ip: 192.168.178.29
    # Set this to the IP address of the router. Often ends with .1
#    gateway: 192.168.178.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
#    subnet: 255.255.255.0  
```

OttoWinter, 2019-01-03:
I don't think that's on the esphomeyaml side. I just checked and there has been no change whatsoever for resolving IP addresses. It's probably some issue with mDNS on your side. `.fritz.box` makes the operating system use the local DNS server (of your fritz box) for IP resolving, instead of the mDNS one when using `.local`.

Styx85, 2019-01-03:
I think this explanation is only valid if there was a change on the router side. The only difference here was compile time (aprox. 2 minutes) and the esphomexaml version.

brandond, 2019-01-04:
Whatever it is, it's not the wifi part, it's mDNS... which is (I believe) handled internally by the Arduino OTA code; there's no way to turn it on or off directly from esphomeyaml, but disabling OTA also disables mDNS. Specifying a static IP or domain in the config simply makes the 'update' command use the specified IP or domain instead of using mDNS's '.local' domain.

On ubuntu at least, you can install the 'avahi-utils' package and test resolution with `avahi-resolve -v -n pir1.local`

OttoWinter, 2019-01-04:
@brandond The mDNS part is not handled by Arduino, it's handled by the OS ([here](https://github.com/OttoWinter/esphomeyaml/blob/dev/esphomeyaml/helpers.py#L99)) - this is all done internally in CPython with a [`getaddrinfo`](http://man7.org/linux/man-pages/man3/getaddrinfo.3.html) syscall.

OttoWinter, 2019-01-04:
@brandond 
> which adds several more records than you do here.

Yes, but since we don't use the `ArduinoOTA` anymore (and that's being removed soon), those records don't do anything anymore.

> but the base SRV record is created even if OTA is not enabled

Yes because that enables users to type `livingroom.local` with the new native API too. 

The new way is kind of like this: First of all, always set some mDNS record - this enables the use of `.local` domains.  Then, depending on which components are used set one of the following mDNS types:

 * `_esphomelib` - for HA automatic native API discovery
 * `arduino` otherwise - it could also just be `foobar` as far as I'm concerned. Since we're not using AruinoOTA anymore, what we choose here doesn't matter.

> So I guess I'm not surprised that the client that @Styx85 is using declines to resolve the hosts via mDNS

Yes, but this line here in the logs give it away:

```
WARNING OTA v2 method failed. Trying with legacy OTA...
```

since dev, that's only printed if there has not been a successful OTAv2 upload in the dev version. So the `dev` code with these mDNS changes was not running on the device.

> Every DNS-SD service MUST have a TXT record in addition to its SRV record

Ok that's interesting 🤔

brandond, 2019-01-04:
I've been dumping mDNS traffic; it looks like the esp8266 mDNS library will add an empty TXT record if none are present, which should comply with the RFC. That said, avahi-browse isn't showing any of my esp8266 nodes that are on dev, but it is showing the esp32 node that's on 1.9.2. I'll see if I can collect more info.

OttoWinter, 2019-02-02:
Closing for https://github.com/esphome/ESPHome/pull/386

closed 2019-02-02 by OttoWinter