esphome/esphome-core #358: PMSx003 calculation

jara1, 2019-01-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
Hi, I would like to report bug in calculation of PMSx003 total size.
pmsx003.cpp, line 81, should be: uint8_t total_size = 4 + payload_length;

Tested with PMS5003 and works correctly.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2019-01-08:
Are you sure that is the actual issue here? There's been a fix recently (in dev, not in 1.9.3) that fixed some of the packet sizes - and for the person reporting the issue that fixed it.

Also line 81 should be correct judging from the datasheet.

jara1, 2019-01-08:
Hi, I've used version 1.9.3. According to datasheet for PMS5003, frame length (payload length) is 2x13+2 = 28bytes. This matches line 62 in pmsx003.cpp, which is correct. There are another 2bytes for start characters and another 2 bytes for frame length indication = 4bytes. So the total size of the message is 28 + 4 = 32bytes.

[image]