esphome/esphome-core #360: ESPNOW as a transport medium instead of Wifi?

xlfe, 2019-01-07:
**Is your feature request related to a problem/use-case? Please describe.**
I'm interested in seeing whether it would be possible to use ESPNOW instead of a normal Wifi connection. This would also remove MQTT from the picture - you would presumably need a ESP device dedicated as a HUB connected to HA via USB which acted as the HUB to the rest of your ESP devices over ESPNOW. 

**Describe the solution you'd like**
Ability to use esphomelib without wifi and mqtt - to enable low latency control of ESP devices from HASS.

**Additional context**
MQTT is slow (or at least slow when you consider a point to point message is <50ms). Wifi is unreliable and requires more power usage for ESP devices than ESPNOW. I have been working on this concept for a little while using https://github.com/xlfe/PJON-cython as a transport mechanism https://github.com/xlfe/reticul8 - but I was targeting Arduino and ESP devices. 

I think it would require:-
* A new platform component in HASS (to connect over serial/USB to the ESP HUB)
* Changes to esphomelib to change the transport mechanism from MQTT/WIfi to something (nanopb?)/ESPNOW

Interested in your thoughts on whether it makes sense to figure out ESPNOW support (and therefore esphomelib integration with HASS without the use of MQTT and Wifi).

OttoWinter, 2019-01-07:
https://beta.esphomelib.com/esphomeyaml/components/api.html + https://rc--home-assistant-docs.netlify.com/blog/2019/01/04/release-85/ :)

You got quite close to how I actually just did it ;) also using protobuf and a custom protocol. But it's just using TCP sockets over WiFi - that's primarily because everything else (MQTT, OTA) should still work side-by-side

The goal is to one day move to ESP-Mesh with that custom protocol. But also not in the near future because I think the new native API is already quite good in a few key metrics - latency is in a similar time range you posted (though I don't recommend writing high perfomance applications with esphomelib anyway), and message sizes are quite small.

Power consumption+Range are the only factors speaking for espnow/mesh for me. For the former: we're talking about devices that use 1W max here (except for deep sleep uses of course)

xlfe, 2019-01-07:
LOL! Excellent. I'll checkout what you have done. Thanks

closed 2019-01-07 by xlfe