esphome/esphome-core #361: Unable compile

itProfi, 2019-01-08:
When i try, i have errors:
C:\Users\User\Documents\PlatformIO\Projects\eee>esphomeyaml livingroom.yaml run
←[32mINFO Reading configuration...←[0m
←[32mINFO Generating C++ source...←[0m
←[32mINFO Compiling app...←[0m
←[32mINFO Running:  platformio run -d ble2mqtt←[0m
Processing ble2mqtt (platform: espressif32@>=1.4.0; board: wemosbat; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemosbat.html
PLATFORM: Espressif 32 > WeMos WiFi & Bluetooth Battery
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <esphomelib> 1.9.3 #9d5e290
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <AsyncTCP> 1.0.1
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <AsyncTCP> 1.0.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0
|   |-- <Wire> 1.0
|   |-- <Preferences> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |-- <Update> 1.0
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\FastLED.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\bitswap.cpp.o
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\colorpalettes.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\colorutils.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\hsv2rgb.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\lib8tion.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\noise.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\FastLED.cpp.o] Error 1
Compiling .pioenvs\ble2mqtt\lib6e1\FastLED_ID126\power_mgt.cpp.o*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\bitswap.cpp.o] Error 1

*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\colorpalettes.cpp.o] Error 1
*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\colorutils.cpp.o] Error 1
*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\hsv2rgb.cpp.o] Error 1
*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\lib8tion.cpp.o] Error 1
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\noise.cpp.o] Error 1
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
"Bluetooth" не является внутренней или внешней
командой, исполняемой программой или пакетным файлом.
*** [.pioenvs\ble2mqtt\lib6e1\FastLED_ID126\power_mgt.cpp.o] Error 1
=============================================================== [ERROR] Took 5.59 seconds ===============================================================

OttoWinter, 2019-01-09:
Is this related? https://github.com/platformio/platformio-vscode-ide/issues/368

itProfi, 2019-01-09:
Yes, it's work!

closed 2019-01-09 by itProfi