esphome/esphome-core #367: fastled_clockless effects probably broken in v1.10.0

DavidDeSloovere, 2019-01-10:
Firstly, thanks Otto for the hard work on v1.10. Feels like it could have been a v2. Make we wonder how much you need to change before you call something a new major version ;-)

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio 0.85 and esphomeyaml addon 1.10

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266 Wemos D1 mini

**Affected component:**
light - fastled_clockless

**Description of problem:**
From the moment I add my fastled_clockless config (with effects), build warnings shows up. The build completes and upload works. But connecting to the node (native API) does not seem to work however. Like it doesn't boot correctly, although I can ping it.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: wemos_nasty_eagle
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.70
    gateway: 192.168.0.1
    subnet: 255.255.255.0

ota:
  password: !secret ota_password

api:

# logger:
#   level: VERBOSE

status_led:
  pin:
    number: D4

web_server:
  port: 80

binary_sensor:
  - platform: status
    name: "nasty_eagle status"

# custom stuff here
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: D5
    num_leds: 150
    rgb_order: GRB
    name: "nasty_eagle one"
    effects:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
```

**Traceback (if applicable):**
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphomeyaml/wemos_nasty_eagle
Processing wemos_nasty_eagle (platform: espressif8266@1.8.0; framework: arduino; board: d1_mini)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <FastLED> 3.2.0
|   |-- <EspSoftwareSerial> 3.4.1
|-- <Hash> 1.0
|-- <esphomelib> 1.10.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESP Async WebServer> 1.1.1
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <FastLED> 3.2.0
|   |   |-- <EspSoftwareSerial> 3.4.1
|   |-- <NeoPixelBus> 2.4.1
|   |   |-- <SPI> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/wemos_nasty_eagle/src/main.cpp.o
In file included from .piolibdeps/esphomelib/src/esphomelib/light/addressable_light_effect.h:9:0,
from .piolibdeps/esphomelib/src/esphomelib/application.h:52,
from src/main.cpp:3:
.piolibdeps/esphomelib/src/esphomelib/light/addressable_light.h: In constructor 'esphomelib::light::ESPColor::ESPColor(uint32_t)':
.piolibdeps/esphomelib/src/esphomelib/light/addressable_light.h:32:17: warning: 'esphomelib::light::ESPColor::<anonymous union>::<anonymous struct>::<anonymous union>::w' will be initialized after [-Wreorder]
uint8_t w;
^
.piolibdeps/esphomelib/src/esphomelib/light/addressable_light.h:20:17: warning:   'uint8_t esphomelib::light::ESPColor::<anonymous union>::<anonymous struct>::<anonymous union>::r' [-Wreorder]
uint8_t r;
^
In file included from .piolibdeps/esphomelib/src/esphomelib/light/addressable_light.h:154:0,
from .piolibdeps/esphomelib/src/esphomelib/light/addressable_light_effect.h:9,
from .piolibdeps/esphomelib/src/esphomelib/application.h:52,
from src/main.cpp:3:
.piolibdeps/esphomelib/src/esphomelib/light/addressable_light.tcc:27:1: warning:   when initialized here [-Wreorder]
ESPColor::ESPColor(uint32_t colorcode)
^
Linking .pioenvs/wemos_nasty_eagle/firmware.elf
Building .pioenvs/wemos_nasty_eagle/firmware.bin
Retrieving maximum program size .pioenvs/wemos_nasty_eagle/firmware.elf
Checking size .pioenvs/wemos_nasty_eagle/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  54.6% (used 44764 bytes from 81920 bytes)
PROGRAM: [====      ]  39.2% (used 409712 bytes from 1044464 bytes)
========================= [SUCCESS] Took 27.88 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.70
INFO Uploading /config/esphomeyaml/wemos_nasty_eagle/.pioenvs/wemos_nasty_eagle/firmware.bin (413856 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.70 using esphomelib API
INFO Connecting to 192.168.0.70:6053 (192.168.0.70)
WARNING Couldn't connect to API (Error connecting to 192.168.0.70: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.70:6053 (192.168.0.70)
WARNING Couldn't connect to API (Error connecting to 192.168.0.70: [Errno 111] Connection refused). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.0.70:6053 (192.168.0.70)
WARNING Couldn't connect to API (Error connecting to 192.168.0.70: [Errno 111] Connection refused). Trying to reconnect in 4 seconds
INFO Connecting to 192.168.0.70:6053 (192.168.0.70)
WARNING Couldn't connect to API (Error connecting to 192.168.0.70: [Errno 111] Connection refused). Trying to reconnect in 8 seconds
INFO Connecting to 192.168.0.70:6053 (192.168.0.70)
WARNING Couldn't connect to API (Error connecting to 192.168.0.70: [Errno 111] Connection refused). Trying to reconnect in 16 seconds
INFO Connecting to 192.168.0.70:6053 (192.168.0.70)
WARNING Couldn't connect to API (Error connecting to 192.168.0.70: [Errno 111] Connection refused). Trying to reconnect in 32 seconds
```

**Additional information:**
[image]

OttoWinter, 2019-01-10:
I think the ESP might be in a boot loop, that would at least explain the behavior. (after a couple of bad boots the ESP goes into OTA safe mode, which just enables OTA+WiFi for upload)

I recently did a big refactor of all addressable lights. It's very likely linked to that. I don't have my neopixel lights with me here, so it will take a few days to debug this. If you can, USB logs would really help (preferably taken through esphomeyaml logs command, as that will auto-decode stacktraces)

DavidDeSloovere, 2019-01-10:
To be sure, the logs you are talking about are the ones from the node, at boot? And I need to connect it via USB to my RPI, hope esphomeyaml discovers it, select USB instead of OTA and hit the Logs button on the dashboard?
(I'm having trouble with node being discovered on USB atm)

Also I think the build warnings also appear if I don't specify effects. I was sure I tried it without the effects and didn't get the warning once, now I am getting the warning with and without effects.
But without effects the boot seems ok; with the effects it's stuck...

OttoWinter, 2019-01-10:
> To be sure, the logs you are talking about are the ones from the node, at boot? 

Yes, exactly

> Also I think the build warnings also appear if I don't specify effects. 

The build warnings are not relevant here. It just means I forgot to turn off specific compiler warnings for that file. Nothing to worry about :) (just C++ specific warnings about reorder intialization, IMO those warnings are quite useless/verbose anyway)

nicop4, 2019-01-10:
Hi,
Thanks for your amazing work on this framework I discovered not a very long time ago.
I confirm something has gone wrong with effects. It is ok without them (I have to retry the API though). The device (an ESP-01) keeps booting in loop.

If I remove the fireworks effect from the config file, the device can boot and I can control on/off and color but crash when I select an effect. I didn't tried other combinations.

I have some logs that might be interesting for debug:
`[21:19:49][I][logger:071]: Log initialized
[21:19:49][C][ota:461]: There have been 9 suspected unsuccessful boot attempts.
[21:19:49]
[21:19:49]Exception (3):
WARNING Exception type: unknown
[21:19:49]epc1=0x4020668a epc2=0x00000000 epc3=0x00000000 excvaddr=0x40000000 depc=0x00000000
WARNING Decoded 0x4020668a: esphomelib::light::ESPColorCorrection::calculate_gamma_table(float)
[21:19:49]
[21:19:49]ctx: cont
[21:19:49]sp: 3ffffc60 end: 3fffffd0 offset: 01a0
[21:19:49]
[21:19:49]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[21:19:49]3ffffe00:  3ffffe50 00000013 3ffffe50 4021f2c8
WARNING Decoded 0x4021f2c8: std::string::_Rep::_M_destroy(std::allocator<char> const&) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:732
[21:19:49]3ffffe10:  3fff2938 3ffffe54 3fff30d3 4021f2ea
WARNING Decoded 0x4021f2ea: std::string::_Rep::_M_dispose(std::allocator<char> const&) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:732
[21:19:49]3ffffe20:  3fff30d5 00000001 3fff2f84 4021f421
WARNING Decoded 0x4021f421: std::string::_M_leak() at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:732
[21:19:49]3ffffe30:  3ffffe60 3fff2950 3fff28fc 4021f44c
WARNING Decoded 0x4021f44c: std::string::_M_data() const at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:732
 (inlined by) std::string::end() at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:627
[21:19:49]3ffffe40:  3fff2cf4 3fff2f7c 3fff2f84 00000000
[21:19:49]3ffffe50:  00000100 3ffffec0 00000000 4020666a
WARNING Decoded 0x4020666a: esphomelib::light::ESPColorCorrection::calculate_gamma_table(float)
[21:19:49]3ffffe60:  3fff3034 40203362 00000034 3fffff68
WARNING Decoded 0x40203362: esphomelib::Application::register_light(esphomelib::light::LightState*)
[21:19:49]3ffffe70:  3fff3378 3fff2cf4 3ffffe90 4020820c
WARNING Decoded 0x4020820c: esphomelib::light::FastLEDLightOutputComponent::setup_state(esphomelib::light::LightState*)
[21:19:49]3ffffe80:  3fff32f4 40210c0c 00000000 40207f89
WARNING Decoded 0x40210c0c: operator new(unsigned int)
WARNING Decoded 0x40207f89: esphomelib::light::AddressableFireworksEffect::init()
[21:19:49]3ffffe90:  3fff2cf4 4021f2b0 0000002e 3fffff68
WARNING Decoded 0x4021f2b0: std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator<char> const&) at /Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:732
[21:19:49]3ffffea0:  00000000 40210c0c 00000018 401004e8
WARNING Decoded 0x40210c0c: operator new(unsigned int)
WARNING Decoded 0x401004e8: malloc
[21:19:49]3ffffeb0:  3fffff50 3fff3384 3fff2f7c 40216f8a
WARNING Decoded 0x40216f8a: esphomelib::light::LightEffect::init_(esphomelib::light::LightState*)
[21:19:49]3ffffec0:  03020100 07060504 0b0a0908 0f0e0d0c
[21:19:49]3ffffed0:  13121110 17161514 1b1a1918 1f1e1d1c
[21:19:49]3ffffee0:  23222120 27262524 2b2a2928 2f2e2d2c
[21:19:49]3ffffef0:  33323130 37363534 3b3a3938 3f3e3d3c
[21:19:49]3fffff00:  43424140 47464544 4b4a4948 4f4e4d4c
[21:19:49]3fffff10:  53525150 57565554 5b5a5958 5f5e5d5c
[21:19:49]3fffff20:  63626160 67666564 6b6a6968 6f6e6d6c
[21:19:49]3fffff30:  73727170 77767574 7b7a7978 7f7e7d7c
[21:19:49]3fffff40:  83828180 87868584 8b8a8988 8f8e8d8c
[21:19:49]3fffff50:  93929190 97969594 9b9a9998 9f9e9d9c
[21:19:49]3fffff60:  a3a2a1a0 a7a6a5a4 abaaa9a8 afaeadac
[21:19:49]3fffff70:  b3b2b1b0 b7b6b5b4 bbbab9b8 bfbebdbc
[21:19:49]3fffff80:  c3c2c1c0 c7c6c5c4 cbcac9c8 cfcecdcc
[21:19:49]3fffff90:  d3d2d1d0 d7d6d5d4 dbdad9d8 dfdedddc
[21:19:49]3fffffa0:  e3e2e1e0 e7e6e5e4 ebeae9e8 efeeedec
[21:19:49]3fffffb0:  f3f2f1f0 f7f6f5f4 fbfaf9f8 fffefdfc
[21:19:49]3fffffc0:  03020100 07060504 0b0a0908 0f0e0d0c
[21:19:49]<<<stack<<<
[21:19:49]
[21:19:49] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[21:19:49]
[21:19:49]load 0x4010f000, len 1384, room 16
[21:19:49]tail 8
[21:19:49]chksum 0x2d
[... 203 lines cut ...]
# Enable logging
logger:

ota:
  password: 'xxxxxxxxxxxx'

light:
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO2
    num_leds: 108
    rgb_order: GRB
    max_refresh_rate: 30ms
    name: "Leds RGB baie vitrée"
    effects:
      - addressable_rainbow:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_flicker:
      - addressable_flicker:
          name: Flicker Effect With Custom Values
          update_interval: 20ms
          intensity: 10%`

Hope this will help.

DavidDeSloovere, 2019-01-10:
And here's a part of mine (finally got it, learned something new again today)

```
...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[21:57:01]sdlܟ<�l�|�l�b|���\033�{�c�b��o'�log���c8ćdsdrdx�o��lĜco�|l��#��og�l��d`�\033g'd`gs���oc�l{��gc�d�ccc���l`��g�
[21:57:01]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[21:57:01][I][logger:071]: Log initialized
[21:57:01][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[21:57:01]Fatal exception 3(LoadStoreErrorCause):
[21:57:01]epc1=0x40207792, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40000000, depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphomeyaml/wemos_nasty_eagle -t idedata
WARNING Decoded 0x40207792: esphomelib::light::ESPColorCorrection::calculate_gamma_table(float)
[21:57:10]
[21:57:10]Exception (3):
WARNING Exception type: unknown
[21:57:10]epc1=0x40207792 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40000000 depc=0x00000000
WARNING Decoded 0x40207792: esphomelib::light::ESPColorCorrection::calculate_gamma_table(float)
[21:57:10]
[21:57:10]ctx: cont 
[21:57:10]sp: 3ffffbd0 end: 3fffffd0 offset: 01a0
[21:57:10]
[21:57:10]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[21:57:10]3ffffd70:  3fff610c 00000002 3fff60d8 4020b5bd  
WARNING Decoded 0x4020b5bd: void std::vector<std::function<void ()>, std::allocator<std::function<void ()> > >::_M_emplace_back_aux<std::function<void ()> >(std::function<void ()>&&)
[21:57:10]3ffffd80:  3fff5494 3fff5484 3fff5494 3fff530c  
[21:57:10]3ffffd90:  3fff604c 3fff5538 00000014 3fff530c  
[21:57:10]3ffffda0:  3fff604c 3fff5538 3fff604c 4020b606  
WARNING Decoded 0x4020b606: esphomelib::light::LightState::add_new_remote_values_callback(std::function<void ()>&&)
[21:57:10]3ffffdb0:  3fff5d94 3fff604c 3fff604c 40215886  
WARNING Decoded 0x40215886: esphomelib::StoringUpdateListenerController::register_light(esphomelib::light::LightState*)
[21:57:10]3ffffdc0:  00000100 3ffffe30 00000000 40207772  
WARNING Decoded 0x40207772: esphomelib::light::ESPColorCorrection::calculate_gamma_table(float)
[21:57:10]3ffffdd0:  3fff604c 00000010 3fff3e84 3ffffeec  
[21:57:10]3ffffde0:  3fff6ab0 3fff578c 3ffffe00 40209a5c  
WARNING Decoded 0x40209a5c: esphomelib::light::FastLEDLightOutputComponent::setup_state(esphomelib::light::LightState*)
[21:57:10]3ffffdf0:  3fff69a4 3ffe8a24 00000000 402096c1  
WARNING Decoded 0x402096c1: esphomelib::light::AddressableFireworksEffect::init()
[21:57:10]3ffffe00:  00000030 00000000 3fff3e84 3ffffeec  
[21:57:10]3ffffe10:  00000000 00000000 00000030 401007c5  
WARNING Decoded 0x401007c5: malloc
[21:57:10]3ffffe20:  00000024 3fff6ab8 3fff59f4 40220d5e  
WARNING Decoded 0x40220d5e: esphomelib::light::LightEffect::init_(esphomelib::light::LightState*)
[21:57:10]3ffffe30:  03020100 07060504 0b0a0908 0f0e0d0c  
[21:57:10]3ffffe40:  13121110 17161514 1b1a1918 1f1e1d1c  
[21:57:10]3ffffe50:  23222120 27262524 2b2a2928 2f2e2d2c  
[21:57:10]3ffffe60:  33323130 37363534 3b3a3938 3f3e3d3c  
[21:57:10]3ffffe70:  43424140 47464544 4b4a4948 4f4e4d4c  
[21:57:10]3ffffe80:  53525150 57565554 5b5a5958 5f5e5d5c  
[21:57:10]3ffffe90:  63626160 67666564 6b6a6968 6f6e6d6c  
[21:57:10]3ffffea0:  73727170 77767574 7b7a7978 7f7e7d7c  
[21:57:10]3ffffeb0:  83828180 87868584 8b8a8988 8f8e8d8c  
[21:57:10]3ffffec0:  93929190 97969594 9b9a9998 9f9e9d9c  
[21:57:10]3ffffed0:  a3a2a1a0 a7a6a5a4 abaaa9a8 afaeadac  
[21:57:10]3ffffee0:  b3b2b1b0 b7b6b5b4 bbbab9b8 bfbebdbc  
[21:57:10]3ffffef0:  c3c2c1c0 c7c6c5c4 cbcac9c8 cfcecdcc  
[21:57:10]3fffff00:  d3d2d1d0 d7d6d5d4 dbdad9d8 dfdedddc  
[21:57:10]3fffff10:  e3e2e1e0 e7e6e5e4 ebeae9e8 efeeedec  
[21:57:10]3fffff20:  f3f2f1f0 f7f6f5f4 fbfaf9f8 fffefdfc  
[21:57:10]3fffff30:  03020100 07060504 0b0a0908 0f0e0d0c  
[21:57:10]3fffff40:  13121110 17161514 1b1a1918 1f1e1d1c  
[21:57:10]3fffff50:  23222120 27262524 2b2a2928 2f2e2d2c  
[21:57:10]3fffff60:  33323130 37363534 3b3a3938 3f3e3d3c  
[21:57:10]3fffff70:  43424140 47464544 4b4a4948 4f4e4d4c  
[21:57:10]3fffff80:  53525150 57565554 5b5a5958 5f5e5d5c  
[21:57:10]3fffff90:  63626160 67666564 6b6a6968 6f6e6d6c  
[... 71 lines cut ...]
[21:57:11]3fffff80:  53525150 57565554 5b5a5958 5f5e5d5c  
[21:57:11]3fffff90:  63626160 67666564 6b6a6968 6f6e6d6c  
[21:57:11]3fffffa0:  73727170 77767574 7b7a7978 7f7e7d7c  
[21:57:11]3fffffb0:  83828180 87868584 8b8a8988 8f8e8d8c  
[21:57:11]3fffffc0:  93929190 97969594 9b9a9998 9f9e9d9c  
[21:57:11]<<<stack<<<
[21:57:11]
[21:57:11] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[21:57:11]
[21:57:11]load 0x4010f000, len 1384, room 16 
[21:57:11]tail 8
[21:57:11]chksum 0x2d
[21:57:11]csum 0x2d
[21:57:11]vbb28d4a3
[21:57:11]~ld
[21:57:11]
[21:57:11]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[21:57:11][I][logger:071]: Log initialized
[21:57:11][C][ota:461]: There have been 2 suspected unsuccessful boot attempts.
...
[21:57:15][C][ota:461]: There have been 10 suspected unsuccessful boot attempts.
[21:57:15][E][ota:466]: Boot loop detected. Proceeding to safe mode.
[21:57:15][C][wifi:029]: Setting up WiFi...
...
```

nicop4, 2019-01-10:
I did other tests and tried to activate the API option and upload the firmware.
It was running fine at first, I could change color, but it seems an issue also occurs when I start using the API:

```
﻿﻿[22:40:21][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[22:40:21][I][application:053]: Running through setup()...
[22:40:21][C][light.fast_led:035]: Setting up FastLED light...
[22:40:21][C][light.fast_led:042]:     Max refresh rate: 30000
[22:40:21][C][light.state:157]: Setting up light 'Leds RGB baie vitrée'...
[22:40:21][C][wifi:029]: Setting up WiFi...
[22:40:21][D][wifi:247]: Starting scan...
[22:40:27][D][wifi:262]: Found networks:
[22:40:27][I][wifi:295]: [redacted]
[22:40:27][D][wifi:296]: [redacted]
[22:40:27][D][wifi:297]: [redacted]
[22:40:27][D][wifi:299]: [redacted]
[22:40:27][D][wifi:299]: [redacted]
[22:40:27][D][wifi:299]: [redacted]
[22:40:27][D][wifi:299]: [redacted]
[22:40:27][D][wifi:299]: [redacted]
[22:40:27][I][wifi:166]: WiFi Connecting to 'MyNetwork'...
[22:40:29][I][wifi:336]: WiFi connected!
[22:40:29][C][wifi:227]:   SSID: 'MyNetwork'
[22:40:29][C][wifi:228]:   IP Address: 192.168.3.33
[22:40:29][C][wifi:230]:   BSSID: 18:D6:C7:XX:XX:XX
[22:40:29][C][wifi:232]:   Hostname: 'rgbwindow1'
[22:40:29][C][wifi:237]:   Signal strength: -57 dB ▂▄▆█
[22:40:29][C][wifi:238]:   Channel: 8
[22:40:29][C][wifi:239]:   Subnet: 255.255.255.0
[22:40:29][C][wifi:240]:   Gateway: 192.168.3.1
[22:40:29][C][wifi:241]:   DNS1: 8.8.8.8
[22:40:29][C][wifi:242]:   DNS2: 8.8.4.4
[22:40:29][C][api:023]: Setting up Home Assistant API server...
[22:40:29][C][ota:129]: Over-The-Air Updates:
[22:40:29][C][ota:130]:   Address: 0.0.0.0:8266
[22:40:29][C][ota:132]:   Using Password.
[22:40:29][C][mqtt.client:024]: Setting up MQTT...
[22:40:29][I][mqtt.client:170]: Connecting to MQTT...
[22:40:29][I][mqtt.client:209]: MQTT Connected!
[22:40:29][I][application:089]: setup() finished successfully!
[22:40:29][I][application:097]: You're running esphomelib v1.10.0 compiled on Jan 10 2019, 22:36:41
[22:40:29][C][wifi:328]: WiFi:
[22:40:29][C][wifi:227]:   SSID: 'MyNetwork'
[22:40:29][C][wifi:228]:   IP Address: 192.168.3.33
[22:40:29][C][wifi:230]:   BSSID: 18:D6:C7:XX:XX:XX
[22:40:29][C][wifi:232]:   Hostname: 'rgbwindow1'
[22:40:29][C][wifi:237]:   Signal strength: -58 dB ▂▄▆█
[22:40:29][C][wifi:238]:   Channel: 8
[22:40:29][C][wifi:239]:   Subnet: 255.255.255.0
[22:40:29][C][wifi:240]:   Gateway: 192.168.3.1
[22:40:29][C][wifi:241]:   DNS1: 8.8.8.8
[22:40:29][C][wifi:242]:   DNS2: 8.8.4.4
[22:40:29][C][light.fast_led:045]: FastLED light:
[22:40:29][C][light.fast_led:046]:     Num LEDs: 108
[22:40:29][C][light.fast_led:047]:     Max refresh rate: 30000
[22:40:29][C][logger:099]: Logger:
[22:40:29][C][logger:100]:   Level: DEBUG
[22:40:29][C][logger:101]:   Log Baud Rate: 115200
[22:40:30][C][light.state:269]: Light 'Leds RGB baie vitrée'
[22:40:30][C][light.state:271]:   Default Transition Length: 1000 ms
[22:40:30][C][light.state:272]:   Gamma Correct: 2.80
[22:40:30][C][api:072]: API Server:
[22:40:30][C][api:073]:   Address: 0.0.0.0:6053
[22:40:30][C][ota:129]: Over-The-Air Updates:
[22:40:30][C][ota:130]:   Address: 0.0.0.0:8266
[22:40:30][C][ota:132]:   Using Password.
[22:40:30][C][mqtt.client:058]: MQTT:
[22:40:30][C][mqtt.client:060]:   Server Address: 192.168.3.3:1883 (192.168.3.3)
[22:40:30][C][mqtt.client:061]:   Username: 'xxxx'
[22:40:30][C][mqtt.client:062]:   Client ID: 'rgbwindow1-18fe34f26316'
[... 153 lines cut ...]
[22:42:08]3ffff1b0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff1c0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff1d0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff1e0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff1f0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff200:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff210:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff220:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff230:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff240:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff250:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff260:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff270:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff280:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff290:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff2a0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff2b0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff2c0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff2d0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff2e0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff2f0:  feefeffe feefeffe feefeffe feefeffe
[22:42:08]3ffff300:  feefeffe feefeffe feefeffe feefeffe
[22:42:13]3ffff310:  feefe
```
The device started to boot and crash in loop when I tried to access the API through Home Assistant

bruhautomation, 2019-01-11:
Can confirm.  Similar experiences on NodeMCU esp8266 and esp32 devices.  Everything runs great until I combine the api + fastled_clockless components. 

Also, this is an Incredible project. Esphomeyaml + HassIO is a game changer. Thanks much for your work and pushing the ball so far forward!

jchambers2012, 2019-01-12:
Effect are also broken with neopixel, since I think they are the same functions for effects, that would make sense:

[D][light.state:457]: 'Stairs Lights' Turning ON
[D][light.state:499]:   Transition Length: 1000 ms
[D][light.state:506]:   Effect: 'Rainbow'

Exception (9):
epc1=0x40208244 epc2=0x00000000 epc3=0x00000000 excvaddr=0xed011d41 depc=0x00000000

ctx: cont
sp: 3ffffac0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffc60:  3fff475c 3ffffca0 3fff3e44 40213f96
3ffffc70:  3fff3eb0 3fff38bc 3fff3e44 4020a809
3ffffc80:  00000000 00000000 4bc6a7f0 00000000
3ffffc90:  00000034 00000034 401067d4 000078fe
3ffffca0:  3fff3ecc 3fff38bc 00000034 00000000
3ffffcb0:  00000000 00000001 3fff386c 40212a5d
3ffffcc0:  3fff38bc 3ffffd60 3fff2714 3ffe9104
3ffffcd0:  00000034 3fff38bc 3fff3e44 4020a8dd
3ffffce0:  40202b10 00000001 3fff44cc 40202b2b
3ffffcf0:  00000004 00000000 3fff4814 3fff38bc
3ffffd00:  3fff5a84 3fff475c 3fff5a64 3fff4994
3ffffd10:  000003e8 3ffe9104 3ffffe30 40209a39
3ffffd20:  3ffffd70 3ffffd60 0000000c 4020843e
3ffffd30:  3ffffd70 3ffffd60 0000000c 4021f800
3ffffd40:  3f800000 3f800000 3f800000 40209a5c
3ffffd50:  3ffffd70 3ffffd60 0000000c 3fff4994
3ffffd60:  000003e8 00000001 3fff4a4c 4020756b
3ffffd70:  000001fa 3fff4a40 3fff475c 40207f51
3ffffd80:  000003e8 3ffe9104 3ffffe30 40208641
3ffffd90:  3ffe90bc 00000361 00000361 4010020c
3ffffda0:  3ffe9414 3fff6e64 3ffffe30 401006dc
3ffffdb0:  00010101 00000000 00000000 40213db0
3ffffdc0:  3fff4d24 3fff6e64 3ffffe30 40229838
3ffffdd0:  3f800000 3f800000 3f800000 00000000
3ffffde0:  00000000 3f800000 00000000 00010101
3ffffdf0:  00000000 00000000 00000000 3fff4994
3ffffe00:  3ffe9414 3fff6e64 3ffffe30 40208f70
3ffffe10:  3fff4d30 3ffffeb0 00000022 3fff4994
3ffffe20:  3ffffee4 3fff499c 3fff6e64 402094ca
3ffffe30:  3fff475c 00000101 3fff2600 402061cd
3ffffe40:  3ffffe00 00000010 3fff2600 40206242
3ffffe50:  3fffff00 3fff6e64 3ffffe00 4021f3c8
3ffffe60:  3ffffe00 3fff6e64 3ffffe00 402066b3
3ffffe70:  00000000 3fff6e92 3fff6e01 00000001
3ffffe80:  3fff6e7c 3fff4354 3fff2694 40209e10
3ffffe90:  00000000 00007250 00000004 40205f96
3ffffea0:  3ffffee4 3fff499c 3fff2694 40206755
3ffffeb0:  3fff2694 3fff4cb4 3fff2694 4021f80a
3ffffec0:  00000000 3fff499c 3fff5d2c 4020a75d
3ffffed0:  3fff4cc8 3fff4354 2e147ae1 402094d8
3ffffee0:  402094a8 3fff5d2c 4bc6a7f0 4020a6b4
3ffffef0:  40209df8 4bc6a7f0 401067d4 4021f800
3fffff00:  00000000 00000001 3fff4dec 40209e6c
3fffff10:  3fff4e0c 00000000 4bc6a7f0 00000000
3fffff20:  00000000 3fff2028 401067d4 3fff3e44
3fffff30:  00000000 000078f9 00000000 4020516a
3fffff40:  008be01b 3fff5d4c 3ffe8648 3fff3d74
3fffff50:  3fff4730 00000002 3fff3d74 3fff3e44
3fffff60:  3fff4734 00000002 3fff3e44 402052a0
3fffff70:  3fff4748 00000002 3fff2638 40203731
3fffff80:  00000000 3fff4748 3fff4a4c 3fff475c
3fffff90:  00000001 3fff381c feefeffe 3fff27d4
3fffffa0:  3fffdad0 00000000 3fff27cd 40203478
3fffffb0:  3fffdad0 00000000 3fff27cd 40213f68
3fffffc0:  feefeffe feefeffe 3ffe8648 40100721
<<<stack<<<

[... 171 lines cut ...]
  ssid: 'JAC-I'
  password: '###########'
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.10.85
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.10.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

mqtt:
  broker: '192.168.10.40'
  username: '###########'
  password: '###########'
  discovery: True
  reboot_timeout: 1h

# Enable logging
logger:

ota:

web_server:
  port: 80
```

RaveGun, 2019-01-12:
Hi Otto,

The warnings are legit.
It is about the initialisation order of the ESPColor.
The typedef is R,G,B,W but the initialisation in the addresable_lights.tcc oder is W,R,G,B.
I don't know if this helps.

MFG, DG.

OttoWinter, 2019-01-12:
Update: working on a fix, I think I figured it out now after *a lot* of headbanging. It's really hard to debug stack smashing issues with these ESPs without a JTAG debugger 😝 (Seems to have to do with C++ polymorphism rules together with `static_cast`, rtti saved the day for debugging :) )

@RaveGun Yes the warnings are real (and can be fixed), but they don't affect anything here. AFAIK -Wreorder is just there to protect you from scenarios like `MyClass::MyClass() : field_a_(call_second_method()), field_b_(call_first_method())`. In this case though, as RGBW are all uint8_t (and no function call is involved) it shouldn't be a problem.

OttoWinter, 2019-01-12:
Ok, any brave users wanting to try the patch? 😅 

```yaml
esphomeyaml:
  esphomelib_version:
    branch: fix-addressable-lights
```

jchambers2012, 2019-01-12:
As requested, do i need to run a command to purge the folder?  I will join the chat room

```yaml
esphomeyaml:
  esphomelib_version:
    branch: fix-addressable-lights
  name: stairs
  platform: ESP8266
  board: nodemcuv2
  on_boot:
    then:
        - light.turn_off:
            id: stairs_light
switch:
  - platform: restart
    name: "Stairs Restart"
light:
  - platform: neopixelbus
    variant: WS2813
    pin: D4
    num_leds: 100
    type: RGB
    name: "Stairs Lights"
    id: stairs_light
#    method: BIT_BANG
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: "Rainbow Effect With Custom Values"
          speed: 50
          width: 100
sensor:
  - platform: wifi_signal
    name: "Stairs WiFi Signal Sensor"
    update_interval: 15s
wifi:
  ssid: 'JAC-I'
  password: ''
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.10.85
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.10.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

mqtt:
  broker: '192.168.10.40'
  username: ''
  password: ''
  discovery: True
  reboot_timeout: 1h

# Enable logging
logger:

ota:

web_server:
  port: 80

```
```
root@HA02:/home/homeassistant/esphomeyaml# esphomeyaml stairs.yaml run  --upload-port 192.168.10.85 --no-logs
INFO Reading configuration...
INFO Generating C++ source...
WARNING Local changes in esphomelib copy from git. Will not auto-update.
INFO Core config or version changed, cleaning build files...
INFO Deleting stairs/.piolibdeps
INFO Deleting stairs/.pioenvs
[... 91 lines cut ...]
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
/root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp: In m                                                                                                                        ember function 'bool BearSSL::WiFiClientSecure::_connectSSL(const char*)':
/root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:771:6                                                                                                                        : internal compiler error: in iterative_hash_expr, at tree.c:7078
bool WiFiClientSecure::_connectSSL(const char* hostName) {
^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/stairs/lib4d9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o] Error 1
============================================ [ERROR] Took 72.27 seconds ============================================

```

OttoWinter, 2019-01-12:
@jchambers2012 Interesting, gcc has a bug apparently, but that's out of scope of this issue (and esphomeyaml frankly). To make sure, yes I would recommend calling clean (overflow menu in dashboard, `clean` command in cli). Btw, haven't tried neopixelbus yet, will try once I can confirm fastled is working.

jchambers2012, 2019-01-12:
> `clean`

I tested it on a fastLED, but no effect display now. at least it doesn't crash anymore.

```
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:486]:   Red=78%, Green=100%, Blue=14%
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:486]:   Red=0%, Green=100%, Blue=25%
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:486]:   Red=14%, Green=100%, Blue=100%
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:503]:   Effect: 'Rainbow'
[I][ota:154]: Boot seems successful, resetting boot loop counter.
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:486]:   Red=78%, Green=100%, Blue=71%
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:486]:   Red=14%, Green=100%, Blue=14%
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:486]:   Red=14%, Green=100%, Blue=14%
[D][light.state:496]:   Transition Length: 1000 ms
[D][light.state:454]: 'Dining Room Cabinet Under' Turning ON
[D][light.state:496]:   Transition Length: 1000 ms
```

```yaml
esphomeyaml:
  esphomelib_version:
    branch: fix-addressable-lights
  name: diningroomserver
  platform: ESP8266
  board: nodemcuv2
  on_boot:
    then:
        - light.turn_off:
            id: dining_room_cabinet_inside_a
        - light.turn_off:
            id: dining_room_cabinet_inside_b
        - light.turn_off:
            id: dining_room_cabinet_inside_c
        - light.turn_off:
            id: dining_room_cabinet_inside_d
        - light.turn_off:
            id: dining_room_cabinet_under

switch:
  - platform: restart
    name: "Dining Room Cabinet Restart"
  - platform: template
    name: "Dining Room Cabinet Automation"
    optimistic: true
    id: dining_room_cabinet_automation
  - platform: template
    name: "Dining Room Cabinet Party Mode"
    optimistic: true
    id: dining_room_cabinet_party

light:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D3
    num_leds: 36
    rgb_order: GRB
    color_correct: [80%, 80%, 80%]
    name: "Dining Room Cabinet Inside Left (left)"
    id: dining_room_cabinet_inside_a
    effects:
      - addressable_rainbow:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D8
    num_leds: 36
    rgb_order: GRB
    color_correct: [80%, 80%, 80%]
    name: "Dining Room Cabinet Inside Left (right)"
    id: dining_room_cabinet_inside_b
    effects:
      - addressable_rainbow:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D7
    num_leds: 30
    rgb_order: GRB
    color_correct: [80%, 80%, 80%]
    name: "Dining Room Cabinet Inside Right (left)"
    id: dining_room_cabinet_inside_c
    effects:
      - addressable_rainbow:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D1
    num_leds: 30
    rgb_order: GRB
    color_correct: [80%, 80%, 80%]
    name: "Dining Room Cabinet Inside Right (right)"
    id: dining_room_cabinet_inside_d
    effects:
      - addressable_rainbow:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D4
    num_leds: 56
    rgb_order: GRB
    color_correct: [80%, 80%, 80%]
    name: "Dining Room Cabinet Under"
    id: dining_room_cabinet_under
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: "Rainbow Effect Fast"
          speed: 50
          width: 100
binary_sensor:
  - platform: gpio
    name: "Dining Room Cabinet Left Door"
    pin:
     number: D2
    device_class: door
    id: dining_room_cabinet_door_left
    filters:
      - heartbeat: 3600s
    on_release:
#     if:
#      condition:
#        lambda: "return id(dining_room_cabinet_door_right).value == id(dining_room_cabinet_door_left).value;"
      then:
        - light.turn_off:
            id: dining_room_cabinet_inside_a
        - light.turn_off:
            id: dining_room_cabinet_inside_b
    on_press:
      then:
        - light.turn_on:
            id: dining_room_cabinet_inside_a
            brightness: 80%
        - light.turn_on:
            id: dining_room_cabinet_inside_b
            brightness: 80%

  - platform: gpio
    name: "Dining Room Cabinet Right Door"
    pin:
     number: D6
    device_class: door
    id: dining_room_cabinet_door_right
    filters:
      - heartbeat: 3600s
    on_release:
#     if:
#      condition:
#        lambda: "return id(dining_room_cabinet_door_right).value == id(dining_room_cabinet_door_left).value;"
      then:
        - light.turn_off:
            id: dining_room_cabinet_inside_c
        - light.turn_off:
            id: dining_room_cabinet_inside_d
    on_press:
      then:
        - light.turn_on:
            id: dining_room_cabinet_inside_c
            brightness: 80%
        - light.turn_on:
            id: dining_room_cabinet_inside_d
            brightness: 80%

sensor:
  - platform: wifi_signal
    name: "Dining Room Cabinet WiFi Signal Sensor"
    update_interval: 15s

wifi:
  ssid: 'JAC-I'
  password: ''

mqtt:
  broker: '192.168.10.40'
  username: ''
  password: ''
  discovery: True
  reboot_timeout: 1h

# Enable logging
logger:

ota:

web_server:
   port: 80
```

OttoWinter, 2019-01-12:
at least an improvement 😉

egon0, 2019-01-12:
i also tested the `fix-addressable-lights` branch, the nodemcu still reboots after selecting any `addressable_` effect.

jchambers2012, 2019-01-12:
have you done the clean option too?

egon0, 2019-01-12:
yep.

OttoWinter, 2019-01-12:
Ok, fixed a couple of issues again. One thing was a divide-by-zero mistake in the color correction code. And another one was ... well all colors were "scaled up" to 0 while an effect was active 🤣

jchambers2012, 2019-01-12:
working for me :D

jchambers2012, 2019-01-12:
I just noticed that the green channel has a strange "flicker": https://photos.app.goo.gl/RLhrbgiRJaW1kNAz5 Hope the video can show you what im talking about

jchambers2012, 2019-01-12:
(HD video is still uploading, sorry thought it finished....)

nicodemus, 2019-01-12:
Tossing my experience into the ring. With the fix-addressable-lights, I'm having the experience of massive delays between updates. Using the built-in rainbow effect, the strip seems to partially update; first ~5 LEDs update, rest of the strip is off. Several seconds later, the strip will jump to some phase of the rainbow pattern, and then go static again. This will repeat. I'm using a Wemos D1 Mini, using the DMA method. UART methods are somewhat similar, but the updating seems to be more frequent, often updating every 2-3 seconds. Using a demo Arduino sketch, with the same physical connections, results in the expected result of a smooth color transition across the entire strip.

```yaml
esphomeyaml:
  esphomelib_version:
    branch: fix-addressable-lights
  name: esp1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi0_ssid
  password: !secret wifi0_password
  domain: !secret wifi_domain

mqtt:
  broker: !secret broker_fqdn
  username: !secret broker_username
  password: !secret broker_password

# Enable logging
#logger:

ota:
  password: !secret ota_password

sensor:
  - platform: dht
    model: DHT22
    pin: D7
    temperature:
      name: "Office Temperature"
    humidity:
      name: "Office Humidity"
    update_interval: 15s

light:
  - platform: neopixelbus
    variant: WS2812X
    pin: D4
    type: GRB
    num_leds: 60
    name: "Lightstrip 1"
    effects:
      - addressable_rainbow:
```

The demo sketch that works perfectly:
```
#include <NeoPixelBus.h>

const uint16_t PixelCount = 60; // this example assumes 4 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 2;  // make sure to set this to the correct pin, ignored for Esp8266

#define colorSaturation 128
NeoPixelBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> strip(PixelCount, PixelPin);

RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

HslColor hslRed(red);
HslColor hslGreen(green);
HslColor hslBlue(blue);
HslColor hslWhite(white);
HslColor hslBlack(black);

void setup()
{
    Serial.begin(115200);
    while (!Serial); // wait for serial attach

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();

    Serial.println();
    Serial.println("Running...");
}
RgbColor color;
uint8_t pos;

void loop() {
  for(uint16_t j=0; j<256*5; j++) // complete 5 cycles around the color wheel
  { 
    for(uint16_t i=0; i<PixelCount; i++)
    {
      // generate a value between 0~255 according to the position of the pixel
      // along the strip
      pos = ((i*256/PixelCount)+j) & 0xFF;
      // calculate the color for the ith pixel
      color = Wheel( pos );
      // set the color of the ith pixel
      strip.SetPixelColor(i, color);
    }
    strip.Show();
    delay(50);
  }  
}

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
RgbColor Wheel(uint8_t WheelPos) 
{
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) 
  {
    return RgbColor(255 - WheelPos * 3, 0, WheelPos * 3);
  } else if(WheelPos < 170) 
  {
    WheelPos -= 85;
    return RgbColor(0, WheelPos * 3, 255 - WheelPos * 3);
  } else 
  {
    WheelPos -= 170;
    return RgbColor(WheelPos * 3, 255 - WheelPos * 3, 0);
  }
}
```

OttoWinter, 2019-01-12:
@jchambers2012 I don't think the flickering is caused by ESPHome, it might be more of a hardware issue.

@nicodemus The behavior you're seeing suggests that the ESP is crashing every few seconds. Also, as you're not using any RGBW strips, I would first try with the fastled_clockless platform. At least right now, the former is more mature from the ESPHome integration side (+ I like the library way more, it's really nice C++ code)

cklit, 2019-01-12:
> I just noticed that the green channel has a strange "flicker": https://photos.app.goo.gl/RLhrbgiRJaW1kNAz5 Hope the video can show you what im talking about

Just to confirm, I'm seeing the same behaviour with the green color, using this fix below :) But at least the effects will run now!
Keep up the good work (and thank you).

```
esphomeyaml:
  esphomelib_version:
    branch: fix-addressable-lights
```

OttoWinter, 2019-01-12:
@jchambers2012 I think this might fix your issue: https://github.com/OttoWinter/esphomelib/commit/35efcf18d96f27c3cb41aac01382e3d80c99717c - There was a small typo in the HSV->RGB code and it affects green to yellowish colors.

cklit, 2019-01-12:
> @jchambers2012 I think this might fix your issue: [35efcf1](https://github.com/OttoWinter/esphomelib/commit/35efcf18d96f27c3cb41aac01382e3d80c99717c) - There was a small typo in the HSV->RGB code and it affects green to yellowish colors.

That fixed it :D

nicodemus, 2019-01-12:
I can confirm that it appears most of my issues got fixed along the way, too. 

One small bit lingers, not sure if anyone else can confirm: when set to a static color, the brightness seems to be offset quite a bit. The brightness slider in HA needs to be about 60% before the LEDs begin to light at all, where as with at least the rainbow pattern, the slider seems to behave more as expected, with minimal brightness down near the ~5% mark.

nicodemus, 2019-01-12:
To add, it does seem to go to full brightness when the slider is at maximum, there's just a much narrow active range. I confirmed via serial logging the LEDs first turn on at the 56% mark, at effectively a few % brightness. Anything below that and they are off completely.

OttoWinter, 2019-01-12:
@nicodemus https://github.com/OttoWinter/esphomelib/commit/768c1ec78802b682cddb788548a13f6e40887cdf :)

egon0, 2019-01-12:
the effects work! thanks a lot! \o/

nicodemus, 2019-01-12:
> 
> 
> @nicodemus [768c1ec](https://github.com/OttoWinter/esphomelib/commit/768c1ec78802b682cddb788548a13f6e40887cdf) :)

That's closer to the mark, but still not matching the brightness values for effects (rainbow). Now, static colors begin to turn on at 36%.

RomRider, 2019-01-13:
@OttoWinter, the `AddressableFlickerEffect` is still broken in the latest dev merge.
Whenever you specify the intensity, it breaks as the intensity is defined as being a `uint8_t` but it should  be a `float`.

RomRider, 2019-01-14:
Hey @OttoWinter thanks for the latest `1.10.1` release!

With this release, it seems that:
* all the addressable effects are lower in brightness than the static colors or the same non-addressable effect.
* Also for me static colors start to light up at `11%` (Using a WS2812B led strip and the default gamma settings)
* If you define an intensity value for the addressable flicker effect (eg: `10%`), the addressable flicker effect goes all the way to shutting down all leds, in a nice way though :). It seems that it's reusing the previous iteration color of each led as the new color minus the flicker effect instead of oscillating around the defined color. https://github.com/OttoWinter/esphomelib/blob/dev/src/esphomelib/light/addressable_light_effect.cpp#L351

OttoWinter, 2019-01-15:
> all the addressable effects are lower in brightness than the static colors or the same non-addressable effect.

Do you mean the `addressable_flicker` effect specifically or all in general?

> Also for me static colors start to light up at 11% (Using a WS2812B led strip and the default gamma settings)

That sounds about right. With a gamma setting of 2.8, it will activate at `(0.5/255)^(1/2.8) = 10.78%` (`0.5` because of rounding up and `2.8` for gamma factor) - You can tune down the gamma correction if you want it to activate at lower values.

OttoWinter, 2019-01-15:
I'll close this since the `fastled_clockless` is fixed. If some individual effects are broken issues should be opened for them separately.

closed 2019-01-15 by OttoWinter

DavidDeSloovere, 2019-01-15:
[image]