esphome/esphome-core #369: Cannot connect to hidden SSIDs

jjlawren, 2019-01-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266/Board/Sonoff):**

esp8266 (nodemcu)

**Affected component:**

wifi (https://esphomelib.com/esphomeyaml/components/wifi.html)

**Description of problem:**
After upgrading from a quite old version to 1.10, I can no longer connect to hidden SSIDs. The wifi component repeatedly scans the available wifi networks and fails with "No matching network found!".

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: 'hidden_ssid'
  password: 'mypassword'
```

**Traceback (if applicable):**

```
[20:46:30][W][wifi:304]: No matching network found!
[20:46:35][D][wifi:247]: Starting scan...
[20:46:41][D][wifi:262]: Found networks:
[20:46:41][D][wifi:299]: - '' (46:D9:E7:91:1B:CC) ▂▄▆█
[20:46:41][D][wifi:299]: - 'myssid' (44:D9:E7:91:1B:CC) ▂▄▆█
[20:46:41][D][wifi:299]: - '' (56:D9:E7:91:1B:CC) ▂▄▆█
[20:46:41][D][wifi:299]: - '' (02:9F:C2:74:EF:55) ▂▄▆█
[20:46:41][D][wifi:299]: - '' (F2:9F:C2:74:EF:55) ▂▄▆█
[20:46:41][D][wifi:299]: - 'myssid' (F0:9F:C2:74:EF:55) ▂▄▆█
[20:46:41][W][wifi:304]: No matching network found!
[20:46:46][D][wifi:247]: Starting scan...
```

Note that 'myssid' above is a broadcasted SSID, and not the hidden one I'm trying to connect to.

**Additional information:**

jjlawren, 2019-01-11:
I can confirm one of the listed BSSIDs with an empty listed SSID is the network I'd like to connect to.

psbaltar, 2019-01-11:
Coincidentally, I just sat down to report the same issue.

It won't connect if the SSID is hidden. Telling my AP to unhide it allows it to connect (no other changes). ESPEasy is able to connect to the same AP while the SSID is hidden. So I'm pretty sure there's a bug in esphomelib.

OttoWinter, 2019-01-11:
I think the way I'll solve it is by adding a new `hidden` configuration option.

Workaround is: use BSSID

```yaml
wifi:
  networks:
  - bssid: 46:D9:E7:91:1B:CC  # replace
    password: ...
```

Misak, 2019-01-12:
Same here, I can't connect to hidden ssid with d1mini. When I switch my ssid to be visible it can connect without problems. I am using latest docker image ottowinter/esphomeyaml.
I agree with subzero79, great work. Thanks

quazzie, 2019-01-12:
Yeah, BSSID does not work for me either. "Wifi Connecting to ''..." 
then "Timeout while connecting to wifi"

meijerwynand, 2019-01-12:
Greetings, 

Just upgraded to 1.10.0. Also experiencing the same issue. 

Unable to connect to hidden wifi network, ```bssid``` option also not working and I am running multiple AP's on same SSID name, so no luck on that option either.

The new cli looks kickass =]

oscar-b, 2019-01-13:
Would it be possible to specify multiple BSSID so that the closest AP is used?

OttoWinter, 2019-01-13:
@oscar-b https://esphomelib.com/esphomeyaml/components/wifi.html#connecting-to-multiple-networks - but as said here using `bssid` also doesn't work.

oscar-b, 2019-01-13:
@OttoWinter  I want it to connect to the same hidden SSID though, using the closest BSSID (APs), not different networks. I would have to specify the same network and password multiple times with the current implementation, as far as I can tell?

subzero79, 2019-01-14:
it works now. :+1:

meijerwynand, 2019-01-15:
Confirmed working to connect to hidden WiFi (NodeMCU)

For the sake of easy access, clarity, and those who just google to copy paste the correct configs...

To connect to hidden on 1.10.1

```yaml
...
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  fast_connect: true
...
```

commit 371a9f8b (2019-01-13): Add WiFi fast connect mode (#385)

Fixes #369
commit 5361e462 (2019-02-17): Add WiFi fast connect mode (#385)

Fixes #369