esphome/esphome-core #370: Lost Connection Using Remote Transmitter And Repeat

JCrankWV, 2019-01-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, Homeassistant v0.85.0

**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP32

**Affected component:**

[Remote Transmiter](https://esphomelib.com/esphomeyaml/components/switch/remote_transmitter.html)

**Description of problem:**
Using the Homeassistant Native API
When repeat is used under remote transmitter.
When I control the remote transmitter switch from home assistant the command runs one time then I loose connection to the ESP32 also I can not connect to the web server of the ESP32, but I still receive messages in the remote ESP32 log "WARNING Disconnected from API".  I must power cycle the ESP32 to get it to come back up.

If I remove the repeat section from the yaml file it functions fine without loosing connection.

If I control the IR transmitter from the ESP32s web server it works fine with the repeat in place.

The repeats where working fine when I was using MQTT, but I was also using older versions of both homeassistant and EPSHomeYAML then.

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  pin: GPIO33
  carrier_duty_percent: 50%

switch:
  - platform: remote_transmitter
    name: "Aquarium Light Power"
    icon: "mdi:remote"
    nec:
      address: 0x807F
      command: 0x00FF
  - platform: remote_transmitter
    name: "Aquarium Light Max"
     repeat:
       times: 3
       wait_time: 3s
    icon: "mdi:remote"
    nec:
      address: 0x807F
      command: 0xC23D
```

**Traceback (if applicable):**

Not receiving any errors from the ESP except for
```
[22:21:53][VV][api:946]: on_switch_command_request_
[22:21:53][D][switch:033]: 'Aquarium Light M1' Turning ON.
[22:21:53][D][switch:055]: 'Aquarium Light M1': Sending state ON
[22:21:57][VV][component:134]: Running interval '':0 with interval=10000 last_execution=333440 (now=343449)
[22:21:59][VV][component:134]: Running interval 'update':0 with interval=15000 last_execution=329980 (now=344986)
[22:21:59][V][sensor.sensor:020]: 'Aquarium Light WiFi Signal': Received new state -55.000000
WARNING Disconnected from API.
INFO Connecting to 192.168.1.50:6053 (192.168.1.50)
WARNING Couldn't connect to API (Error connecting to 192.168.1.50: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 192.168.1.50:6053 (192.168.1.50)
WARNING Couldn't connect to API (Error connecting to 192.168.1.50: timed out). Trying to reconnect in 2 seconds
```

Errors in home assistant log at disconnect
```
2019-01-10 22:36:06 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.1.50                         │
  │2019-01-10 22:36:07 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 192.168.1.50: Timeout while waiting fo│
  │r API response!  
```

Errors in home assistant after I power cycle the ESP32
```
│2019-01-10 22:42:06 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.1.50                         │
Tr│2019-01-10 22:42:13 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to 192.168.1.50                                 │
  │2019-01-10 22:42:13 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved                                 │
  │Traceback (most recent call last):                                                                                                              │
  │  File "/usr/local/lib/python3.6/site-packages/aioesphomeapi/client.py", line 542, in run_forever                                               │
  │    await self._run_once()                                                                                                                      │
  │  File "/usr/local/lib/python3.6/site-packages/aioesphomeapi/client.py", line 520, in _run_once                                                 │
  │    preamble = await self._recv(1)                                                                                                              │
At│  File "/usr/local/lib/python3.6/site-packages/aioesphomeapi/client.py", line 507, in _recv                                                     │
20│    ret = await self._socket_reader.readexactly(amount)                                                                                         │
20│AttributeError: 'NoneType' object has no attribute 'readexactly'  
```

**Additional information:**

OttoWinter, 2019-01-11:
Wait time is not specifically designed for such long wait times, but I guess it could be handled better.
Will be fixed in 1.10.1, workaround is don't use such a long wait time - even after the fix you'll suspend execution of the main loop by that duration

JCrankWV, 2019-01-11:
Thank you, I was using the delay as a fix for the ir command not reaching the device all the time.  I didn’t think about it suspending code execution for the duration.

I will send multiple commands from homeassistsnt rather then using the repeat in ESPHomelib

BTW:  Thanks for all the time you have put into the awesome software.

commit f5a3e34c (2019-01-13): Fix remote transmitter ESP32 repeat wait (#376)

* Fix ESP32 remote transmitter repeat wait

Fixes #370

* Defer Send
commit adaeee6f (2019-02-17): Fix remote transmitter ESP32 repeat wait (#376)

* Fix ESP32 remote transmitter repeat wait

Fixes #370

* Defer Send
closed 2019-01-13 by OttoWinter