esphome/esphome-core #372: API connects first node, but not second.

JAAlperin, 2019-01-11:
Uploaded new yaml file containing MQTT discovery: clean. OK
Followed procedure for revising registry (with hass stopped).
Removed MQTT, added api and reloaded the node.  Got errors:

INFO Successfully compiled program.
INFO Connecting to 192.168.0.139
INFO Uploading /config/esphomeyaml/nodemcu03/.pioenvs/nodemcu03/firmware.bin (346688 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.139 using esphomelib API
INFO Connecting to 192.168.0.139:6053 (192.168.0.139)
WARNING Couldn't connect to API (Error connecting to 192.168.0.139: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.139:6053 (192.168.0.139)
WARNING Couldn't connect to API (Error connecting to 192.168.0.139: [Errno 111] Connection refused). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.0.139:6053 (192.168.0.139)
WARNING Couldn't connect to API (Error connecting to 192.168.0.139: [Errno 111] Connection refused). Trying to reconnect in 4 seconds

Restarted hass.  Confirmed that previous mqtt entity was removed.
Tried to add second integration for espHome, got error:

ESPHome
Please enter connection settings of your ESPHome node.

Can't connect to ESP. Please make sure your YAML file contains an 'api:' line.
 
Host
192.168.0.139
 
Port
6053

Using dashboard, confirmed that yaml file valid and contains api.
Restarted hass.
Tried to configure second integration again.  Still errors as above.

Am I doing something wrong?

OttoWinter, 2019-01-11:
Let's leave the HA side for a bit. The error here is on the ESP, even the esphomeyaml client can't connect to the native API, so HA most certainly can't either.

For debugging this I'll need USB logs - for some reason the native API server is not coming up. Maybe it's the OTA safe mode kicking in because there are repeated reboots.

JAAlperin, 2019-01-11:
Thanks, Otto.

I haven't seen the OTA reboots about which you speculate, but maybe.

Regarding "the native API server is not coming up" - it is up and functioning for another node, but won't engage for the second node.

By USB logs I gather you want to update via USB versus OTA.  Any specific directions or just repeat prior upload over USB?

talondnb, 2019-01-12:
> In prior releases I could get into usb flash mode by connecting the esp8266 (nodemcu) and restarting the esphomeyaml addon. Tried everything I could think of, but cannot get 1.10.0 to see the usb connection. Open to suggestions.

Same issue here, multiple devices. ESPHome unable to pick up the device, even though HA shows the device listed (e.g. /dev/ttyUSB1).

I'm also having API connection issues since moving to native, tried with and without password but to no avail. Not even able to connect first node. It seems to have cause an intermittent connection issue now with my garage opener, dropping every so often.

JAAlperin, 2019-01-13:
Tried 1.10.1 on 85.1 OTA.  Does not fix my API problem.  Will try USB later.

JAAlperin, 2019-01-14:
Hello, Otto.

Two USB flash attempts.  Yaml for API fails in boot.  Yaml for MQTT is OK.

[Compile And Upload nodemcu03 API.txt](https://github.com/OttoWinter/esphomelib/files/2755234/Compile.And.Upload.nodemcu03.API.txt)
[Compile And Upload nodemcu03 MQTT.txt](https://github.com/OttoWinter/esphomelib/files/2755236/Compile.And.Upload.nodemcu03.MQTT.txt)

Thanks for your help and for EspHomeLib.

OttoWinter, 2019-01-15:
Ah I see the issue now. `expire_after` is an option for MQTT based installs - the native API does not have that functionality anymore. So each time the `set_expire_after` function is called it will fail because the MQTT client does not exist. So just remove `expire_after` from your config, that should fix the issue.

I will create a patch that creates an error in the YAML validation stage if `expire_after` is used without MQTT.

JAAlperin, 2019-01-17:
Spot on, Mr. Winter.  Three more nodemcu devices now using the api.

Thanks again for your excellent contributions.

closed 2019-01-17 by JAAlperin

agengler11, 2019-01-20:
Is there an upgrade or a reinstall that needs to happen for this to get fixed? I too am having the same issues on 1.10.1