esphome/esphome-core #389: GPIO output behaving weird

AlexDanault, 2019-01-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

n/a

**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266

**Affected component:**

GPIO Output

**Description of problem:**
I have a simple setup with only two components, a GPIO switch commected to a push button and a GPIO outpout connected to a buzzer. When using the switch's on_press/on_release everything works fine, but if I try to add a delay to have the buzzer on for only a small period of time, the buzzer goes on as soon as the ESP is booting, and never stops.

**Problem-relevant YAML-configuration entries:**
This works as expected:
```yaml
output:
  - platform: gpio
    id: buzzer
    pin:
      number: D8
      inverted: True

binary_sensor:
  - platform: gpio
    id: push_button
    pin: D3
    name: "Push button"
    on_click:
      then:
        - output.turn_on: buzzer
    on_release:
      then:
        - output.turn_off: buzzer
```
This doesn't (buzzer is always on):
```
output:
  - platform: gpio
    id: buzzer
    pin:
      number: D8
      inverted: True

binary_sensor:
  - platform: gpio
    id: push_button
    pin: D3
    name: "Push button"
    on_click:
      then:
        - output.turn_on: buzzer
        - delay: 0.1s
        - output.turn_off: buzzer
```

**Traceback (if applicable):**

```

```

**Additional information:**

talondnb, 2019-01-15:
I concur with this, I'm also having an issue with the following:

```
switch:
  - platform: gpio
    pin: D7
    restore_mode: RESTORE_DEFAULT_OFF
    id: relay
  - platform: template
    name: "Garage Door Trigger"
    optimistic: yes
    turn_on_action:
    - switch.turn_on: relay
    - delay: 100ms
    - switch.turn_off: relay
```

Moving to this and using a script within HA is more reliable so far:

```
switch:
  - platform: gpio
    pin: D7
    name: "Garage Door Trigger"
```

talondnb, 2019-01-17:
Just to follow up; I am still running completely fine with the second config I pasted in the above post.

OttoWinter, 2019-01-18:
I was not able to replicate this issue (using an LED, but essentially the same).

@AlexDanault Could you provide a full example for me to test? Of course strip out all secrets etc first.

And @talondnb: Could you please provide more information as to _what_ is not working as expected with your setup? Is it also constantly having the relay on? I tried your config, but it also worked as expected for me.

AlexDanault, 2019-01-18:
@OttoWinter I think you may have fixed a big part of it in #415 , I'll give a good test tonight and I'll report back here.

talondnb, 2019-01-18:
@OttoWinter when using the below config with a relay it would work initially but after a short while the output stops working. Even a reboot of the Nodemcu would not bring it back.

```
switch:
  - platform: gpio
    pin: D7
    restore_mode: RESTORE_DEFAULT_OFF
    id: relay
  - platform: template
    name: "Garage Door Trigger"
    optimistic: yes
    turn_on_action:
    - switch.turn_on: relay
    - delay: 100ms
    - switch.turn_off: relay
```

Changing back to my previous configuration resolved the issue, the only different being the delay and using the relay id.

AlexDanault, 2019-01-19:
Alright @OttoWinter , sorry for the delay. I was able to reproduce the problem with a led and push button.

This basic config works as expected. Expected means: The led is off at boot, and the led responds to the button.

``` yaml
esphomeyaml:
  name: test_gpio
  platform: ESP8266
  board: d1_mini

logger:
  level: VERBOSE

wifi:
  ssid: 'Alex'
  password: 'PASSWORD'

output:
  - platform: gpio
    id: my_led    
    pin:
      number: D8
      inverted: True

binary_sensor:
  - platform: gpio
    id: my_button
    pin: D3
    name: "My button"
    on_press:
      then:
        - output.turn_on: my_led
    on_release:
      then:
        - output.turn_off: my_led
```

But if we want the led to stay on and go off, the following config modification should work, but doesn't. What fails is: The led is fully on at boot and doesn't respond to the push button.

``` yaml
binary_sensor:
  - platform: gpio
    id: my_button
    pin: D3
    name: "My button"
    on_click:
      then:
        - output.turn_on: my_led
        - delay: 1s
        - output.turn_off: my_led
```

After some testing, the delay doesn't seem to cause the problem because removing it doesn't change anything.

This following config modification works partially. The led is still on at boot, but pressing the push button does flash the led for 1 second, but reversed (it is always on, and when clicked it goes off for a second.

``` yaml
binary_sensor:
  - platform: gpio
    id: my_button
    pin: D3
    name: "My button"
    on_press:
      then:
        - output.turn_on: my_led
        - delay: 1s
        - output.turn_off: my_led
```

So it seems that the constant may be having a output.turn_on and output.turn_off in the same event causes the code to ignore the invert set on the led.

OttoWinter, 2019-01-27:
> does flash the led for 1 second, but reversed

Then your LED is active-high. You have inverted the signal with `inverted: True`, please remove that.

If `on_press` works but `on_click` it could be due to two things:

- You're not pressing for the correct amount of time. See the parameters to `on_click`
- The button is inverted. Then `on_click` won't work. Apply an `inverted` to the `binary_sensor.gpio` pin.

closed 2019-01-27 by OttoWinter