esphome/esphome-core #391: cant connect to home assistant

sandro-defender, 2019-01-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
[05:39:33][I][application:114]: Running through first loop()
[05:39:33][I][application:141]: First loop finished successfully!
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: timed out). Trying to reconnect in 1 seconds
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: [Errno 111] Connection refused). Trying to reconnect in 2 seconds
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: [Errno 111] Connection refused). Trying to reconnect in 4 seconds
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: [Errno 111] Connection refused). Trying to reconnect in 8 seconds
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: [Errno 111] Connection refused). Trying to reconnect in 16 seconds
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: [Errno 111] Connection refused). Trying to reconnect in 32 seconds
WARNING Disconnected from API.
INFO Connecting to 1.1.1.93:6053 (1.1.1.93)
WARNING Couldn't connect to API (Error connecting to 1.1.1.93: [Errno 111] Connection refused). Trying to reconnect in 64 seconds

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: switch
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'Sandro-Home'
  password: '************'
  manual_ip:
    static_ip: 1.1.1.93
    gateway: 1.1.1.1
    subnet: 255.255.255.0

api:
#  password: '******'
  reboot_timeout: 5min

mqtt:
  broker: '1.1.1.76'
  username: 'pi'
  password: '******'
  discovery: clean
# Enable logging
logger:
#  level: WARN
#  level: DEBUG
#  level: VERY_VERBOSEWARN

ota:
  safe_mode: True
  password: '******'

web_server:
  port: 80

switch:
  - platform: gpio
    pin:
      number: 15
      inverted: false
    name: "Reley N1"
    id: Reley1
#
  - platform: gpio
    pin:
      number: 12
      inverted: false
    name: "Reley N2"
    id: Reley2
#
  - platform: gpio
    pin:
      number: 13
      inverted: false
    name: "Reley N3"
    id: Reley3
#
  - platform: gpio
    pin:
      number: 14
      inverted: false
    name: "Reley N4"
    id: Reley4
#
  - platform: gpio
    pin:
      number: 05
      inverted: false
    name: "Reley N5"
    id: Reley5
#
  - platform: shutdown
    name: "Mcu.Switch-Shutdown"
  - platform: restart
    name: "Mcu.Switch-Restart"
dallas:
  - pin: GPIO4

sensor:
  - platform: dallas
    address: 0x9A031572938CFF28
    name: "Temperature 8"

binary_sensor:
  - platform: status
    name: "Switch esp"
#
#
  - platform: gpio
    pin:
      number: D3
      inverted: True
    name: switch input 1
    filters:
      - delayed_on: 40ms
    on_press:
      then:
        - switch.turn_on:
            id: Reley1
        - delay: 1s
        - switch.turn_on:
            id: Reley2
    on_release:
      then:
        - delay: 1s
        - switch.turn_off:
            id: Reley2
        - delay: 1s
        - switch.turn_off:
            id: Reley1
#
  - platform: gpio
    pin:
      number: D4
      inverted: True
    name: switch input 2
    filters:
      - delayed_on: 40ms
    on_press:
      then:
        - switch.turn_on:
            id: Reley3
    on_release:
      then:
        - delay: 1s
        - switch.turn_off:
            id: Reley3
```

**Traceback (if applicable):**

```

```

**Additional information:**

ichbestimmtnicht, 2019-01-14:
Did you created the integration in homeassistant? (config -> integration -> esphome) I waited for it at first as well. As soon as I added the Ip there I got prompted for my password afterwards and almost everything (see discord) worked fine.

sandro-defender, 2019-01-15:
Removed mqtt everuthing works fine...

soppilif, 2019-04-05:
I have the same problem with my esphome (hassio 0.91.0) and nodemcu.
When i press the Show Logs:

> INFO Reading configuration...
INFO Starting log output from esp_55ec28.lan using esphome API
INFO Connecting to esp_55ec28.lan:6053 (192.168.1.100)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.100: [Errno 111] Connection refused). Re-Trying in 1 seconds

and my yaml file is (the bruh multisensor):

```
esphome:
  name: esp_55ec28
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret ssid_name
  password: !secret wifi_pwd
  domain: .lan

# Enable logging
logger:

# Enable Home Assistant API
api:
  #password: !secret wifi_pwd

ota:
  password: !secret wifi_pwd

sensor:
  - platform: dht
    pin: D7
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"

  - platform: adc
    pin: A0
    name: "Brightness"
    unit_of_measurement: lux
    filters:
      - lambda: |-
          return (x / 10000.0) * 2000000.0;

binary_sensor:
  - platform: gpio
    pin: D5
    name: "Motion"
    device_class: motion
```

I have followed the instructions for Hassio.
ESPHome integration though fails but i suspect it is not needed since its not mentioned in the instruction page of esphome.

So what now?