esphome/esphome-core #394: sliding_window_moving_average causes boot looping.

tomlut, 2019-01-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.85.1, ESPhome 1.10.0
**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266 (Wemos D1 mini)

**Affected component:**

https://esphomelib.com/esphomeyaml/components/sensor/index.html#sensor-filters

**Description of problem:**
Including the recommended configuration to bring back a windowed moving average causes validation errors:
```
  - platform: adc
    pin: A0
    name: "Master Bed Sensor"
    update_interval: 15s
    filters:
      - multiply: 3.25
      - sliding_window_moving_average:
```

```
sensor.adc: [source /config/esphomeyaml/master_bed.yaml:37]
  platform: adc
  pin: A0
  name: Master Bed Sensor
  update_interval: 15s
  filters:  [source /config/esphomeyaml/master_bed.yaml:42]
    - multiply: 3.25
    - [source /config/esphomeyaml/master_bed.yaml:43]
      
      expected a dictionary.
      sliding_window_moving_average: 
```

Including the full sliding window filter configuration causes boot looping:

```
  - platform: adc
    pin: A0
    name: "Master Bed Sensor"
    update_interval: 15s
    filters:
      - multiply: 3.25
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15
```
```
[18:51:56] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[18:51:56]
[18:51:56]wdt reset
[18:51:56]load 0x4010f000, len 1384, room 16 
[18:51:56]tail 8
[18:51:56]chksum 0x2d
[18:51:56]csum 0x2d
[18:51:56]vbb28d4a3
[18:51:56]~ld
[18:51:56][I][logger:071]: Log initialized
[18:51:56][C][ota:461]: There have been 3 suspected unsuccessful boot attempts.
[18:51:56][I][application:053]: Running through setup()...
[18:51:56][D][binary_sensor:027]: 'Master Bed Sensor Status': Sending state OFF
[18:51:56][C][sensor.adc:028]: Setting up ADC 'Master Bed Sensor'...
[18:51:56][C][wifi:029]: Setting up WiFi...
[18:51:56][D][wifi:260]: Starting scan...
```

Removing the sliding window average configuration compiles and runs as expected. Though the sensor is not averaged (as is wanted):

```
  - platform: adc
    pin: A0
    name: "Master Bed Sensor"
    update_interval: 15s
    filters:
      - multiply: 3.25
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
```
As per descrition above.
**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2019-01-15:
You're right about the first part, indeed only specifying `sliding_window_moving_average:` is invalid.

But with the configuration you gave here, I don't see any boot looping. The log you provided only contains the last part of the log, if it were to crash there would be a long stacktrace printed before that. I would need that stacktrace to debug this issue.

tomlut, 2019-01-15:
I'll give it another go tomorrow and post the full log. Getting late here now.

tomlut, 2019-01-16:
I tried updating to the following (clean build) OTA:

```
esphomeyaml:
  name: master_bed
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'WAPLO'
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.71
    gateway: 10.1.1.1
    subnet: 255.255.255.0

mqtt:
  discovery: true
  discovery_prefix: homeassistant
  broker: '10.1.1.100'
  username: !secret mqtt_usr
  password: !secret mqtt_pwd

logger:
  # level: WARN

ota:
  password: !secret esp_pwd

binary_sensor:
  - platform: status
    name: "Master Bed Sensor Status"

sensor:
  - platform: wifi_signal
    name: "Master Bed Sensor WiFi Signal"
    update_interval: 15s
    filters:
      - sliding_window_moving_average: {}

  - platform: adc
    pin: A0
    name: "Master Bed Sensor"
    update_interval: 4s
    filters:
      - multiply: 3.25
      - sliding_window_moving_average: {}
```

I can ping the device but I receive no sensor updates from it via the log or in HA. The device is shown as offline in ESPhome and the status is 'unavailable' in HA. I waited until the log terminated and a further 5 minutes for HA to show any sign of this sensor.

I removed the sliding window filters and tried re-uploading a clean build OTA.  The ESP then correctly started sending sensor readings again.

There really does seem to be something amiss with the sliding window average filter.

tomlut, 2019-01-16:
Update, I tried removing the multiplication filter and the sliding average from the wifi senso leaving just this sensor:

```
  - platform: adc
    pin: A0
    name: "Master Bed Sensor"
    update_interval: 4s
    filters:
      - sliding_window_moving_average: {}
```

Same problem. The ESP never publishes any mqtt messages. It can be seen as online by ESPhome and can be pinged and updated OTA. Removing the sliding window filter and re-flashing resumes normal operation.

tomlut, 2019-01-17:
I tried this using the exponential average instead. 

```
  - platform: adc
    pin: A0
    name: "Master Bed Sensor"
    update_interval: 3s
    filters:
      - multiply: 3.25
      # - sliding_window_moving_average: {}
      - exponential_moving_average:
          alpha: 0.1
          send_every: 15
```

With this result:

```
[12:52:16]
[12:52:16] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[12:52:16]
[12:52:16]wdt reset
[12:52:16]load 0x4010f000, len 1384, room 16 
[12:52:16]tail 8
[12:52:16]chksum 0x2d
[12:52:16]csum 0x2d
[12:52:16]vbb28d4a3
[12:52:16]~ld
[12:52:16][E][ota:466]: Boot loop detected. Proceeding to safe mode.
[12:52:23][W][ota:136]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
```

tomlut, 2019-01-17:
I just tried a new Wemos D1 Mini board. Same result. Boot loops with `sliding_window_moving_average: {}`, for the A0 sensor, works fine without the filter.

OttoWinter, 2019-01-18:
@tomlut Ok so your logs provide some good info. Specifically, `rst cause:4` means that there was a hardware WDT reset.

However, what is weird that _nothing_ is displayed on boot. If the code would even _reach_ the point where the moving average or anything else for that matter is initialized, the logs are set up. 
And so I would _except_ to see a line like `Logs initialized!` on each reboot. But that message is not there, so that means there's something fundamentally wrong with the ESP. 
Do you have another ESP to check if it's something to do with the ESP?

I also tried your configuration files and was not able to replicate the issue.

tomlut, 2019-01-18:
Ok if you can use A0 with the filter and other sensors specified in my config on the same hardware then it has to be a problem with my D1 board. I’ve tried reinstalling the addon and using the alternate command line approach on a whole different platform (Windows) so that rules out a problem with my ESPhome installation.

I’ll dig out the other D1 and try again tomorrow - with discovery disabled and a correct static IP assignment.

Thanks for your time.

tomlut, 2019-01-19:
I deleted all reference to my other (spare-ish) D1 mini from my router and uploaded this config:

```
esphomeyaml:
  name: test_d1_mini
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'WAPLO'
  password: !secret wifi_pwd
  # manual_ip:
    # static_ip: 10.1.1.71
    # gateway: 10.1.1.1
    # subnet: 255.255.255.0

mqtt:
  # discovery: true
  # discovery_prefix: homeassistant
  broker: '10.1.1.100'
  username: !secret mqtt_usr
  password: !secret mqtt_pwd

logger:
  # level: WARN

ota:
  password: !secret esp_pwd

binary_sensor:
  - platform: status
    name: "D1 Test Sensor Status"

sensor:
  - platform: wifi_signal
    name: "D1 Test WiFi Signal"
    update_interval: 15s
    filters:

  - platform: adc
    pin: A0
    name: "D1 A0 Test Sensor"
    update_interval: 4s
    filters:
      - multiply: 3.25
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15
```

And it works. No issue at all. Looks like my initial D1 mini must have an issue with flash memory at one of the higher addresses - assuming that putting back discovery or a static IP address is not going to reintroduce the fault. But I cant test that until I can remove the existing device integration from HA.

I will close this issue now but would appreciate guidance on how best to remove the master_bed sensor from my HA integration. I'll ask over on the HA community forums though, to prevent cluttering up your issues feed. 

Thanks for your help Otto.

tomlut, 2019-01-19:
Oh damn. Buoyed by the above success I decided to update anotehr ESP D1 mini. And immediately lost contact with it. The config:

```
esphomeyaml:
  name: lounge_dht
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'WAPLO'
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.75
    gateway: 10.1.1.1
    subnet: 255.255.255.0

mqtt:
  discovery: true
  discovery_prefix: homeassistant
  broker: '10.1.1.100'
  username: !secret mqtt_usr
  password: !secret mqtt_pwd

logger:
  # level: WARN

ota:
  password: !secret esp_pwd

binary_sensor:
  - platform: status
    name: "Lounge Room DHT Status"

sensor:
  - platform: wifi_signal
    name: "Lounge Room DHT WiFi Signal"
    update_interval: 15s
    filters:
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15

  - platform: dht
    pin: D5
    model: DHT22
    update_interval: 15s
    temperature:
      name: "Lounge Room Temperature"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 15
    humidity:
      name: "Lounge Room Humidity"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 15
```
The log:

```
NFO Reading configuration...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[14:49:16][I][ota:154]: Boot seems successful, resetting boot loop counter.
[14:49:24][E][ota:485]: No OTA attempt made, restarting.
[14:49:24][I][helpers:277]: Forcing a reboot... Reason: 'ota-safe-mode'
[14:49:24][W][wifi:727]: Event: Disconnected ssid='WAPLO' bssid=FC:EC:DA:38:94:62 reason='Association Leave'
[14:49:24]
[14:49:24] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[14:49:24]
[14:49:24]wdt reset
[14:49:24]load 0x4010f000, len 1384, room 16 
[14:49:24]tail 8
[14:49:24]chksum 0x2d
[14:49:24]csum 0x2d
[14:49:24]vbb28d4a3
[14:49:24]~ld
[14:49:24][I][logger:071]: Log initialized
[14:49:24][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[14:49:24][I][application:053]: Running through setup()...
[14:49:24][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[14:49:24][C][sensor.dht:030]: Setting up DHT...
[14:49:24][C][wifi:029]: Setting up WiFi...
[14:49:24][D][wifi:260]: Starting scan...
[14:49:32]
[14:49:32] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[14:49:32]
[14:49:32]wdt reset
[14:49:32]load 0x4010f000, len 1384, room 16 
[14:49:32]tail 8
[14:49:32]chksum 0x2d
[14:49:32]csum 0x2d
[14:49:32]vbb28d4a3
[14:49:32]~ld
[14:49:32][I][logger:071]: Log initialized
[14:49:32][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[14:49:32][I][application:053]: Running through setup()...
[14:49:32][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[14:49:32][C][sensor.dht:030]: Setting up DHT...
[14:49:32][C][wifi:029]: Setting up WiFi...
[14:49:32][D][wifi:260]: Starting scan...
[14:49:41]
[14:49:41] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[14:49:41]
[14:49:41]wdt reset
[14:49:41]load 0x4010f000, len 1384, room 16 
[14:49:41]tail 8
[14:49:41]chksum 0x2d
[14:49:41]csum 0x2d
[14:49:41]vbb28d4a3
[14:49:41]~ld
[14:49:41][I][logger:071]: Log initialized
[14:49:41][C][ota:461]: There have been 2 suspected unsuccessful boot attempts.
[14:49:41][I][application:053]: Running through setup()...
[14:49:41][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[14:49:41][C][sensor.dht:030]: Setting up DHT...
[14:49:41][C][wifi:029]: Setting up WiFi...
[14:49:41][D][wifi:260]: Starting scan...
[14:49:49]
[14:49:49] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[14:49:49]
[14:49:49]wdt reset
[14:49:49]load 0x4010f000, len 1384, room 16 
[14:49:49]tail 8
[14:49:49]chksum 0x2d
[14:49:49]csum 0x2d
[14:49:49]vbb28d4a3
[14:49:49]~ld
[14:49:49][I][logger:071]: Log initialized
[14:49:49][C][ota:461]: There have been 3 suspected unsuccessful boot attempts.
[14:49:49][I][application:053]: Running through setup()...
[14:49:49][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[14:49:49][C][sensor.dht:030]: Setting up DHT...
[14:49:49][C][wifi:029]: Setting up WiFi...
[14:49:49][D][wifi:260]: Starting scan...
[14:49:58]
[14:49:58] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[14:49:58]
[14:49:58]wdt reset
[14:49:58]load 0x4010f000, len 1384, room 16 
[14:49:58]tail 8
[14:49:58]chksum 0x2d
[14:49:58]csum 0x2d
[14:49:58]vbb28d4a3
[14:49:58]~ld
[14:49:58][I][logger:071]: Log initialized
[14:49:58][C][ota:461]: There have been 4 suspected unsuccessful boot attempts.
[14:49:58][I][application:053]: Running through setup()...
[14:49:58][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[14:49:58][C][sensor.dht:030]: Setting up DHT...
[14:49:58][C][wifi:029]: Setting up WiFi...
[14:49:58][D][wifi:260]: Starting scan...
[14:50:06]
[14:50:06] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[14:50:06]
```

brandond, 2019-01-19:
FYI, you can't change the flash layout via OTA. It just re-uploads the new FW to a flash region just after the current sketch, and then overwrites it as it reboots. If you think that there might be some issue with your flash, you should hook it up to a UART and reflash it from there so that it can actually rewrite the flash map.

I had to do this to get extra space on my ESP8266s with 1M of flash after disabling the SPIFFS reservation. That was fun, especially for the smart bulbs and plugs that I'd glued back together...

tomlut, 2019-01-19:
I'm currently erasing all trace of this config from the esphome folder and will start again from creating a new device and uploading via serial.

tomlut, 2019-01-19:
It failed. 

I am beginning to suspect it is not a problem with the D1 mini's but a problem associating with wifi. I'm about to run some more tests.

tomlut, 2019-01-19:
Ok this is very odd. For the lounge room dht sensor D1:

I just deleted the static IP mapping from my router and removed the manual IP from the D1 mini config. No success after a re-upload via serial. It would still not connect to wifi with DHCP and looped.

I removed mqtt discovery from the D1 mini and re-uploaded - **it worked.** 

I put back the manual IP and static mapping in my router - **it worked.**

I put back mqtt discovery - **it failed.**

Log:
```
Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:23:04]{$l��|�l�<�d�c|����s�c�c��gn�dg'���#x��lrd;lx�o��l��#'�|l��#��o'�d��l �\033gnd`'{���'#�lsۓoc�l�cbc���d`��o�[I][logger:071]: Log initialized
[15:23:04][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[15:23:04][I][application:053]: Running through setup()...
[15:23:04][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[15:23:04][C][sensor.dht:030]: Setting up DHT...
[15:23:04][C][wifi:029]: Setting up WiFi...
[15:23:04][D][wifi:260]: Starting scan...
[15:23:12]
[15:23:12] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[15:23:12]
[15:23:12]wdt reset
[15:23:12]load 0x4010f000, len 1384, room 16 
[15:23:12]tail 8
[15:23:12]chksum 0x2d
[15:23:12]csum 0x2d
[15:23:12]vbb28d4a3
[15:23:12]~ld
[15:23:12][I][logger:071]: Log initialized
[15:23:12][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[15:23:12][I][application:053]: Running through setup()...
[15:23:12][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[15:23:12][C][sensor.dht:030]: Setting up DHT...
[15:23:12][C][wifi:029]: Setting up WiFi...
[15:23:12][D][wifi:260]: Starting scan...
[15:23:20]
[15:23:20] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[15:23:20]
[15:23:20]wdt reset
[15:23:20]load 0x4010f000, len 1384, room 16 
[15:23:20]tail 8
[15:23:20]chksum 0x2d
[15:23:20]csum 0x2d
[15:23:20]vbb28d4a3
[15:23:20]~ld
[15:23:20][I][logger:071]: Log initialized
[15:23:20][C][ota:461]: There have been 2 suspected unsuccessful boot attempts.
[15:23:20][I][application:053]: Running through setup()...
[15:23:20][D][binary_sensor:027]: 'Lounge Room DHT Status': Sending state OFF
[15:23:20][C][sensor.dht:030]: Setting up DHT...
[15:23:20][C][wifi:029]: Setting up WiFi...
[15:23:21][D][wifi:260]: Starting scan...
```

I took the mqtt discovery config out of my original problem child D1 - the master_bed sensor (and left in **all** the filtering and sensors that were causing trouble originally) - it connected to wifi and it did not boot loop and worked as expected.

I just noticed that the mqtt broker IP address in my configs is in single quotes (the component configuration example does not have these). Removing them did not change the situation.

I do not understand how having HA mqtt discovery on prevents wifi connection if using sliding windowed averaging. This is insane.

brandond, 2019-01-19:
Just out of curiosity, what happenes if you set send_first_at: 5 on the sliding_window_moving_average filter?

brandond, 2019-01-19:
You see anything interesting if you set the log level to VERY_VERBOSE?

tomlut, 2019-01-19:
Hang on this is even odder. If I dont define mqtt discover are the defaults used?

> discovery (Optional, boolean): If Home Assistant automatic discovery should be enabled. Defaults to True.

> discovery_prefix (Optional, string): The prefix to use for Home Assistant’s MQTT discovery. Should not contain trailing slash. Defaults to homeassistant.

brandond, 2019-01-19:
Yeah, you only need to set the broker address (plus creds if required).

tomlut, 2019-01-19:
So why when I actually define the defaults is it failing?

Unless it's a case sensitive problem?

true =/= True

tomlut, 2019-01-19:
I don't believe it! That was the problem. 

Last version, `discovery: true` worked. This version it has to be `discovery: True`

Let me try one more sensor update to be sure before closing.

tomlut, 2019-01-19:
Nope. It was enabling very_verbose mode that actually fixed the problem. Ugh!

tomlut, 2019-01-19:
This config, that failed ([shown in post above](https://github.com/OttoWinter/esphomelib/issues/394#issuecomment-455754162)) when using the default logger level:

```
esphomeyaml:
  name: lounge_dht
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'WAPLO'
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.75
    gateway: 10.1.1.1
    subnet: 255.255.255.0

mqtt:
  discovery: true
  discovery_prefix: homeassistant
  broker: 10.1.1.100
  username: !secret mqtt_usr
  password: !secret mqtt_pwd

logger:
#  level: WARN

ota:
  password: !secret esp_pwd

binary_sensor:
  - platform: status
    name: "Lounge Room DHT Status"

sensor:
  - platform: wifi_signal
    name: "Lounge Room DHT WiFi Signal"
    update_interval: 15s
    filters:
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15
          send_first_at: 15

  - platform: dht
    pin: D5
    model: DHT22
    update_interval: 15s
    temperature:
      name: "Lounge Room Temperature"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 15
            send_first_at: 15
    humidity:
      name: "Lounge Room Humidity"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 15
            send_first_at: 15
```

Now successfully uploads and runs when the log level is set to very verbose:

```
Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[... 182 lines cut ...]
[18:10:31][VV][sensor.dht:167]: Data: Hum=0b0000001000000000, Temp=0b0000000011101011, Checksum=0b11101101
[18:10:31][D][sensor.dht:066]: Got Temperature=23.5°C Humidity=51.2%
[18:10:31][V][sensor.sensor:020]: 'Lounge Room Temperature': Received new state 23.500000
[18:10:31][VV][sensor.filter:022]: Filter(0x3fff5e1c)::input(23.500000)
[18:10:32][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff5e1c)::new_value(23.500000) -> 23.500000
[18:10:32][V][sensor.sensor:020]: 'Lounge Room Humidity': Received new state 51.200001
[18:10:32][VV][sensor.filter:022]: Filter(0x3fff6084)::input(51.200001)
[18:10:32][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff6084)::new_value(51.200001) -> 51.233334
[18:10:41][VV][component:134]: Running interval 'update':0 with interval=15000 last_execution=20454 (now=35460)
[18:10:41][V][sensor.sensor:020]: 'Lounge Room DHT WiFi Signal': Received new state -34.000000
[18:10:41][VV][sensor.filter:022]: Filter(0x3fff5bb4)::input(-34.000000)
[18:10:41][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff5bb4)::new_value(-34.000000) -> -35.666668
[18:10:46][VV][component:134]: Running interval 'update':0 with interval=15000 last_execution=25940 (now=40956)
[18:10:46][VV][sensor.dht:167]: Data: Hum=0b0000001000000000, Temp=0b0000000011101011, Checksum=0b11101101
[18:10:47][D][sensor.dht:066]: Got Temperature=23.5°C Humidity=51.2%
[18:10:47][V][sensor.sensor:020]: 'Lounge Room Temperature': Received new state 23.500000
[18:10:47][VV][sensor.filter:022]: Filter(0x3fff5e1c)::input(23.500000)
[18:10:47][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff5e1c)::new_value(23.500000) -> 23.500000
[18:10:47][V][sensor.sensor:020]: 'Lounge Room Humidity': Received new state 51.200001
[18:10:47][VV][sensor.filter:022]: Filter(0x3fff6084)::input(51.200001)
[18:10:47][VV][sensor.filter:071]: SlidingWindowMovingAverageFilter(0x3fff6084)::new_value(51.200001) -> 51.224998

```

I have no idea what is going on. I don't think it has anything to do with discovery or filtering. At a guess the framework for wifi on this new build could have some sort of race condition problem. But really I have no idea.

brandond, 2019-01-19:
Honestly, I'm kinda surprised that it's trying to read the DHT at the same time wifi setup is going on. Seems to me like that component shouldn't start sampling until later. Sure seems like you're running into some weird race condition with the wifi scan + sensor poll going on at the same time. Maybe @OttoWinter will have some new idea with the additional information you've collected.

Just out of curiosity, does putting the DHT on a different pin change anything? I know on the ESP32s at least, some pins are shared with internal components. Not sure if that's a possibility for ESP8266s or not.

tomlut, 2019-01-19:
If I'm being honest, at the moment it's too much hassle to change the DHT input and the fact that there was a problem with the A0 input on the other device tends to make me believe this is not the issue.

Also the fact that changing seemingly unrelated components (even just defining default options) and even altering the log level changes the fault to me says that there is a much deeper issue and changing the input could send me up another false path.

brandond, 2019-01-19:
Oh right, this is ADC. There are a BUNCH of issues with reading the ADC while wifi is active.
* https://github.com/esp8266/Arduino/issues/513
* https://github.com/esp8266/Arduino/issues/1634
* https://github.com/esp8266/Arduino/issues/2070

One of those issues links to [this page](https://github.com/krzychb/EspScopeA0/tree/master/Bravo) which says:
> If sampling is done too often, then Wi-Fi connection is lost. This happens because we are competing for access to internal ADC with ESP's routines responsible for Wi-Fi management. The same ADC is used by ESP to calibrate Wi-Fi modem. Espressif ESP8266 FAQs document dated on 2016.08, provides us with related recommendations on ADC usage. On page 18/32 it specifies that with "Wi-Fi modem normally active" you may sample ADC at 1k samples/sec.

I am guessing that the ADC is used even more extensively during initial wifi setup and scanning. This suggests that setting `fast_connect: true` in the YAML wifi settings might improve the issue but honestly, it sounds like the ADC shouldn't be used at when wifi is particularly active; even if we are sampling at far less than 1khz there's still likely to be a lot of noise in the reading.

tomlut, 2019-01-19:
Is an ADC call every 4 seconds too often though? The WDT seems to time out a lot quicker than that when things go pear shaped.

Or more importantly, why does leaving the default mqtt discovery parameters alone alleviate the problem (with the same sample frequency)?

I've kind of found a work around - don't define default mqtt discovery parameters - that seems to be working across every device I try it on. I've tried other boards that have I2C sensors rather than A0 or DHT22 sensors. All work as long as I just let the mqtt discovery defaults define themselves.

tomlut, 2019-01-19:
> c) In this comment the error is appearing without the ADC, that is also unexpected.

That's because that particular device does not have an ADC sensor (unless the ADC is being used internally to measure the RSSI?).

> And then comes the part that I cannot replicate this. I tried all of the configs above with a bunch of my own devices (nodemcu, wemos d1, sonoff, direct ESP8266 chip, etc) and could not replicate the behavior. 

Did you include these config lines:

```
mqtt:
  discovery: true
  discovery_prefix: homeassistant
```

It's the common factor across all my D1 minis. Including these (superfluous) config parameters causes the boot looping for me (unless logging level is set to very verbose, then the config is not an issue).  It does not matter whatr sensors I'm using, ADC, I2C, DTH, these lines seem to be the common factor in runtime failure.

Either way, thank you for taking the time to try to replicate the issue.

I may try setting discovery false when I get my next batch of D1 minis just see what happens. I learned today the perrils of using already discovered devices for testing. It was quite a chore to delete the duplicates from my HA integrations.

I'll close this for now as there is nothing concrete we can do, no rhyme or reason as to why it is happening and I have a workaround (dont include unneeded default discovery parameters).

Might be wise to keep it in mind for any future issues that are reported though.

closed 2019-01-19 by tomlut

tomlut, 2019-01-19:
Also just FYI. I purchased all the Wemos D1 minis from the [official Aliexpress store](https://www.aliexpress.com/store/product/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/1331105_32529101036.html). These are not knock-off imitations.