esphome/esphome-core #396: All actions on a device just say "Terminated process."

nhorvath, 2019-01-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Linux htpc 4.15.0-42-generic #45-Ubuntu SMP Thu Nov 15 19:32:57 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
running a Virtualbox vm running HassOS 1.13

**ESP (ESP32/ESP8266/Board/Sonoff):**

NodeMCU

**Affected component:**

n/a

**Description of problem:**
I created a barebones device with the wizard. When I click any action (Validate, build, clean, compile, show logs, etc.) the popup appears with a single black line and says "Terminated process." in a black hover in the top right. Nothing proceeds. The only logs generated are http 400 (see below).
The only action that works is Edit.

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: Geothermal Monitor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'horvath'
  password: 'redacted'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'redacted'

ota:
  password: 'redacted'

```

**Traceback (if applicable):**

```
n/a
```

**Additional information:**
Logs from hassio addon page:
```
INFO 200 GET /edit?configuration=Geothermal%20Monitor.yaml (0.0.0.0) 3.89ms
INFO 200 GET / (0.0.0.0) 0.87ms
WARNING 400 GET /validate (0.0.0.0) 1.67ms
WARNING 400 GET /clean (0.0.0.0) 1.60ms
WARNING 400 GET /compile (0.0.0.0) 3.48ms
WARNING 400 GET /clean-mqtt (0.0.0.0) 2.04ms
INFO 304 GET / (0.0.0.0) 1.69ms
WARNING 400 GET /compile (0.0.0.0) 0.25ms
WARNING 400 GET /clean-mqtt (0.0.0.0) 2.05ms
WARNING 400 GET /validate (0.0.0.0) 2.00ms
```

OttoWinter, 2019-01-15:
What browser is it? I think the logs are suggesting the websocket connection is not created properly.

nhorvath, 2019-01-15:
I didn't realize you were using websockets. That was it. I enabled websocket forwarding in my nginx proxy and all is well. Thanks!

closed 2019-01-15 by nhorvath

bk8190, 2019-02-09:
I had the same problem and the nginx proxy was behind it too :)

nhorvath, 2019-02-09:
Here is my nginx proxy section for future people who might have the same issue: 
```
proxy_set_header Host $http_host;
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Forwarded-Proto $scheme;
proxy_http_version 1.1;
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection $connection_upgrade;
proxy_redirect off;
proxy_ssl_verify off;
```

hpirila, 2019-06-24:
I have same problem. Works fine with direct connection but not via nginx proxied connection. I tried to modify my proxy as written below but no luck. Can you share your entire nginx configuration?