esphome/esphome-core #401: Default 'ON' time /  Reset delay timer

meijerwynand, 2019-01-16:
**Is your feature request related to a problem/use-case? Please describe.**
Feature request:

Use case scenario

1. I would like my lights to go off if there is no movement detected for 10 minutes.
2. I can accomplish this using ```delay: 10m```
3. I need to wait for the 10 minutes to pass, lights go off, activate the motion sensor to get another 10 minutes of light. _(this get scary in the kitchen busy with a knife)_

**Describe the solution you'd like**

Will it be possible to define an optional time/delay reset on a device?

1. I would like my lights to go off if there is no movement detected for 10 minutes.
2. Once motion is detected and a time/delay reset value is set, the delay value is reset to the given value.
3. This means if motion was detected 8 minutes into the current 10 minute delay, the light delay/timer will revert back to 10 minutes.
4. Only if there was truly no motion for 10 minutes the lights will turn off

```yaml 
...
light:
  - platform: binary
    name: "esp_xxx_gpio12"
    id: "esp_xxx_gpio12"
    output: output_esp_xxx_gpio12

output:
  - platform: gpio
    id: "output_esp_xxx_gpio12"
    pin: 12
    delay_reset: 10m

...
```

Thanks

meijerwynand, 2019-01-22:
Greetings, 

It appears this is done via home assistant.

This is roughly what was done to achieve the the solution.

### Scenario

_this is for HA 0.85.1_

- When it's dark and there is motion detected
- Set a timer
- Turn on device (lights)
- Once timer has lapsed / completed, turn off device (lights)
- Any time during running timer motion is detected, reset the timer

### Configurations
_this is a split configuration_

configuration.yaml
```yaml
...
automation: !include_dir_list automation
timer: !include timer/all_timers.yaml
...
```

timer/all_timers.yaml
```yaml
...
## this sets the default duration of the timer
esp_lounge_gpio5:
  duration: '00:05:00'
...
```

automation/lounge/lights_on.yaml
```yaml
id: 'motion_lounge_lights_on'
alias: Motion in lounge detected, lights on

## triggered by the motion sensor / pir
trigger:
- entity_id: binary_sensor.esp_office_d2
  platform: state
  to: 'on'

## run only when dark
condition:
  condition: or
  conditions:
    - condition: sun
      after: sunset
    - condition: sun
      before: sunrise

action:
## turn on the lights
- service: homeassistant.turn_on
  entity_id: switch.esp_lounge_gpio5

## each time 'timer.start' is run, it resets the timer
- service: timer.start
  entity_id: timer.esp_lounge_gpio5
```

automation/lounge/lights_off.yaml
```yaml
id: 'motion_lounge_lights_off'
alias: No motion in lounge detected, lights off

## when the 'timer.finished' event occurs, switch off
trigger:
- platform: event
  event_type: timer.finished
  event_data:
    entity_id: timer.esp_lounge_gpio5

action:
- service: homeassistant.turn_off
  entity_id: switch.esp_lounge_gpio5
```

### Sources

There are some issues with current documentation and the workings of timer.*. It is documented there is a timer.reset function, however it does not appear to be present in 0.85.1 anymore when checking the services, and the timer.start does the resetting. 

- [Github](https://github.com/home-assistant/home-assistant/issues/12013)
- [Home Assistant documentation](https://www.home-assistant.io/components/timer/)

closed 2019-01-22 by meijerwynand

T3m3z, 2019-05-04:
Your solution is probably better and Home Assistant offers huge flexibility. Nevertheless, I wanted to add here a possible ESPHome automation which can achieve the desired functionality (reset timer every time there is movement). My setup has one PIR sensor on pin D7 on my ESP8266 and addressable led strip on D4. Template binary sensor is used to generate delayed on/off signal using filters (in this case 1min = 60000ms). 

`binary_sensor:
  - platform: gpio
    pin: D7
    name: "WC Motion"
    device_class: motion
    id: wc_motion
  - platform: template
    name: "WC motion delayed"
    lambda: |-
      return id(wc_motion).state;
    filters:
      - delayed_off: 60000ms # Sensor goes to OFF state when this time has passed after last true ON
    on_press:
      then:
        - light.turn_on:
            id: wc_light
            brightness: 100%
            red: 100%
            blue: 100%
            green: 100%
    on_release:
      then:
        - light.turn_off: wc_light

light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: D4
    num_leds: 17
    rgb_order: BRG
    name: "WC Light"
    id: wc_light`