esphome/esphome-core #403: Can not use previous version

tomlut, 2019-01-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 0.85.1, Pi 3B
**ESP (ESP32/ESP8266/Board/Sonoff):**

Not relevant
**Affected component:**

The esphomeyaml addon

**Description of problem:**

I tried to use the previous version by addding the following (as per the instructions in the addon) to my addon config, saving and restarting the addon:

```
{
  "ssl": false,
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem",
  "port": 6052,
  "esphomeyaml_version": "v1.9.0"
}
```

However this just results in:

502 Bad Gateway
nginx/1.14.0 (Ubuntu)

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2019-01-17:
Please read the documentation for that option. Usually the versions must match to guarantee operation - that is also the case here.

In 1.10, a new option was added so that the dashboard could be started on a UNIX socket, which makes specifying versions before 1.10.0 impossible with this version. Future versions (unless something else is changed) will be usable though.

closed 2019-01-17 by OttoWinter

tomlut, 2019-01-17:
Ok, is there any way to revert to an earlier version of the Hassio addon?

OttoWinter, 2019-01-17:
Not until Hass.io adds such an option, no.

tomlut, 2019-01-17:
Right-o. Thanks.

tomlut, 2019-01-17:
Thought I'd ound a method to downgrade. Partial snapshot restore. But the web front end is not reachable.  Log:

```
nginx: [emerg] host not found in "default_server" of the "listen" directive in /etc/nginx/nginx.conf:27
[10:15:33+1100] INFO ----> Starting NGINX...
```

OttoWinter, 2019-01-18:
Snapshots don't restore addon images (only their data). So sorry, not possible :/

tomlut, 2019-01-18:
Ah, right. Thanks for the explanation.