esphome/esphome-core #406: rising/falling edge detect on hlw8012

quazzie, 2019-01-17:
**Is your feature request related to a problem/use-case? Please describe.**

As i've read to support BL0937 or HJL-01 to hlw8012 energy monitor you should use falling edge detection. They also have inverted sel pin but that is supported already with inverted: True on sel_pin.

Theses chips are in for lots of products. 
// * Blitzwolf (https://www.amazon.es/Inteligente-Temporización-Dispositivos-Cualquier-BlitzWolf/dp/B07BMQP142)
// * HomeCube (https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG)
// * Coosa (https://www.amazon.com/COOSA-Monitoring-Function-Campatible-Assiatant/dp/B0788W9TDR)
// * Gosund (http://www.gosund.com/?m=content&c=index&a=show&catid=6&id=5)
// * Ablue (https://www.amazon.de/Intelligente-Steckdose-Ablue-Funktioniert-Assistant/dp/B076DRFRZC)

**Describe the solution you'd like**

Setting to select rising/falling edge on pulse counters for hlw8012

**Additional context**

OttoWinter, 2019-01-17:
Could you specify further what you mean by that?

Also, I *think* the duty cycle was 50% in the datasheet when I implemented the integration. So what would be the difference in changing to falling edge detection?

quazzie, 2019-01-18:
Hmm i was reading around trying to figure out if i could use esphomelib and saw:
https://github.com/xoseperez/espurna/issues/737#issuecomment-398878301

But since i've tried it and got it working, i think..
```
  - platform: hlw8012
    sel_pin: 
      number: GPIO12
      inverted: True
    cf_pin: GPIO05
    cf1_pin: GPIO14
    current_resistor: 0.0025
    voltage_divider: 840
    current:
      name: "current"
      unit_of_measurement: A
    voltage:
      name: "voltage"
      unit_of_measurement: V
    power:
      name: "power"
      id: my_power
      unit_of_measurement: W
    change_mode_every: 1
    update_interval: 10s
```

The shunt resistor on the device is an R001 same as Pow R1 but it was reporting wrong values with that value so had to tweak it to 0.0025 to get what i think is somewhat accurate and
i couldn't figure out the upstream/downstream values so i just tweaked voltage_divider until i got somewhat accurate results.

quazzie, 2019-01-25:
I've been running it now for a while and it seems fine so closing.

closed 2019-01-25 by quazzie

ei-ke, 2019-04-22:
@quazzie Do you actually get current reports from the device or only voltage and power? My device ([Teckin SP22](https://www.amazon.de/dp/B07CDCYLQ6)) shows current readings when using Tasmota but not when using ESPHome with the following configuration:
`substitutions:
  plug_name: sp22_01
  # Higher value gives lower watt readout
  current_res: "0.0023298"
  # Lower value gives lower voltage readout
  voltage_div: "756"

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: esp8285

wifi:
  ssid: ""
  password: ""

# Enable logging
logger:

# Enable Web server
web_server:
  port: 80

# Enable Home Assistant API
api:
  password: ''

ota:
  password: ''

time:
  - platform: homeassistant
    id: homeassistant_time
    
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO1
      inverted: True
    name: "${plug_name}_button"
    filters:
      - delayed_on: 100ms
      - delayed_off: 100ms
    on_press:
      then:
        - delay: 500ms
        - switch.toggle: "${plug_name}_Relay"
    
switch:
- platform: gpio
  name: "${plug_name}_Relay"
  id:  "${plug_name}_Relay"
  pin: GPIO14
  restore_mode: RESTORE_DEFAULT_OFF
  on_turn_on:
    - switch.turn_off: "${plug_name}_LED_Blue"
    - switch.turn_on: "${plug_name}_LED_Red"
  on_turn_off:
    - switch.turn_off: "${plug_name}_LED_Red"
    - switch.turn_on: "${plug_name}_LED_Blue"
    
- platform: gpio
  name: "${plug_name}_LED_Blue"
  id:  "${plug_name}_LED_Blue"
  pin: GPIO13
  inverted: True
  restore_mode: RESTORE_DEFAULT_ON
- platform: gpio
  name: "${plug_name}_LED_Red"
  id:  "${plug_name}_LED_Red"
  pin: GPIO03
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
- platform: restart
  id: esphome00_restart
  name: "ESPHome00 restart"
sensor:
- platform: hlw8012
  sel_pin:
    number: GPIO12
    inverted: True
  cf_pin: GPIO04
  cf1_pin: GPIO05
  current_resistor: ${current_res}
  voltage_divider: ${voltage_div}
  current:
    name: "${plug_name}_current"
    unit_of_measurement: A
  voltage:
    name: "${plug_name}_voltage"
    unit_of_measurement: V
  power:
    name: "${plug_name}_power"
    unit_of_measurement: W
    id: "${plug_name}_power"
  change_mode_every: 1
  update_interval: 10s
- platform: total_daily_energy
  name: "${plug_name}_Total Daily Energy"
  power_id: "${plug_name}_power"
  filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
  unit_of_measurement: kWh

#Extra sensor to keep track of plug uptime
- platform: uptime
  name: ${plug_name}_Uptime Sensor
    
text_sensor:
- platform: version
  name: "${plug_name}_version"`

OttoWinter, 2019-04-22:
@ei-ke please see calibrate_linear