esphome/esphome-core #407: Can't compile and can't load up

flemssound, 2019-01-17:
I have completle fresh installed my home assistant and esphomelib.
Now I want to compile the binarys for my node.
I have running an Raspberry pi 3 with the 32-bit version of hass.io.

This is the config of my node

esphomeyaml:
  name: test
  platform: ESP8266
  board: d1_mini
wifi:
  ssid: 'txxxxxx'
  password: 'xxxxxxxxxxxx'
#Enable Home Assistant API
api:
  password: 'xxxxx'
# Enable logging
logger:
ota:
  password: 'xxxxx'

And this is the error that is be thrown:

[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7573bdd0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)

OttoWinter, 2019-01-17:
What version of HassOS is this? I think there was an issue with older versions of HassOS that had a broken `resolv.conf` for docker.

Could you check the HassOS version, and update to the latest one if there's an update available?

flemssound, 2019-01-17:
HassOS 1.13