esphome/esphome-core #412: Deep Sleep makes home assistant list sensor as "Unknown"

rutberg, 2019-01-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hassio

**ESP (ESP32/ESP8266/Board/Sonoff):** Nodemcuv2

**Affected component:** https://esphomelib.com/esphomeyaml/components/deep_sleep.html

**Description of problem:** The suggested fix in the component readme about changing the will_message and birth_message dosen't yield the desired result. Instead the MQTT connection times out and the status in Home Assistant becomes Unknown

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: garage
  platform: ESP8266
  board: nodemcuv2

mqtt:
  broker: 192.168.1.159
  birth_message:
  will_message:
  on_message:
    topic: garage/ota_mode
    payload: "ON"
    then:
      - deep_sleep.prevent: deep_sleep_1
      - logger.log: "OTA_mode enabled, halting deep sleep"

wifi:
  ssid: 'WIFISSID'
  password: 'WIFIPSWD'
  manual_ip:
    static_ip: 192.168.1.140
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:
 level: INFO
# Enable Home Assistant API
api:
  password: 'PSWD'
ota:
  password: 'PSWD'
sensor:
  - platform: dht
    model: DHT22
    pin: D5
    temperature:
      name: "Garage Temperature"
    humidity:
      name: "Garage Humidity"
    update_interval: 8s

deep_sleep:
  id: deep_sleep_1
  run_duration: 10s
  sleep_duration: 60s

```

**Traceback (if applicable):**

```

```

**Additional information:**
I've also tried the mentioned fix; to set the keep alive to a number greater than the sleep duration, but home assistant still drops the sensor during sleep.
I'm running MQTT Server & Web client component as MQTT broker.

rutberg, 2019-01-18:
To clarify, I've followed the instructions you mentioned and set those values as indicated above in the YAML entry. However, Home Assistant still reports "Unknown" this is also reported in the comments to the component on esphomelib.com

OttoWinter, 2019-01-18:
The documentation there is mostly for binary sensors, for sensors you additionally need to disable **expire_after**.

I've made #417 which makes that the default if deep sleep is enabled.

PR #417 (2019-01-19): Default expire after to 0 with deep sleep
## Description:

**Related issue (if applicable):** fixes #412

**Pull request in [esphomeyaml](https://github.com/OttoWinter/esphomeyaml) with YAML changes (if applicable):** OttoWinter/esphomeyaml#<esphomeyaml PR number goes here>
**Pull request in [esphomedocs](https://github.com/OttoWinter/esphomedocs) with documentation (if applicable):** OttoWinter/esphomedocs#<esphomedocs PR number goes here>

## Checklist:
  - [ ] The code change is tested and works locally.

If user exposed functionality or configuration variables are added/changed:
  - [ ] Do [... cut]
closed 2019-01-19 by OttoWinter