esphome/esphome-core #416: Wifi connection issue with Sonoff Basic with light component

timmmmmmmmm, 2019-01-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**
Sonoff Basic

**Affected component:**
Wifi functionality

**Description of problem:**
I'm trying to flash an Sonoff Basic with a firmware that makes the relay shows up in Home Assistant as a light component. I defined an output and consequently link the light component to it. The code runs fine without the light component defined, but once I add it . The moment I include the light component in the code the ESP8266 isn't able to detect any wifi networks on boot. The serial logs repeatingly:
```
[0;36m[D][wifi:260]: Starting scan...[0m
[0;36m[D][wifi:275]: Found networks:[0m
[0;36m[D][wifi:277]:   No network found![0m
```
After keeping this running for a while (5 minutes) it finally detects (only) my network and connects, but disconnects right after. When I don't include the light component it lists all networks around me and stays connected to my network without a problem. Did I configure something wrong or did I encounter a bug here?

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: gpio
    id: sonoff_relay
    pin: GPIO12

light:
  - platform: binary
    name: "Moonlight"
    output: sonoff_relay
```

**Traceback (if applicable):**
n/a

```

```

**Additional information:**
n/a

OttoWinter, 2019-01-18:
I don't think a software bug would manifest itself in such a way. If there were a bug, I would expect it to crash the ESP, not make wifi issues.

What _can_ be the case here is that the relay is adding noise to the WiFi signal; I've seen this a couple of times before with custom NodeMCU setups with relays.

Could you try enabling the WiFi `fast_connect` mode? I *think* maybe that could have something to do with that.

timmmmmmmmm, 2019-01-18:
> Could you try enabling the WiFi `fast_connect` mode? I _think_ maybe that could have something to do with that.

This fixed the issue. Thanks for the quick reply.

closed 2019-01-18 by timmmmmmmmm