esphome/esphome-core #418: Switch as light

9guiajkbk2, 2019-01-18:
**What new integration would you wish to have?**

Using switches as lights. Home assistant already provides for [such functionality](https://www.home-assistant.io/components/light.switch/) but it would be even nicer if this was part of Esphomelib. Especially if you have lights behind Sonoff switches. 

**If possible, provide a link to an existing library for the integration:**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context:**

brandond, 2019-01-18:
Why don't you make a light then, instead of a switch?

```yaml
output:
  - platform: gpio
    pin: D1
    id: gpio_d1

light:
  - platform: binary
    name: "Desk Lamp"
    output: gpio_d1
```

TheJulianJES, 2019-01-18:
I'm guessing because you can't use this solution for template switches.

brandond, 2019-01-18:
If you're doing it with a template switch on the esphomelib side you should be able to control either the gpio or the light from automation lambdas.

OttoWinter, 2019-01-19:
Agree with @brandond here. This is entirely possible already. Closing

closed 2019-01-19 by OttoWinter

9guiajkbk2, 2019-01-19:
> Why don't you make a light then, instead of a switch? I haven't tested this myself, but it would give you both a light and a switch on the same GPIO pin.
> 
> ```yaml
> output:
>   - platform: gpio
>     pin: D1
>     id: gpio_d1
> 
> light:
>   - platform: binary
>     name: "Desk Lamp"
>     output: gpio_d1
> 
> switch:
>   - platform: output
>     name: "Desk Lamp Switch"
>     output: gpio_d1
> ```

Ah, thanks for pointing that out :)

NoNaMoe, 2019-05-24:
I've come across a use case where this may be useful, atm i am using a template switch, but would prefer to expose as a light:
```
output:
  - platform: gpio
    pin: 12
    id: relay
    
  - platform: gpio
    pin: 13
    inverted: yes
    id: led

switch:
  - platform: template
    name: "Big Lamp"
    id: lamp
    turn_on_action:
      - output.turn_on: relay
      - output.turn_on: led
    turn_off_action:
      - output.turn_off: relay
      - output.turn_off: led
```

I can't really see a neat way to do this using lambdas... currently light and output do not have triggers so would have to create a binary_sensor polling the light or gpios state. This would just be a duplicate of what we already know (and a busy loop?).

Perhaps adding those triggers, or a template light would be alternative solution/feature request?

brandond, 2019-05-24:
What are you not getting out of HA by setting this up as a binary switch instead of a light?