esphome/esphome-core #420: [New integration] H801 Wifi LED control

lichtamberg, 2019-01-18:
**What new integration would you wish to have?**
H801 Wifi LED control

**If possible, provide a link to an existing library for the integration:**
Tasmota supports this module already: 
https://github.com/arendst/Sonoff-Tasmota/wiki/H801

http://tinkerman.cat/closer-look-h801-led-wifi-controller/

**Please describe your use case for this integration and alternatives you've tried:**
I would like to power my leds with this module and add it to my homeassistant server.

**Additional context:**

brandond, 2019-01-18:
This doesn't seem like it would need any specific support - what are you looking for?

The page you linked even gives you the pins you'd need to configure in the [RGBWW Light](https://esphomelib.com/esphomeyaml/components/light/rgbww.html) component:

```
    #define RGBW_RED_PIN            15
    #define RGBW_GREEN_PIN          13
    #define RGBW_BLUE_PIN           12
    #define RGBW_WHITE_PIN          14
    #define RGBW_WHITE2_PIN         4
```

```yaml
output:
  - platform: gpio
    pin: 4
    id: gpio_4
  - platform: gpio
    pin: 12
    id: gpio_12
  - platform: gpio
    pin: 13
    id: gpio_13
  - platform: gpio
    pin: 14
    id: gpio_14
  - platform: gpio
    pin: 15
    id: gpio_15

light:
  - platform: rgbww
    name: "H801 Light"
    red: gpio_15
    green: gpio_13
    blue: gpio_12
    cold_white: gpio_14
    warm_white: gpio_4
    cold_white_color_temperature: 153 mireds
    warm_white_color_temperature: 500 mireds
```

lichtamberg, 2019-01-18:
Oh cool! Sorry, I'm quite new to this - thank you so much!

closed 2019-01-18 by lichtamberg