esphome/esphome-core #429: MQTT Causing A boot loop

agengler11, 2019-01-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
Boot Looping /  API connection Errors

**Problem-relevant YAML-configuration entries:**
esphomeyaml:
  name: garage_door
  platform: ESP8266
  board: d1_mini
  
wifi:
  ssid: 'FalconPP'
  password: ''
  manual_ip:
    static_ip: 192.168.1.176
    gateway: 192.168.1.1
    subnet: 255.255.255.0

mqtt:
  broker: '192.168.1.253'
  username: ''
  password: ''
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ''

ota:
  password: ''

**Traceback (if applicable):**

```

```

**Additional information:**
Serial Data Above / However this may be the important data.

Panic /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:103 __yield

agengler11, 2019-01-20:
`[0;36m[D][api:531]: Client 'esphomeyaml v1.10.1 (192.168.1.253)' connected successfully![0m

Panic /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:103 __yield

ctx: sys 
sp: 3fffeab0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffec60:  00000000 00000000 4010670a 96041893  
3fffec70:  00000000 00000000 3fff2b74 402131f8  
3fffec80:  00000000 00000000 3fff2b74 402145e0  
3fffec90:  3ffe7030 00fe8304 00000211 40219c39  
3fffeca0:  00000011 00000001 3fff252c 00000001  
3fffecb0:  3fff2ea0 00000000 00000072 40225cf0  
3fffecc0:  0000005a 40225cf0 00000284 4020ffcd  
3fffecd0:  3fff2b58 3fff252c 3fff2aec 40209895  
3fffece0:  00000000 00000000 00000000 3ffe8967  
3fffecf0:  0000005d 0000005d 3fff17cc 4020ebf0  
3fffed00:  3fff2b74 3fff252c 0000005d 00000000  
3fffed10:  00000000 00000001 3fff35dc 40203770  
3fffed20:  3fff252c 3fffedc0 3fff17cc 3ffe8967  
3fffed30:  0000005d 3fff252c 3fff2aec 40209969  
3fffed40:  0000005d 3fff37b4 3fff37c4 40208d88  
3fffed50:  00000004 00000000 4bc6a7f0 3fff252c  
3fffed60:  00000003 3fff3cf4 4010670a 3fff3dbc  
3fffed70:  00000000 00000001 3fff3dbc 40208bfd  
3fffed80:  3fffedd0 3fffedc0 0000000c 402131f8  
3fffed90:  3fffedd0 3fffedc0 0000000c 40203138  
3fffeda0:  3fffedb0 00000003 0000000f 40208c20  
3fffedb0:  3fffedd0 3fffedc0 0000000c 3fff3f70  
3fffedc0:  00000000 0000000d 00000002 3ffe8967  
3fffedd0:  00000213 3fff4308 00000001 3fff3dbc  
3fffede0:  00000000 00000001 3fff3dbc 40203388  
3fffedf0:  3fff3e24 00000000 00000100 3fff3d00  
3fffee00:  00000100 40216e60 3fff3dbc 40203a25  
3fffee10:  40216e60 3fff4340 40102242 00000100  
3fffee20:  7fffff01 3fff3ab1 00000000 40203cbc  
3fffee30:  7fffff01 00000003 000031ee 3fff1b98  
3fffee40:  3fff3ab2 00000003 4bc6a7f0 00000000  
3fffee50:  3fffc278 40101fdc 4010670a 3fff3d48  
3fffee60:  00000001 3fff3a5c 3fff3cf4 40203cf5  
3fffee70:  3fff422c 3fff3a5c 3fff3cf4 402135e1  
3fffee80:  3fff422c 00000001 00000020 00000000  
3fffee90:  00000004 00000063 00000463 3fff1b98  
3fffeea0:  3fff422c 3fff1b7c 3fff1b78 4021362c  
3fffeeb0:  3fff422c 3fff1b7c 3fff1b7d 40221518  
3fffeec0:  b001a8c0 00000000 00000000 00000000  
3fffeed0:  00000000 40225cf0 00000058 3fff3a7a  
3fffeee0:  3fff1de4 3fff3a5c 3fff1ecc 4022547d  
3fffeef0:  00000014 00000489 00000489 3fff1de4  
3fffef00:  3fffdc80 3fff238c 3fff4334 3fff23d4  
3fffef10:  00000008 3fff1de4 3fff3a5c 4021e910  
3fffef20:  3fffdc80 3fff238c 3fff4334 4021e74b  
3fffef30:  4024180e 3fff238c 3fff4334 4024181f  
3fffef40:  3fff3a6c 3fff3a5c 00000000 3fffdcc0  
3fffef50:  402363e7 00000000 3fff4334 4023d027  
3fffef60:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffef70:  40000e19 40001878 00000002 0a98063c  
3fffef80:  3fffff10 aa55aa55 000000a4 4010441c  
3fffef90:  40104422 00000002 0a98063c 3ffea8b2  
3fffefa0:  4010000d 3ffea8b2 3ffea8b2 3ffea8b2  
3fffefb0:  40100764 3fffef4c 40100705 3fffff48  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
[... 232 lines cut ...]
3ffffec0:  00000042 3fff252c 3fff2aec 40209969  
3ffffed0:  00000042 3fff37b4 3fff37c4 40208d88  
3ffffee0:  00000003 00000000 00000000 3fff252c  
3ffffef0:  00000000 00000001 40217408 3fff32dc  
3fffff00:  3fff321c 00003017 3fff16fc 40208bfd  
3fffff10:  00000000 4bc6a7f0 000031df 3fff2aec  
3fffff20:  00000000 4bc6a7f0 000031df 3fff32dc  
3fffff30:  40104d38 00277be6 3fff18e0 00000000  
3fffff40:  3fff1110 3fff18e0 3ffe856c 3fff18e0  
3fffff50:  3fff321c 3fff32dc 4020ff04 3fffefb0  
3fffff60:  3fff18e0 3fff321c 00000010 4020162b  
3fffff70:  3fff321c 000031df 3fff16fc 40204be7  
3fffff80:  00000000 3fff321c 3fff306c feefeffe  
3fffff90:  00000001 3fff3314 3fff324c 3fff188c  
3fffffa0:  3fffdad0 00000000 3fff1884 40202728  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld`

AlexDanault, 2019-01-20:
Having both mqtt and api enabled at the same time does this on some devices, I have the same results on an ESP8266, but this is clearly documented in the docs that having both enabled is unstable for the moment.