esphome/esphome-core #437: Unable to connect to wifi with sensor in configuration

DonHugo, 2019-01-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP8266/D1_min

**Affected component:**
Wifi

**Description of problem:**
if i have a configuration with only wifi and mqtt settings it works, but as soon as i add any sensor it cant connect to the wifi anymore.

first logentry is from executing first yaml configuration and second logentry is from executing second yaml configuration

**Problem-relevant YAML-configuration entries:**
```yaml
#Does not work
esphomeyaml:
  name: varmevaxlare
  platform: ESP8266
  board: d1_mini
  arduino_version: latest
  esphomelib_version: latest
  

wifi:
  ssid: 'Parlan'
  password: 'password'
  hostname: 'D1_mini'

# Enable logging
logger:
  level: DEBUG

mqtt:
  broker: '192.168.0.111'
  username: 'name'
  password: 'password'
  topic_prefix: 'test/esphome/1'
  
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 10s

```

```yaml
#This one works
esphomeyaml:
  name: varmevaxlare
  platform: ESP8266
  board: d1_mini
  arduino_version: latest
  esphomelib_version: latest
  

wifi:
  ssid: 'Parlan'
  password: 'password'
  hostname: 'D1_mini'

# Enable logging
logger:
  level: DEBUG

mqtt:
  broker: '192.168.0.111'
  username: 'name'
  password: 'password'
  topic_prefix: 'test/esphome/1'
```
**Traceback (if applicable):**

```
Checking size .pioenvs/varmevaxlare/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  45.2% (used 37040 bytes from 81920 bytes)
PROGRAM: [===       ]  31.2% (used 325884 bytes from 1044464 bytes)
======================== [SUCCESS] Took 308.36 seconds ========================
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB1 write_flash 0x0 /config/esphomeyaml/varmevaxlare/.pioenvs/varmevaxlare/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB1
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: b4:e6:2d:27:01:b0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 330032 bytes to 230411...
Wrote 330032 bytes (230411 compressed) at 0x00000000 in 20.8 seconds (effective 126.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB1 with baud rate 115200
[10:46:48]�b��oo�lgg���cx��drd{lp�n��l��cn�|l��c��nn�l��d`�\033ngl`g{���nc�d{��nc�d�ccc���l`��o�[I][logger:071]: Log initialized
[10:46:48][I][application:053]: Running through setup()...
[10:46:48][C][wifi:029]: Setting up WiFi...
[10:46:48][D][wifi:260]: Starting scan...
[10:46:57]
[10:46:57] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[10:46:57]
[10:46:57]wdt reset
[10:46:57]load 0x4010f000, len 1384, room 16 
[10:46:57]tail 8
[10:46:57]chksum 0x2d
[10:46:57]csum 0x2d
[10:46:57]vbb28d4a3
[10:46:57]~ld
[10:46:57][I][logger:071]: Log initialized
[10:46:57][I][application:053]: Running through setup()...
[10:46:57][C][wifi:029]: Setting up WiFi...
[10:46:57][D][wifi:260]: Starting scan...
```

```
[17:48:59][I][application:053]: Running through setup()...
[17:48:59][C][wifi:029]: Setting up WiFi...
[17:48:59][D][wifi:260]: Starting scan...
[17:49:05][D][wifi:275]: Found networks:
[17:49:05][I][wifi:308]: - 'Parlan' (40:16:7E:EB:46:48) ▂▄▆█
[17:49:05][D][wifi:309]:     Channel: 11
[17:49:05][D][wifi:310]:     RSSI: -32 dB
[17:49:05][D][wifi:312]: - '' (FA:8F:CA:63:F0:AF) ▂▄▆█
[17:49:05][I][wifi:179]: WiFi Connecting to 'Parlan'...
[17:49:06][I][wifi:349]: WiFi connected!
[17:49:06][C][wifi:240]:   SSID: 'Parlan'
[17:49:06][C][wifi:241]:   IP Address: 192.168.0.192
[17:49:06][C][wifi:243]:   BSSID: 40:16:7E:EB:46:48
[17:49:06][C][wifi:245]:   Hostname: 'D1_mini'
[17:49:06][C][wifi:250]:   Signal strength: -36 dB ▂▄▆█
[17:49:06][C][wifi:251]:   Channel: 11
[17:49:06][C][wifi:252]:   Subnet: 255.255.255.0
[17:49:06][C][wifi:253]:   Gateway: 192.168.0.1
[17:49:06][C][wifi:254]:   DNS1: 192.168.0.129
[17:49:06][C][wifi:255]:   DNS2: 192.168.0.1
[17:49:06][C][mqtt.client:024]: Setting up MQTT...
[17:49:06][I][mqtt.client:170]: Connecting to MQTT...
[17:49:06][I][mqtt.client:209]: MQTT Connected!
[17:49:06][I][application:089]: setup() finished successfully!
[17:49:06][I][application:097]: You're running esphomelib v1.10.1 compiled on Jan 21 2019, 17:43:49
[17:49:06][C][wifi:341]: WiFi:
[17:49:06][C][wifi:240]:   SSID: 'Parlan'
[17:49:06][C][wifi:241]:   IP Address: 192.168.0.192
[17:49:06][C][wifi:243]:   BSSID: 40:16:7E:EB:46:48
[17:49:06][C][wifi:245]:   Hostname: 'D1_mini'
[17:49:06][C][wifi:250]:   Signal strength: -36 dB ▂▄▆█
[17:49:06][C][wifi:251]:   Channel: 11
[17:49:06][C][wifi:252]:   Subnet: 255.255.255.0
[17:49:06][C][wifi:253]:   Gateway: 192.168.0.1
[17:49:06][C][wifi:254]:   DNS1: 192.168.0.129
[17:49:06][C][wifi:255]:   DNS2: 192.168.0.1
[17:49:06][C][logger:099]: Logger:
[17:49:06][C][logger:100]:   Level: DEBUG
[17:49:06][C][logger:101]:   Log Baud Rate: 115200
[17:49:06][C][mqtt.client:058]: MQTT:
[17:49:06][C][mqtt.client:060]:   Server Address: 192.168.0.111:1883 (192.168.0.111)
[17:49:06][C][mqtt.client:061]:   Username: 'hugo'
[17:49:06][C][mqtt.client:062]:   Client ID: 'varmevaxlare-b4e62d2701b0'
[17:49:06][C][mqtt.client:064]:   Discovery prefix: 'homeassistant'
[17:49:06][C][mqtt.client:065]:   Discovery retain: YES
[17:49:06][C][mqtt.client:067]:   Topic Prefix: 'test/esphome/1'
[17:49:06][C][mqtt.client:069]:   Log Topic: 'test/esphome/1/debug'
[17:49:06][C][mqtt.client:072]:   Availability: 'test/esphome/1/status'
[17:49:06][I][application:114]: Running through first loop()
[17:49:06][I][application:141]: First loop finished successfully!
```

**Additional information:**

DonHugo, 2019-01-22:
Tested to use api instead of mqtt and then it works with  wifi_signal but not if i add ota in the configuration.

```
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 10s
```

and i can't get a dallas component to work either.
```
dallas:
  - pin: GPIO23
```

mjrider, 2019-01-22:
I'm seeing a similar issue,

i had a working setup yesterday, without uptime and wifi sensor, added them, and now it wil not connect
the serial logs tell me that it find the ssid, and then it complains that the wifi is not connectable

```
[I][wifi:179]: WiFi Connecting to 'fropnetiot'...
[W][wifi:727]: Event: Disconnected ssid='fropnetiot' bssid=00:00:00:00:00:00 reason='AP Not Found'
[W][wifi:369]: Error while connecting to network.
[W][wifi:399]: Restarting WiFi adapter...
[I][wifi:179]: WiFi Connecting to 'fropnetiot'...
[W][wifi:727]: Event: Disconnected ssid='fropnetiot' bssid=92:2A:A8:84:FA:12 reason='Auth Expired'
[W][wifi:369]: Error while connecting to network.
[W][wifi:399]: Restarting WiFi adapter...
```