esphome/esphome-core #439: Support for ACS712 Current Sensor

bgoll, 2019-01-22:
**What new integration would you wish to have?**
I would like to have the possibility to use an ACS712 current sensor with esphomelib and esphomeyaml. The sensor allows you to read the current of a connected device.

**If possible, provide a link to an existing library for the integration:**
https://github.com/rkoptev/ACS712-arduino

**Please describe your use case for this integration and alternatives you've tried:**
I personally use this sensor for automating the dust collection in my workshop. If a power tool connected to the sensor gets turned on, I turn on the dust collection and switch it off (with a small delay) if the tool gets turned off.
Currently I achieve this with normal Arduino code but I would love to be able to integrate this device with the rest of my home.
Besides my use case this sensor can also be used for very cheap power consumption measurement.

**Additional context:**
Using the sensor is quite straightforward. It outputs an analog signal which indicates the current that the connected device consumes.

bgoll, 2019-01-23:
Oh, I was not aware of that. Thank you very much! I think the issue can be closed then.

closed 2019-01-25 by OttoWinter

HFsi, 2019-06-03:
For what I see, to sense AC current, you have to take several samples, not only the value in a certain moment, am I right?
If so, the ADS1115 supports that kind of measure?