esphome/esphome-core #440: Error getting initial data: Timeout while waiting for API response!

rgabrielson11, 2019-01-23:
Home Assistant (0.85.1) regularly loses connection to device (shows unavailable).  Shows as 'on line' using esphome dashboard.

```
esphomeyaml:
  name: dishwasher
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'my'
  password: 'secret'
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.195
    gateway: 192.168.1.1
    subnet: 255.255.255.0
ota:
  safe_mode: True
  password: secret

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

uart:
  rx_pin: RX
  baud_rate: 4800

sensor:
  - platform: cse7766
    current:
      name: "Dishwasher Current"
    voltage:
      name: "Dishwasher Voltage"
    power:
      name: "Dishwasher Power"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Dishwasher Button"
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: "Dishwasher Relay"
    pin: GPIO12
    id: relay

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    name: "Dishwasher Green LED"
    output: basic_green_led

```
log from device
```
INFO Reading configuration...
INFO Starting log output from 192.168.1.195 using esphomelib API
INFO Connecting to 192.168.1.195:6053 (192.168.1.195)
INFO Successfully connected to 192.168.1.195
[02:43:27][I][application:097]: You're running esphomelib v1.10.1 compiled on Jan 19 2019, 14:38:50
[02:43:27][C][wifi:341]: WiFi:
[02:43:27][C][wifi:240]:   SSID: 'secret'
[02:43:27][C][wifi:241]:   IP Address: 192.168.1.195
[02:43:27][C][wifi:243]:   BSSID: 4C:ED:FB:9A:C1:C0
[02:43:27][C][wifi:245]:   Hostname: 'dishwasher'
[02:43:27][C][wifi:250]:   Signal strength: -53 dB ▂▄▆█
[02:43:27][C][wifi:251]:   Channel: 4
[02:43:27][C][wifi:252]:   Subnet: 255.255.255.0
[02:43:27][C][wifi:253]:   Gateway: 192.168.1.1
[02:43:27][C][wifi:254]:   DNS1: 0.0.0.0
[02:43:27][C][wifi:255]:   DNS2: 0.0.0.0
[02:43:27][C][uart:130]: UART Bus:
[02:43:27][C][uart:131]:   TX Pin: GPIO-1
[02:43:27][C][uart:132]:   RX Pin: GPIO3
[02:43:27][C][uart:133]:   Baud Rate: 4800 baud
[02:43:27][C][uart:137]:   Using software serial
[02:43:27][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'Dishwasher Button'
[02:43:27][C][binary_sensor.gpio:024]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[02:43:27][C][switch.gpio:049]: GPIO Switch 'Dishwasher Relay'
[02:43:27][C][switch.gpio:050]:   Pin: GPIO12 (Mode: OUTPUT)
[02:43:27][C][switch.gpio:066]:   Restore Mode: Restore (Default to OFF)
[02:43:27][C][output.esp8266_pwm:028]: ESP8266 PWM:
[02:43:27][C][output.esp8266_pwm:029]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[02:43:27][C][output.esp8266_pwm:030]:   Frequency: 1000.0 Hz
[02:43:27][C][logger:099]: Logger:
[02:43:27][C][logger:100]:   Level: DEBUG
[02:43:27][C][logger:101]:   Log Baud Rate: 0
[02:43:27][C][light.state:266]: Light 'Dishwasher Green LED'
[02:43:27][C][light.state:268]:   Default Transition Length: 1000 ms
[02:43:27][C][light.state:269]:   Gamma Correct: 2.80
[02:43:27][C][sensor.cse7766:182]: CSE7766:
[02:43:27][C][sensor.cse7766:183]:   Update Interval: 60000 ms
[02:43:27][C][sensor.cse7766:184]:   Voltage 'Dishwasher Voltage'
[02:43:27][C][sensor.cse7766:184]:     Unit of Measurement: 'V'
[02:43:27][C][sensor.cse7766:184]:     Accuracy Decimals: 1
[02:43:27][C][sensor.cse7766:184]:     Icon: 'mdi:flash'
[02:43:27][C][sensor.cse7766:185]:   Current 'Dishwasher Current'
[02:43:27][C][sensor.cse7766:185]:     Unit of Measurement: 'A'
[02:43:27][C][sensor.cse7766:185]:     Accuracy Decimals: 1
[02:43:27][C][sensor.cse7766:185]:     Icon: 'mdi:flash'
[02:43:27][C][sensor.cse7766:186]:   Power 'Dishwasher Power'
[02:43:27][C][sensor.cse7766:186]:     Unit of Measurement: 'W'
[02:43:27][C][sensor.cse7766:186]:     Accuracy Decimals: 1
[02:43:27][C][sensor.cse7766:186]:     Icon: 'mdi:flash'
[02:43:27][C][api:072]: API Server:
[02:43:27][C][api:073]:   Address: 192.168.1.195:6053
[02:43:27][C][ota:129]: Over-The-Air Updates:
[02:43:27][C][ota:130]:   Address: 192.168.1.195:8266
[02:43:27][C][ota:132]:   Using Password.
```

ant0n-z, 2019-01-24:
I suppose that connection to device loses regularly because poor wi-fi signal quality. I moved the device (sensor with esp8266) to another room and the connection is stable. But one question, why the link between HA and device is not reconnected until the home-assistant is rebooted.

OttoWinter, 2019-01-25:
@ant0n-z Use HA 0.86.0 please

mrmantas, 2019-01-31:
I am getting the same using HA 0.86.4. Seems to be a Wifi connection issue, keeps disconnecting for a few seconds and then reconnects again. Might have to go back to MQTT actually, since the signal is definitely quite strong in my case - other devices are working fine even further away and behind a wall

OttoWinter, 2019-02-01:
@mrmantas Seems to be a Wifi connection issue - so how should MQTT fix that?