esphome/esphome-core #444: Sonoff POW R2 continuous restarts after adding power sensor cse7766

alvarea, 2019-01-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
Hi, I have a sonoff POW R2 flashed with ESPhome 1.10.1 and connected to HA using API. It's working properly if I don't add any code of the CSE7766 Power Sensor. When I add the configuration entries described in the sensor chapter of ESPhomelib documentation, the sonoff starts disconnecting from API and restarting the switch.  If I remove the power sensor entries of the configuration and I flash it again, the behavior is normal and it remains connected during the whole day.

**Problem-relevant YAML-configuration entries:**
```yaml
sphomeyaml:
  name: esphome_pow_sw2
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'xxx'
  password: 'xxx'
  manual_ip:
    static_ip: 192.168.0.23
    gateway: 192.168.0.1
    subnet: 255.255.255.0

api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "esphome pow sw2 boton"
  - platform: status
    name: "esphome pow sw2 status"

switch:
  - platform: gpio
    name: "esphome pow sw2"
    pin: GPIO12

output:
  # Register the green LED as a dimmable output ....
  - platform: esp8266_pwm
    id: pow_sw2_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "esphome pow sw2 led"
    output: pow_sw2_green_led

# Example configuration entry
# Disable logging over USB
logger:
  baud_rate: 0

uart:
  rx_pin: RX
  baud_rate: 4800

sensor:
  - platform: cse7766
    current:
[... 114 lines cut ...]
[22:55:16][D][sensor.cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
[22:56:16][D][sensor.cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
[22:57:16][D][sensor.cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
[22:58:16][D][sensor.cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
[22:59:16][D][sensor.cse7766:147]: Got voltage=0.0V current=0.0A power=0.0W
WARNING Disconnected from API.
INFO Connecting to 192.168.0.37:6053 (192.168.0.37)
Exception in thread Thread-1836:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 1073, in run
    self.function(*self.args, **self.kwargs)
  File "/home/homeassistant/venv/local/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 136, in func
    self._fatal_error()
  File "/home/homeassistant/venv/local/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 242, in _fatal_error
    self.on_disconnect()
  File "/home/homeassistant/venv/local/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 444, in try_connect
    cli.connect()
  File "/home/homeassistant/venv/local/lib/python2.7/site-packages/esphomeyaml/api/client.py", line 198, in connect
    self._socket.settimeout(0.1)
AttributeError: 'NoneType' object has no attribute 'settimeout'
```

**Additional information:**

novikovav69, 2019-01-25:
I have a similar problem. at first it clicks. Then the connection falls off. The device hangs, the button stops working. What you need to send and how to get it?
`esphomeyaml:
  name: tt
  platform: ESP8266
  board: esp01_1m
  arduino_version: latest

wifi:
  ssid: 'Aurelia2_4'
  password: '****'
  power_save_mode: light
  fast_connect: true
  manual_ip:
    static_ip: 192.168.100.44
    gateway: 192.168.100.1
    subnet: 255.255.255.0

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: '****'

ota:
  password: '****'

switch:
  - platform: gpio
    pin: GPIO12
    name: "Living Room Torsher"
    id: relay
    
uart:
  rx_pin: RX
  baud_rate: 4800

sensor:
  - platform: total_daily_energy
    name: "Total Daily Energy"
    power_id: my_power
  - platform: cse7766
    current:
      name: "Sonoff Pow R2 Current"
    voltage:
      name: "Sonoff Pow R2 Voltage"
    power:
      name: "Sonoff Pow R2 Power"
      id: my_power
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kW

time:
  - platform: sntp
    id: my_time

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "Sonoff S20 Status"`

novikovav69, 2019-01-25:
I think this is critical. I have many such devices and they control the heating.

novikovav69, 2019-01-25:
`INFO Reading configuration...
INFO Detected timezone 'MSK' with UTC offset 3
INFO Starting log output from 192.168.100.44 using esphomelib API
INFO Connecting to 192.168.100.44:6053 (192.168.100.44)
WARNING Couldn't connect to API (Error connecting to 192.168.100.44: [Errno 113] No route to host). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.100.44:6053 (192.168.100.44)
WARNING Couldn't connect to API (Error connecting to 192.168.100.44: [Errno 113] No route to host). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.100.44:6053 (192.168.100.44)
WARNING Couldn't connect to API (Error connecting to 192.168.100.44: [Errno 113] No route to host). Trying to reconnect in 4 seconds
INFO Connecting to 192.168.100.44:6053 (192.168.100.44)
WARNING Couldn't connect to API (Error connecting to 192.168.100.44: [Errno 113] No route to host). Trying to reconnect in 8 seconds
INFO Connecting to 192.168.100.44:6053 (192.168.100.44)
WARNING Couldn't connect to API (Error connecting to 192.168.100.44: [Errno 113] No route to host). Trying to reconnect in 16 seconds`

novikovav69, 2019-01-25:
removed the sensor cse7766 settings. Let's see what will happen.

brandond, 2019-01-25:
@novikovav69 as mentioned above, you need to pull logs via USB, not API or MQTT. If it's crashing and/or not connecting to the network, you obviously can't use the network to collect the logs necessary to debug the issue.

novikovav69, 2019-01-26:
I am not a very good specialist in this area. How to remove logs via USB? Does it have to be powered? Or logs are stored on the device like file?

novikovav69, 2019-01-26:
I'll be back when there will be a stable job.