esphome/esphome-core #447: force sensitive resistor

stoneobscurity, 2019-01-26:
**What new integration would you wish to have?**

force sensitive resistor
https://www.adafruit.com/product/1075

**If possible, provide a link to an existing library for the integration:**

**Please describe your use case for this integration and alternatives you've tried:**

problem: leaving a phone plugged in after it reaches 100% is really, really bad for the battery.

here's my plan:
combine a nodemcu, a FSR, a power plug switch, and a QI charger. 

set the phone on the charger, it reads the increased weight, turns on the charger, for either a set duration, or using the battery sensor in HASS when it gets to 100%, turn off the charger. 

only reset when the phone is picked up (the weight drops) to prevent it from continuing to charge when it drops back to 99%.

**Additional context:**

Mynasru, 2019-01-26:
Please read more careful before asking for new integrations, this same issue has just been closed 20 hours ago. https://github.com/OttoWinter/esphomelib/issues/439

Mynasru, 2019-01-26:
They are both **analog sensors** having a **voltage** as output. All analog sensors can be used with the following [method](https://esphomelib.com/esphomeyaml/cookbook/temt6000.html). I hope that is helpful.

stoneobscurity, 2019-01-26:
yeah okay, i'll try it, thank you.

also, then the website needs to be updated to reflect this. a FSR needs to be added to the sensors list. there are no mention of them at all. if you're annoyed cause you keep getting questions about these types of sensors, this is why.

brandond, 2019-01-26:
I don't think Otto would be interested in documenting every chip that outputs an analog reading as supported. Stuff that needs a device-specific driver or protocol will be documented as supported, things that can be connected as a simple analog or binary sensor already have examples posted.

OttoWinter, 2019-01-27:
Yes there are hundreds (if not thousands) of sensors that output their values using analog voltage, so it's simply impossible to add docs for all of them.

I don't have such a sensor, so I can't create docs for it. Though if you want you can 100% add a cookbook entry for it like the TEMT6000 example.