esphome/esphome-core #448: Individual addressable leds in pixel strips

mvturnho, 2019-01-27:
**Is your feature request related to a problem/use-case? Please describe.**

I want to use a pixel strip to display sensor status like door open/close or temperature/pressure. 

For example I have 8 door sensors an three temperature sensors.

With one pixel strip I would like to have the first 8 leds show the sensor status. Red is open. Green is closed. 

The next 3 ĺeds for the temperature. Blue is cold, yellow, orange, red is hot.
**Describe the solution you'd like**

Is it possible to address just a number (1-n) of leds in a strip with an mqtt message?

**Additional context**

Individual addressable lights for neopixel or fastled

OttoWinter, 2019-01-27:
You can use a custom effect for that - see `lambda` effect. From there you can address all LEDs like you wish.

Closing as this issue is a duplicate.

closed 2019-01-27 by OttoWinter

mvturnho, 2019-01-30:
Thanks I will try this approach. 

And this project is really great to use. Thanks for that.

mvturnho, 2019-02-01:
@OttoWinter 
It seems I am unable to get this working.

Here is the relevant part in my yaml config
```
esphomeyaml:
  name: m5stack
  platform: ESP32
  board: nodemcu-32s
  esphomelib_version: latest

wifi:
  ssid: 'xxxxxxxx'
  password: 'xxxxxxxx'

mqtt:
#  broker: '192.168.2.xx'
  broker: '192.168.xx.xx'
  username: ''
  password: ''

# Enable logging
logger:

ota:

light:
  - platform: fastled_clockless
    id: light_fastled
    chipset: WS2812B
    pin: GPIO21
    num_leds: 8
    rgb_order: GRB
    name: 'LED Strip'
    effects:
      - addressable_lambda:
          name: statustest
          update_interval: 500ms
          lambda: |-           
            ESP_LOGD("statustest", "switch on led 1");
            fadeToBlackBy(it.leds(), it.size(), 10);
            int pos = 1;
            it.leds()[pos] = CHSV(0, 255, 150);
            
            it.schedule_show();
```

when I compile this I get:

```
src\main.cpp: In lambda function:
src\main.cpp:32:26: error: 'class esphomelib::light::AddressableLight' has no member named 'leds'
fadeToBlackBy(it.leds(), it.size(), 10);
^
src\main.cpp:34:12: error: 'class esphomelib::light::AddressableLight' has no member named 'leds'
it.leds()[pos] = CHSV(0, 255, 150);
^
src\main.cpp:36:12: error: 'class esphomelib::light::AddressableLight' has no member named 'schedule_show'
it.schedule_show();
^
```

Could you help me out on this? Maybe there is some instruction on howto go about this, 
When I open the project in platformio my generated main.cpp also gives the errors.

OttoWinter, 2019-02-01:
@mvturnho No, that's not the API. ESPHome has it's own addressable light abstraction.

```cpp
ESP_LOGD("statustest", "switch on led 1");
for (int i = 0; i < it.size(); i++) 
  it[i] = it[i].get() - 10;
it[1] = ESPHSVColor(0, 255, 100);
```

mvturnho, 2019-02-04:
@OttoWinter 
Hi, it seems I keep running into problems with this.

I started with your example above, when compiling I get the folowing error:

```
src\main.cpp: In lambda function:
src\main.cpp:33:40: error: 'ESPHSVColor' was not declared in this scope
it[1] = ESPHSVColor(0, 255, 100);
^
src\main.cpp:33:40: note: suggested alternative:
In file included from .piolibdeps\esphomelib\src/esphomelib/light/addressable_light_effect.h:9:0,
from .piolibdeps\esphomelib\src/esphomelib/application.h:52,
from src\main.cpp:3:
.piolibdeps\esphomelib\src/esphomelib/light/addressable_light.h:63:8: note:   'esphomelib::light::ESPHSVColor'
struct ESPHSVColor {
^
```

Also, how can I use the state, color and brightness that I parse with the MQTT message?

```json
{"effect":"statustest","state":"ON","brightness":255,"color":{"r":255,"g":255,"b":255}}
```

this is my main.cpp that is generated by esphometaml.

```cpp
// Auto generated code by esphomeyaml
// ========== AUTO GENERATED INCLUDE BLOCK BEGIN ===========
#include "esphomelib/application.h"
using namespace esphomelib;
// ========== AUTO GENERATED INCLUDE BLOCK END ==========="

void setup() {
  // ===== DO NOT EDIT ANYTHING BELOW THIS LINE =====
  // ========== AUTO GENERATED CODE BEGIN ===========
  App.set_name("controlpanel");
  App.set_compilation_datetime(__DATE__ ", " __TIME__);
  LogComponent *logcomponent = App.init_log(115200);
  logcomponent->set_global_log_level(ESPHOMELIB_LOG_LEVEL_DEBUG);
  WiFiComponent *wificomponent = App.init_wifi();
  WiFiAP wifiap = WiFiAP();
  wifiap.set_ssid("XXXXXXXXX");
  wifiap.set_password("XXXXXXXXXXXXXXX");
  wificomponent->add_sta(wifiap);
  OTAComponent *otacomponent = App.init_ota();
  otacomponent->set_auth_password("VERYSECURE");
  otacomponent->start_safe_mode();
  mqtt::MQTTClientComponent *mqtt_mqttclientcomponent = App.init_mqtt("192.168.2.1", 1883, "", "");
  I2CComponent *i2ccomponent = App.init_i2c(21, 22, true);
  Application::MakeFastLEDLight application_makefastledlight = App.make_fast_led_light("LED Strip");
  application_makefastledlight.fast_led->add_leds<WS2812B, 4, GRB>(8);
  light::LightState *light_fastled = application_makefastledlight.state;
  light::MQTTJSONLightComponent *light_mqttjsonlightcomponent = application_makefastledlight.mqtt;
  light_fastled->add_effects({
      new light::AddressableLambdaLightEffect("statustest", [=](light::AddressableLight & it) -> void {
        ESP_LOGD("statustest", "switch on led 1");
        for (int i = 0; i < it.size(); i++) 
          it[i] = it[i].get() - 10;
        it[1] = ESPHSVColor(0, 255, 100);
    }, 500),
  });
  // =========== AUTO GENERATED CODE END ============
  // ========= YOU CAN EDIT AFTER THIS LINE =========
  App.setup();
}

void loop() {
  App.loop();
}

```

OttoWinter, 2019-02-05:
> how can I use the state, color and brightness that I parse with the MQTT message?

Store them in a global variable (`globals:`) in a `mqtt.on_message` trigger and then use those in the effect

mvturnho, 2019-02-05:
@OttoWinter 

Hi just for feedback and reference to other users.
I left the light - efects solution and went for the mqtt on_json_message.
In this way I can set the individual leds from the json message. I added an index to address the individual leds. with state I can set them ON or OFF.

```json
{"index":"1","state":"ON","brightness":100,"color":{"r":255,"g":1,"b":0}}
```

```yaml
esphomeyaml:
  name: controlpanel
  platform: ESP32
  board: nodemcu-32s
  esphomelib_version:
    branch: dev

wifi:
  ssid: 'KELPNET_IOT'
  password: 'GROENEPAARDENSCHOENEN'
  domain : .visconti.nl
  
mqtt:
  broker: '192.168.3.100'
  port: 1883

  on_json_message:
    topic: controlpanel/status/led
    then:
        - if:
            condition:
                lambda: 'return x["state"] == "ON" ;'
            then:
                - lambda: |-
                    int length = 1000;
                    if (x.containsKey("length"))
                        length = x["length"];
                    float brightness = x["brightness"];
                    int index = 0;
                    int r = x["color"]["r"];
                    int g = x["color"]["g"];
                    int b = x["color"]["b"];
                    if (x.containsKey("index"))
                        index = x["index"];
                        ESP_LOGD("on_json_message", "LED index %d bright %f state ON",index,brightness);
                        light_fastled->turn_on();
                        light::FastLEDLightOutputComponent *fast_led =  application_makefastledlight.fast_led;
                        fast_led->get_controller()->leds()[index].setRGB(r,g,b);
                        fast_led->get_controller()->leds()[index].maximizeBrightness(brightness);
                        fast_led->schedule_show();
            else:
                - lambda: |-
                    int index = 0;
                    if (x.containsKey("index"))
                        index = x["index"];
                        id(led_index) = index;
                        ESP_LOGD("on_json_message", "LED index %d state OFF",index);
                        light::FastLEDLightOutputComponent *fast_led =  application_makefastledlight.fast_led;
                        fast_led->get_controller()->leds()[index].setRGB(0,0,0);
                        fast_led->schedule_show();
                
            

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

# Example configuration entry
i2c:
  sda: 21
  scl: 22
  scan: True

light:
  - platform: fastled_clockless
    id: light_fastled
    chipset: WS2812B
    pin: GPIO4
    num_leds: 8
    rgb_order: GRB
    name: 'LED Strip' 
```