esphome/esphome-core #460: restore_mode option doesn't restore switch state

vevsvevs, 2019-02-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io (HassOS OVA)

**ESP (ESP32/ESP8266/Board/Sonoff):**
Sonoff Basic (ESP8285)

**Affected component:**
https://esphomelib.com/esphomeyaml/components/switch/gpio.html

**Description of problem:**
`restore_mode` doesn't actually restore previous switch state, even if the option is explicitly set.

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: gpio
    name: "sonoff_kitchen_light"
    pin: GPIO12
    id: relay
    restore_mode: RESTORE_DEFAULT_OFF
```

**Traceback (if applicable):**

```

```

**Additional information:**

brandond, 2019-02-01:
It should restore it across reboots, but unfortunately ESP8266 does not have any NVRAM storage to persist the state to, like the ESP32 does. You can fake it with SPI flash, but this runs the risk of overwearing it. On ESP8266, esphome simply stores the state to RAM, which is cleared if power is lost.

vevsvevs, 2019-02-01:
Ah, i didn't notice that ESP82xx doesn't have EEPROM :( Then what is the best solution for such a task (is it possible at all?): relay turns-on by default when it powered but keeps turned-off in case of Wi-Fi connection loose? For now, if i had set `ALWAYS_ON` - every connection drop causes relay turning on.
Thank you for help!

brandond, 2019-02-01:
The RESTORE_DEFAULT_x setting should work across WiFi drops, it's just power loss that will wipe them. I think there have been some fixes to state restoration made in dev; you might try that if you haven't yet. For remembering across power loss there's no great solution, you just have to pick a suitable default or restore from an input (such as a switch connected to GPIO) on boot.

vevsvevs, 2019-02-01:
Doh.. I've tested only `ALWAYS_ON`, stupid me :) `RESTORE_DEFAULT_ON` fits my needs perfectly, thank you!

closed 2019-02-01 by vevsvevs