esphome/esphome-core #462: wifi domain config option is ignored, causing name resolution to fail for domains != .local

xlfe, 2019-02-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Tried with both Hass.io and plain Docker

**ESP (ESP32/ESP8266/Board/Sonoff):**

Primarily ESP32 but also seems to apply to ESP8266

**Affected component:**

Wifi - https://esphomelib.com/esphomeyaml/components/wifi.html?highlight=wifi

**Description of problem:**

The domain part of the device's hostname is never set (and therefore always defaults to ".local") - this causes problems for mDNS and node IP resolution from node name in LANs where the domain is not .local

Even when the "domain: " config entry is set under the wifi: section, the domain is set to .local

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  hostname: $devicename
  domain: .lan
```

**Traceback (if applicable):**

```

```

**Additional information:**

For example, I have a ESP32 node hostname=test

I should be able to ping it using test.lan, but it only responds to test.local

Looking through the code, it seems like CONF_HOSTNAME is only used for get_upload_host

https://github.com/esphome/ESPHome/blob/98bdfc821ed6084785d6e0d4397a25a18bb48431/esphomeyaml/components/wifi.py#L154

You might be assuming that if a domain is not specified, esp-idf will get it from DHCP or something, but I don't think that's the case. Looking through the esp-idf code, it has a default of .local if the domain is not specified

https://github.com/espressif/esp-idf/blob/58b53942661bd40567c988eaafe86ceb519bde04/components/mdns/mdns.c#L25
https://github.com/esphome/ESPHome/blob/dev/esphomeyaml/components/wifi.py

OttoWinter, 2019-02-01:
You seem to not understand the reason for that option correctly. It is entirely for setups where your router can resolved the hostnames itself (for example in a VLAN).

For mDNS, ONLY `.local` is valid. Please only use the domain option if you have a setup a DNS server that resolves these hosts through its own system (like hostnames and DHCP). local (the default) is for _mDNS_.

closed 2019-02-01 by OttoWinter

xlfe, 2019-02-03:
Sorry @OttoWinter - you're 100% correct about mDNS and .local

The problem I'm having is that mDNS resolution is not working.

```
INFO Starting log output from powermon.local using esphomelib API
WARNING Error resolving IP address of powermon.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphomelib.com/esphomeyaml/components/wifi.html#manual-ips)
```

Am I correct in thinking that mDNS is not enabled in the esphomeyaml docker container? Ie no avahi or libnss-mdns installed and configure?

I can see you're working on integrating mDNS resolution already - which I assume will remove the need for adding mDNS to the underlying OS?

xlfe, 2019-02-03:
When running on Hass.IO it all works - my guess is that HassIO is setting up a dns resolver that handles the lookup - so running esphomeyaml as a docker container on a host which doesn't do that is my problem?