esphome/esphome-core #463: Auto-load esphome component upon discovery by Home Assistant

xlfe, 2019-02-01:
**Is your feature request related to a problem/use-case? Please describe.**

Adding many esphome components to Home Assistant is time consuming. 

**Describe the solution you'd like**

Add an option to enable auto-loading esphome components upon discovery

**Additional context**

See https://developers.home-assistant.io/docs/en/creating_component_discovery.html#auto-loading-your-component-upon-discovery

xlfe, 2019-02-01:
@brandond the discovery works - yes - so you get a notification that there are discovered components - but the next step, adding them to home assistant is still manual?

See SERVICE_HANDLERS in https://github.com/home-assistant/home-assistant/blob/7db28d3d91869cc1b2898342bcfce096930ad449/homeassistant/components/discovery.py

There is no esphome entry, which means they are not auto added...

brandond, 2019-02-01:
I must be confused - what do you need to add them to? The UI? They're available to be used as switches, sensors, lights, etc as soon as they're discovered.

brandond, 2019-02-01:
It sounds like the ask is to have MQTT and esphome native discovery fire a PLATFORM_DISCOVERED event when a new device is added? I believe that both the esphome and mqtt platforms are maintained in the homeassistant codebase, so the issue probably belongs over there.

brandond, 2019-02-01:
I would put in an issue or PR in the home-assistand project referencing:

https://github.com/home-assistant/home-assistant/tree/dev/homeassistant/components/esphome
https://github.com/home-assistant/home-assistant/tree/dev/homeassistant/components/mqtt

xlfe, 2019-02-01:
Hmm if they are available to be used then sorry I thought you had to do something manual. As you were!

xlfe, 2019-02-03:
Okay so let me see if I have made a mistake. 

New Hass.IO install - discovers a whole bunch of ESPHome devices 

[image]

Based on what you said above 

> I must be confused - what do you need to add them to? The UI? They're available to be used as switches, sensors, lights, etc as soon as they're discovered.

I thought they could be used. But there are no entities available :- 

[image]

So they are not available for use unless I manually go into the integrations panel and click configure.

Once I manually configure them, they are "available to be used"

[image]

As an aside - despite being able to discover the devices using their name, esphome doesn't seem to then be able to connect to them 👎 

[image]

OttoWinter, 2019-02-03:
> So they are not available for use unless I manually go into the integrations panel and click configure.

That is how the integrations screen is _supposed_ to work. Previously, Home Assistant would add _all_ integrations automatically and it was really hard to exclude an integration. The new config entries *require* developers to make it an opt-in action (also because otherwise you could pretty much achieve remote code execution without user accepting).

Also, considering adding it to HA is two button presses, I couldn't care less about this - it takes way more time to set up a node on the ESPHome side of things.

> Assuming it is actually online, you might try a more recent Dev release that has a built-in mDNS resolver for hosts that don't support it natively.

That PR isn't merged yet 😉 Also, Home Assistant doesn't use esphomeyaml for the API (aioesphomeapi package). So the mDNS change needs to be done there too (plus PR in Home Assistant, which takes some time to go through PR process)

xlfe, 2019-02-04:
Thanks @OttoWinter and @brandond - ill keep an eye on the releases for when native mDNS support lands in esphome.

My use case is for a stateless deploy of ha with eh nodes to a non Hass.IO docker host. I think I'll just write a custom component to handle adding the nodes into ha automatically.

Thanks again

closed 2019-02-04 by xlfe