esphome/esphome-core #479: MQ-2 Sensor.

Iv711, 2019-02-07:
**What new integration would you wish to have?**
MQ-2 

**If possible, provide a link to an existing library for the integration:**
https://www.mysensors.org/build/gas
**Please describe your use case for this integration and alternatives you've tried:**
For earlier, I apologize for my English.
I want to refuse mysensors, in favor of esphome, but not to call on the support of one sensor - mq-2. I'd love to see him in the foreseeable future.

**Additional context:**
Thx.

OttoWinter, 2019-02-07:
These are just analog sensors - they give the gas value as an analog voltage reading. So the `adc` sensor supports MQ-2s already. All you need will be to come up with a formula that converts the analog voltage reading to a gas reading (for example from datasheet)

Please see TEMT6000 example for how to do this.

closed 2019-02-07 by OttoWinter

Iv711, 2019-02-07:
In my sensors sketch:
`float BaseMQ::calculateResistance(int rawAdc) const {
  // float vrl = rawAdc*(5.0 / 1023);  // Так по класcике 
  float vrl = rawAdc * (5.0 / 1551);  // Так у нас
  float rsAir = (5.0 - vrl) / vrl * getRL();
  if (_corr_factor_set) {
    rsAir = rsAir / _corr_factor;
  }
Help me write the formula for - lambda correctly
Thx.

OttoWinter, 2019-02-08:
It's not too hard to convert it. You can pretty much copy paste it:

```cpp
float vrl = x * 5.0;
// you didn't post getRL method, so no way I know what that does
float rsAir = (5.0 - vrl) / vrl * getRL();
return rsAir;
``

Also I can tell you your formula is probably wrong. The * 5.0 looks a lot like you're using 5V voltage level, but the ESP operates at 3.3V which would require you to use some external circuitry to scale the voltage.

Iv711, 2019-02-08:
This sensor works from 5 volts (as far as I know), but the terminal gives 5.0 volts
getRL - This probably means
protected:
  unsigned long readScaled(float a, float b) const;
  virtual float getRoInCleanAir() const = 0;
  virtual int getRL() const = 0;
And
 private:
  // Резистор установленный на плату (кОм)
  virtual int getRL() const { return 1; }
  // коэффициент чистого воздуха
  virtual float getRoInCleanAir() const { return 9.83; }

Esiravegna, 2019-02-08:
Based on my experience, the analgo reader of the ESP8266 doesn't sit along well with the MQ2. I had better luck using an arduino as an I2C slave, reading the sensors with that and then the i2C to the ESP8266