esphome/esphome-core #489: [RGB Light Platform] Remember brightness and color after restart

pavolholes, 2019-02-10:
**Is your feature request related to a problem/use-case? Please describe.**

Problem: I toggle on the [RGB Light platform](https://esphome.io/components/light/rgb.html), set the brightness to 50%, set the color to warm white.
Now suddenly the electricity blackout happens (caused by me playing with home automation :) ).
When the power is restored, the light is off. So I toggle it on, the brightness goes to 100% and the color is white (not warm white).

**Describe the solution you'd like**

When the powers is restored, light might be off, but when I toggle it on it would be great if it would remember the last brightness level and the color value.

**Additional context**

I'm running esphomelib v1.10.1 compiled on Feb 10 2019, 19:42:20

```
output:
  # Enable three pins as outputs with software PWM output (ESP8266) for RGB light.
  # Lower frequency from default 1000Hz to 100Hz to lower filckering sound.
  - platform: esp8266_pwm
    pin: D5 #GPIO14
    frequency: 100 Hz
    id: 'outputRgbRed'
  - platform: esp8266_pwm
    pin: D6 #GPIO12
    frequency: 100 Hz
    id: 'outputRgbGreen'
  - platform: esp8266_pwm
    pin: D7 #GPIO13
    frequency: 100 Hz
    id: 'outputRgbBlue'

light:
  # Create RGB light which will maintain output defined above.
  - platform: rgb
    id: 'lightRgb'
    name: 'Livingroom RGB light'    
    red: outputRgbRed
    green: outputRgbGreen
    blue: outputRgbBlue
```

brandond, 2019-02-10:
ESP8266s do not have any persistent NVRAM. They can remember states across reboots, but not power loss - so this might be only partially achieveable.

pavolholes, 2019-02-10:
ESP8266 have EEPROM builtin, do you think this might be used? I used it in past in some simple sketches.

#include <EEPROM.h>

https://arduino-esp8266.readthedocs.io/en/latest/libraries.html
Examples: https://github.com/esp8266/Arduino/tree/master/libraries/EEPROM

brandond, 2019-02-10:
On ESP8266 it is faked with flash memory, which has a limited number of write cycles.

> EEPROM library uses one sector of flash located just after the SPIFFS

OttoWinter, 2019-02-11:
Exactly. ESPHome doesn't use the EEPROM functionality because of a few reasons:
* Limited number of write cycles, if some component writes a lot of updates you will wear out the flash over time.
* Slow. Writing to flash on the ESP8266 is slow, when erasing a page for writing you can get delays of up to a couple hundred milliseconds if I remember correctly.
* Buggy. The ESP8266 EEPROM implementation is known to be unstable when using the hw timer at the same time (for example for PWM)
I'm closing this because it's a won't fix - and there are other issues like this one in the issue tracker already.

closed 2019-02-11 by OttoWinter

markcarline, 2019-02-20:
I see this issue too - but wonder if either via the HA API or MQTT if esphome devices can read HA in some way and restore the "on" and "colour" value back to the before power lost state ?