esphome/esphome-core #496: 1549955558: Socket error

Michael0yodi, 2019-02-12:
Device: D1 Mini, Sonoff Basic R2

Device is losing connection all the time, its restored automatically. the same behavior on both devices, tested power saving modes and other wifi settings without success, device responds to ping while losing mqtt connection.

Home Assistant: 0.87.1
Mosquitto broker hassio:
Latest esphomeyaml 

testmini-dc4f221d5129, disconnecting.
1549955558: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549955558: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549955862: Socket error on client testmini-dc4f221d5129, disconnecting.
1549955862: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549955862: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549956166: Socket error on client testmini-dc4f221d5129, disconnecting.
1549956166: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549956166: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549956188: Saving in-memory database to /data/mosquitto.db.
1549956470: Socket error on client testmini-dc4f221d5129, disconnecting.
1549956470: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549956470: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549956774: Socket error on client testmini-dc4f221d5129, disconnecting.
1549956774: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549956774: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').
1549957078: Socket error on client testmini-dc4f221d5129, disconnecting.
1549957078: New connection from 172.16.1.140 on port 1883.
[INFO] found mqttuser on Home Assistant
1549957078: New client connected from 172.16.1.140 as testmini-dc4f221d5129 (c1, k1800, u'mqttuser').

Home assistant logbook:
08:44
Test GP1 turned off
08:44
Test GP1 changed to unavailable
08:44
Test GP1 turned off
08:44
Test GP1 changed to unavailable
08:44
Test GP1 turned off
08:44
Test GP1 changed to unavailable
08:43
Test GP1 turned off
08:43
Test GP1 changed to unavailable
08:37
Test GP1 turned off
08:37
Test GP1 changed to unavailable
08:32
Test GP1 turned off
08:32
Test GP1 changed to unavailable
08:27
Test GP1 turned off
08:27
Test GP1 changed to unavailable
08:22
Test GP1 turned off
08:22
Test GP1 changed to unavailable
08:17
Test GP1 turned off
08:17
Test GP1 changed to unavailable
08:12
Test GP1 turned off
08:12
Test GP1 changed to unavailable
08:07
Test GP1 turned off
08:07
Test GP1 changed to unavailable

aero83, 2019-02-12:
I am having same issue. Socket error every 5 minutes. Causes the relay to click, and the lamp to flash on every disconnection. 

Sonoff Basic. 
Flashed bin file compiled from HASSio this morning, addon. Mosquitto broker. 

HA log. 
Logbook
Showing entries for
February 12, 2019
 
 
Period
1 day
 
 
Entity
February 12, 2019
9:47 AM
Sonoff Basic Relay turned on
9:47 AM
Sonoff Basic Button turned off
9:47 AM
Sonoff Basic Relay changed to unavailable
9:47 AM
Sonoff Basic Button changed to unavailable
9:42 AM
Sonoff Basic Relay turned on
9:42 AM
Sonoff Basic Button turned off
9:42 AM
Sonoff Basic Relay changed to unavailable
9:42 AM
Sonoff Basic Button changed to unavailable
9:37 AM
Sonoff Basic Relay turned on
9:37 AM
Sonoff Basic Button turned off
9:37 AM
Sonoff Basic Relay changed to unavailable
9:37 AM
Sonoff Basic Button changed to unavailable
9:32 AM
Sonoff Basic Relay turned on
9:32 AM
Sonoff Basic Button turned off
9:32 AM
Sonoff Basic Relay changed to unavailable
9:32 AM
Sonoff Basic Button changed to unavailable
9:27 AM
Sonoff Basic Relay turned on
9:27 AM
Sonoff Basic Button turned off
9:27 AM
Sonoff Basic Relay changed to unavailable

**Mosquitto log**
2019-02-12T15:11:37.899736850Z 1549984297: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:16:49.571702120Z 1549984609: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:16:49.586663330Z 1549984609: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:16:49.586681151Z 1549984609: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:22:01.135613692Z 1549984921: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:22:01.139857356Z 1549984921: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:22:01.139869428Z 1549984921: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:27:11.697340263Z 1549985231: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:27:11.710693547Z 1549985231: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:27:11.710711603Z 1549985231: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:32:22.240031053Z 1549985542: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:32:22.247380860Z 1549985542: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:32:22.247496874Z 1549985542: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:37:33.086094060Z 1549985853: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:37:33.092104140Z 1549985853: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:37:33.092255526Z 1549985853: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:42:43.702886966Z 1549986163: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:42:43.708544174Z 1549986163: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:42:43.708591390Z 1549986163: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15).

2019-02-12T15:47:54.243089073Z 1549986474: Socket error on client ripper_lamp-5ccf7f958d57, disconnecting.

2019-02-12T15:47:54.247188368Z 1549986474: New connection from 192.168.10.135 on port 1883.

2019-02-12T15:47:54.247371468Z 1549986474: New client connected from 192.168.10.135 as ripper_lamp-5ccf7f958d57 (c1, k15)

bk8190, 2019-02-12:
I'm having a similar problem.  I use the HTTP API and my Home Assistant shows that all of my boards (Lolin D1 Mini) temporarily become "unavailable" about 20-30 times per day.  They always reconnect within several seconds of becoming unavailable.

Here's a Home Assistant log snippet:

`2019-02-12 11:40:03 INFO (MainThread) [aioesphomeapi.client] 192.168.X.Y: Ping Failed!
2019-02-12 11:40:03 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.X.Y
2019-02-12 11:40:03 INFO (MainThread) [aioesphomeapi.client] 192.168.X.Y: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
2019-02-12 11:40:03 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to 192.168.X.Y
`

Sometimes it happens to multiple devices at once, where multiple devices drop out at once and then rejoin several seconds later.