esphome/esphome-core #513: Error OTA update ESP8266 12e

Maikl3, 2019-02-21:
I received a repeated error related to the GPIO parameter.
mode: WAKEUP_PULLDOWN and mode: WAKEUP_PULLUP.
When using these parameters, the ESP8266 12e module remains on the network, is available to be managed, but it is no longer possible to update its configuration over the air.
The process stops at 2% or 3%
The only way out is to get the module out of the device again and flash it via USB.
And it is very difficult when the module is installed in an inaccessible place.
Maybe there is a solution to this problem?

`switch: 
  - platform: gpio
    pin:
      number: GPIO5
      mode: WAKEUP_PULLDOWN
    name: "sound"
    `
[image]

P.S I checked it on two modules of version ESP8266 12E
At first I used the parameter
mode: WAKEUP_PULLUP
On the second
mode: WAKEUP_PULLDOWN
In the second case, I needed this GPIO5 pin when switching on the module did not turn on the relay that is connected to it. How else can you get a low level when starting a module?

OttoWinter, 2019-02-21:
No description provided?

OttoWinter, 2019-02-21:
Please open in the issues repo

closed 2019-02-21 by OttoWinter